dresden-elektronik/deconz-rest-plugin #6014: Timezone changes not reflected on ConBee II Time Cluster

aryelevin, 2022-05-03:
## Describe the bug
Changing timezone via phoscon app doesn't affect the time of ConBee II.
After changing time zone from Cairo to Moscow, the DST stays the same while Moscow don't have any DST ever.
Tried many other timezones, nothing happens on the ConBee level (on phoscon it saves the new time zone properly).

## Steps to reproduce the behavior
1. Open phoscon -> Gateway Settings -> Advanced -> Change timezone to something different than the current set timezone.

## Expected behavior
The ConBee II time cluster should reflect the change.

## Screenshots
NA

## Environment
 - Host system: Ubuntu 20 on Intel i5 PC with 16GB Ram
 - Running method: Ubuntu
 - Firmware version: (26720700)
 - deCONZ version: (2.15.3)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes, more ConBee II (total 4)
## deCONZ Logs
NA

## Additional context
NA

manup, 2022-05-10:
@SwoopX any ideas? I'm not so familiar with the time cluster code :)

SwoopX, 2022-05-10:
@manup Yeah, right, let the poor intern sweat 😂 

Well, from my perspective (and only having the information at hand above), everything's fine and I wouldn't consider this as a bug. The time presented is based on system time and consequently, any change in Phoscon doesn't impact time cluster data.

aryelevin, 2022-05-10:
So how do you expect accessories to know the correct time and date??
That doesn’t make any sense.

aryelevin, 2022-05-10:
But accessories doesn’t query the system’s time, only the zigbee coordinator time.
Unless this gonna change the zigbee time (system time).

Mimiix, 2022-05-10:
> But accessories doesn’t query the system’s time, only the zigbee coordinator time. Unless this gonna change the zigbee time (system time).

Please read @SwoopX his comment thoroughly.

closed 2022-05-11 by aryelevin