dresden-elektronik/deconz-rest-plugin #6015: _TZ3000_n2egfsli - TS0203 - Tuya Smart Zigbee Door Window Contact Sensor

isaaccs, 2022-05-04:
## Device

- Product name: Zigbee Door sensor
- Manufacturer: _TZ3000_n2egfsli
- Model identifier: TS0203
- Device type : 
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Other clusters that are not mentioned above
[image]
[image]

Smanar, 2022-05-04:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_n2egfsli",
  "modelid": "TS0203",
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

isaaccs, 2022-05-04:
where should I put it ?

Smanar, 2022-05-04:
> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/deconzuser/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/2

isaaccs, 2022-05-04:
I use a docker version on my Unraid system

majorsl, 2022-05-07:
> where should I put it ?

Where ever you mapped /opt/deCONZ to, they'll be a devices directory in there.  Just put the DDF in that.

Smanar, 2022-05-07:
Ha ?
It's a new procedure again ? In my mind the HA plugin use docker so was the same procedure ? it's not the same docker image ?

majorsl, 2022-05-07:
> Ha ?
> It's a new procedure again ? In my mind the HA plugin use docker so was the same procedure ? it's not the same docker image ?

I'm don't know about HA and how they do it.  I run just the version on my RPi with the standard Docker package from deconzcommunity/deconz:latest, my HA is on a different system so I can update deconz or HA and not bring down the entire system for one or the other.  It should be the same for @isaaccs with Unraid assuming he's doing the same as me, which it sounds like.

With that path mapped to the host system, as it should be, all your settings are there so when you update deconz it won't wipe everything out.  To edit, you'll use the text editor of your choice on your docker host system: vim, nano, pico, whatever.  When you and I were testing the DDF for me, I just open a new file in that directory with sim, copy/pasted your excellent DDF and saved with the name you gave.  Worked perfectly and no need to enter the container like on HA.

I could be mistaken, but it seems no other DDFs that are included with deconz are in that directory, it is simply another place that deconz looks for user created ones - which is handy and good design.

isaaccs, 2022-05-10:
`/root/.local/share/dresden-elektronik/deCONZ` is mapped to `/mnt/user/appdata/deconz`
But i dont have a device folder inside, so I'll create it and add my DDF

isaaccs, 2022-05-10:
[image]
[image]

isaaccs, 2022-05-10:
> Hello, can you try this DDF
>
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZ3000_n2egfsli",
>   "modelid": "TS0203",
>   "vendor": "Tuya",
>   "product": "Smart Zigbee Door Window [...]

seems it not works :/

majorsl, 2022-05-10:
https://github.com/deconz-community/deconz-docker

Go down to the command line example section. You should be mapping /opt/deCONZ to a directory on your host. When you do, you should find devices as one of the directories. It should also be the only directory you need to map.

Mimiix, 2022-05-10:
> seems it not works :/

Did you enable Silver ?

Mimiix, 2022-05-10:
I just wrote a little guide:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-change-DDF-modes-and-enable-or-disable-DDF-levels

isaaccs, 2022-05-10:
control is not available in my panel

Mimiix, 2022-05-10:
What deCONZ version are you on?

isaaccs, 2022-05-10:
2.05.77

Smanar, 2022-05-10:
lol, ha yes 2.05, I think you are the user with the older version I have see ^^.

And yes, you need to enable silver, or edit the DDF and put "Gold" as status. "Gold" DDF are always used.

isaaccs, 2022-05-10:
thanks it works !!

closed 2022-05-16 by manup