dresden-elektronik/deconz-rest-plugin #6017: Aqara temperature sensor (v2.15.3)

Arjan-21, 2022-05-04:
**## Describe the bug**

I work with deconz in Home Assistant. Just updated to [v2.15.3] and after the update my temperature is 0 °C. 

**## Steps to reproduce the behavior**

Update to 2.15.3 and after that the first update of temperature its off.

**## Expected behavior**

Hopefully to get the right temperature

**## Screenshots**
In the first picture you see (Sensor Temp KK) the right temperature, but when I click the button I see 0 °C
[image]

[image]

**## Environment**

Version:  2.15.03 / 21-4-2022
Firmware:   26700700

 - Host system: Raspberry Pi 4
 - Running method: Home assistant with deconz add-on
 - Firmware version: 26720700
 - deCONZ version: 2.15.03 / 21-4-2022
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? Yes, SSD disk

LeoeLeoeL, 2022-05-04:
Which was your previous version of firmware?

beaulebens, 2022-05-04:
I'm seeing something similar, [reported over here](https://github.com/home-assistant/addons/issues/2476) originally with additional details. In my case, temperature is dropping to -148F, which happens to be -100C (maybe some kind of hardcoded lower bound?). It's not affecting all sensors -- as far as I can tell only 2 of about 8 in my network.

Mimiix, 2022-05-04:
Closing this issue as it's a duplicatie of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5974

closed 2022-05-04 by Mimiix