dresden-elektronik/deconz-rest-plugin #6018: Nous Smart Zigbee Socket A1Z

TheNice4GH, 2022-05-04:
## Device

- Product name: Smart Zigbee Socket
- Manufacturer: _TZ3000_ksw8qtmt
- Model identifier: Adapter plug
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Switch

## Screenshots

[image]

### Basic
[image]

### On/Off
[image]

### Simple Metering
[image]

### Other clusters that are not mentioned above
[image]

Mimiix, 2022-05-04:
Closing as it's a duplicatie of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6018

closed 2022-05-04 by Mimiix

TheNice4GH, 2022-05-04:
Sorry my bad... I thought I searched correctly

Mimiix, 2022-05-04:
No worries :) I saw you have it working already so that's an amazing outcome! pro tip: search for the manufacturer name in the git next time