dresden-elektronik/deconz-rest-plugin #6023: SUNRICHER 3 in 1 zigbee push button remote

olivier6931, 2022-05-06:
Hello Guys,
I have a device request for my new sunricher 3 in 1 remote.
Thank in advance for your help :)

## Device

- Product name: 3 in 1 zigbee push button remote
- Manufacturer: SUNRICHER
- Model identifier: ZG2855-RGB
- Device type :
  - Remote

## Screenshots

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### Diagnostics
[image]

### ZLL Commissioning
[image]
[image]

### Groups
[image]

### Scenes
[image]
[image]

### On/Off
[image]

### Level Control
[image]

### OTAU
[image]

### Color control
[image]

### ZLL Commissioning
[image]

Thanks!

olivier6931, 2022-06-02:
Keeping the thread active

olivier6931, 2022-06-24:
Hello guys,

Do you have any news on my device request?
Did I forget some information?

Thanks again

olivier6931, 2022-09-22:
updated basic informations

Mimiix, 2022-10-07:
@Smanar  is this something you can do :)?

Smanar, 2022-10-07:
Can try with this DDF

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "Sunricher",
	"modelid": "ZG2855-RGB",
	"product": "3 in 1 zigbee push button remote",
	"sleeper": true,
	"status": "Gold",
	"subdevices": [{
		"type": "$TYPE_SWITCH",
		"restapi": "/sensors",
		"uuid": ["$address.ext", "0x01", "0x0006"],
		"items": [{
				"name": "attr/id"
			},
			{
				"name": "attr/lastannounced"
			},
			{
				"name": "attr/lastseen"
			},
			{
				"name": "attr/manufacturername"
			},
			{
				"name": "attr/modelid"
			},
			{
				"name": "attr/name"
			},
			{
				"name": "attr/swversion"
			},
			{
				"name": "attr/type"
			},
			{
				"name": "attr/uniqueid"
			},
			{
				"name": "config/battery",
				"parse": {
					"at": "0x0021",
					"cl": "0x0001",
					"ep": 1,
					"eval": "Item.val = Attr.val"
				},
				"read": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0001",
					"at": "0x0021"
				},
				"refresh.interval": 4000,
				"awake": true
			},
			{
				"name": "config/on"
			},
			{
				"name": "config/reachable"
			},
			{
				"name": "state/buttonevent",
				"awake": true
			},
			{
				"name": "state/lastupdated"
			}
		]
	}],
	"bindings": [{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0001",
			"report": [{
				"at": "0x0021",
				"dt": "0x20",
				"min": 60,
				"max": 3600,
				"change": "0x00000001"
			}]
		},
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0006"
		},
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0008"
		}
	]
}
```

For the moment I m using the cluster on/off and setlevel, so not all fatures will work.
After having added the DDF the device can be reconised in deconz, but will not work, need to make the button map file.
So just enable log, in deconz / help / debug view with flag "info" and "info_l2" and share the error line you will when using a button.
Will be something like.

`        DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

olivier6931, 2022-10-07:
Hello @Smanar ,

Thanks for the DDF. I managed to put it in deconz & get said logs.
Here are they : 

- Increase lightness button : 
_[INFO] - No button map for: ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP_WITH_ON_OFF (0x06), payload: 00200000, zclSeq: 84_
- Decrease lightness button : 
_[INFO] - No button map for: ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP_WITH_ON_OFF (0x06), payload: 01200000, zclSeq: 85_
- On/OFF Button : 
_[INFO] - No button map for: ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 89_

Nothing seems to happen in the logs for 2 lasts buttons (or I don't know how to read it) : change color (RGB one) and change white temperature (CW one).

Thanks again for your help!

Smanar, 2022-10-07:
Ok, so first I m not enabling the "groupcast" feature, do you need it ? Using it with the old webapp it's used to make direct bind to a remote to a light gruop for exemple ?

And now more complex, you need to edit the file "button_maps.json" present in the folder "devices" too.

```
        "sunricherZG2855Map": {
            "vendor": "Sunricher",
            "doc": "Zigbee 3 in 1 remote ZG2855-RGB",
            "modelids": ["ZG2855-RGB"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF ", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Increase lightness"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF ", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Decrease lightness"],
            ]
        },
```

(You can find other "Sunricher" inside).

for the missing button, try with adding to the DDF at the end

```
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0300"
		}
```

Take care at comma, need to add one on previous binding

```
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0008"
		},
		{
			"bind": "unicast",
			"src.ep": 1,
			"cl": "0x0300"
		}
```

Thoses 5 line will enable the binding for the color cluster, as I think it's the used cluster for the feature you describe.
Better to re-include the device, without deleting the previous entry, for deconz reconfigure bind (after having edited the DDF)

olivier6931, 2022-10-08:
Hello,

> Ok, so first I m not enabling the "groupcast" feature, do you need it ? Using it with the old webapp it's used to make direct bind to a remote to a light gruop for exemple ?

I'm not sure if using it : I managed to make my button work with "direct pairing" with touchlink function from the device (like a normal pairing but with other buttons sequence & very near the light it manage).

> And now more complex, you need to edit the file "button_maps.json" present in the folder "devices" too.
>         "sunricherZG2855Map": {
>          ...
>         },

I found the file and edited it (though I'm not sure if it will be permanent as the file doesn't seem to be on a directory binded to another one on the docker host).

> for the missing button, try with adding to the DDF at the end

done!

> Better to re-include the device, without deleting the previous entry, for deconz reconfigure bind (after having edited the DDF)

Not sure how to do it, as I think I have to factory reset my device to reinclude it. Doesn't this action remove it from deconz too ?

Smanar, 2022-10-08:
>I'm not sure if using it : I managed to make my button work with "direct pairing" function from the device (like a normal pairing but with other buttons sequence & very near the light it manage).

I m  not sure to understand, bu if you use a procedure to "direct bind" a remote to a bulb, bothe device will leave the network to make a new one to be together, so you will not be able to make something with them.
There is a way to make a "direct bind" but using zigbee request, like that devices will not move.

>I found the file and edited it (though I'm not sure if it will be permanent as the file doesn't seem to be on a directory binded to another one on the docker host).

Ha yes, you are using HA or a clean deconz docker ? If you are using the second one, I m almost sure you have it, it's the same folder you are using for DDF.

>Not sure how to do it, as I think I have to factory reset my device to reinclude it. Doesn't this action remove it from deconz too ?

Yes exaclty, but at every re-inclusion deconz will reconfigure them, it's a good way because this devices is a slepper, so hard to set a bind. During the inclusion the device is awake, and deconz remake all bind, so it's a the procedure the more sure for me.

olivier6931, 2022-10-08:
> I m not sure to understand, bu if you use an hardware procedure to "direct bind" a remote to a bulb, bothe device will leave the network to make a new one to be together, so you will not be able to make something with them. There is a way to make a "direct bind" but using zigbee request, like that devices will not move.

I try many things with direct link and in deconz (with groups, and DDF files i tried to create), and finaly get it to work without it being really known by deconz. I did that on may or april, so now i'm not really sure to remind all of my tests (and I'm not a Zigbee expert to be honest).
But since then, it's working not that bad (except none information get to HA, like battery, status, etc...) :)

> Ha yes, you are using HA or a clean deconz docker ? If you are using the second one, I m almost sure you have it, it's the same folder you are using for DDF.
 
Exactly, I'm using HA with deconz "core addon". In this case, I had to put my modified ZG2855-RGB.json file on a docker bind to keep the modifications (/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices on the host which is mounted on /data/.local/share/dresden-elektronik/deCONZ/devices/ in the container)
but the button_maps.json is only accessible from within the container in /usr/share/deCONZ/devices/button_maps.json

> Yes exaclty, but at every re-inclusion deconz will reconfigure them, it's a good way because this devices is a slepper, so hard to set a bind. During the inclusion the device is awake, and deconz remake all bind, so it's a the procedure the more sure for me. And no need to delete it, like that, all will be invisible.

Did that, and I think it's working, there are the logs :
```
[INFO] - Button 1002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 20
[INFO] - Button 2002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Off, payload: None, zclSeq: 40
```

Smanar, 2022-10-08:
To make direct bind with deconz, you can use the old web app (the link is on phoscon / help), you just need to put light you want to bind in the remote group.
But the remote need to support the group, and ATM I haven't added it in the DDF yet.

>Did that, and I think it's working, there are the logs :

Yep, the button map is working but no more error message for not working buttons ?
The only missing cluster is the Scene cluster, but from your description thoses 2 button are using the color cluster.

olivier6931, 2022-10-08:
> To make direct bind with deconz, you can use the old web app (the link is on phoscon / help), you just need to put light you want to bind in the remote group. But the remote need to support the group, and ATM I haven't added it in the DDF yet.

Ok I understand. No problem ;)

> Yep, the button map is working but no more error message for not working buttons ? The only missing cluster is the Scene cluster, but from your description thoses 2 button are using the color cluster.

No more error, I confirm that.
Thanks again !

Smanar, 2022-10-09:
>Ok I understand. No problem ;)

Not a problem, if you want them, I can add them, just need more work for you, more tests to do.

>I had these kind of info :

Nice so new button map (have made a typo on last one for Level control, one more space)

```
        "sunricherZG2855Map": {
            "vendor": "Sunricher",
            "doc": "Zigbee 3 in 1 remote ZG2855-RGB",
            "modelids": ["ZG2855-RGB"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Increase lightness"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Decrease lightness"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR", "0xEB", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color"],,
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR_TEMPERATURE", "0x30", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color temperature"],
            ]
        },
```

olivier6931, 2022-10-09:
uhhh... something bad happend :

```
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0xAD01, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 59
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0xAD01, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP_WITH_ON_OFF (0x06), payload: 00200000, zclSeq: 61
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0xAD01, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP_WITH_ON_OFF (0x06), payload: 00200000, zclSeq: 63
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0xAD01, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), command: MOVE_TO_COLOR (0x07), payload: 3F553F5500000000, zclSeq: 65
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0xAD01, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), command: MOVE_TO_COLOR_TEMPERATURE (0x0A), payload: 300100000000, zclSeq: 66
[INFO] - No button map for: ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 67
```

I modified the button map as you said and reincluded the device.
After I tried to remove and readd the device, same error.
I even tried to restart the container, and add again the device, same error.

Don't know what I miss...

Smanar, 2022-10-09:
It's my fault there is too much comma

```
        "sunricherZG2855Map": {
            "vendor": "Sunricher",
            "doc": "Zigbee 3 in 1 remote ZG2855-RGB",
            "modelids": ["ZG2855-RGB"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Increase lightness"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Decrease lightness"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR", "0xEB", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR_TEMPERATURE", "0x30", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color temperature"]
            ]
        },
```

olivier6931, 2022-10-10:
Hello,

Same state :
```
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0x0901, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 48
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0x0901, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP_WITH_ON_OFF (0x06), payload: 00200000, zclSeq: 49
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0x0901, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP_WITH_ON_OFF (0x06), payload: 01200000, zclSeq: 50
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0x0901, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), command: MOVE_TO_COLOR (0x07), payload: F37D566E00000000, zclSeq: 51
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0x0901, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), command: MOVE_TO_COLOR_TEMPERATURE (0x0A), payload: B90000000000, zclSeq: 52
[INFO] - No button map for: ZG2855-RGB, broadcast to: 0x0901, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 53
```

olivier6931, 2022-10-16:
Hello,

Sorry for the late answer, I was quite busy this week...
So, now I understand that it work, doesn't it ?

```
[INFO] - Button 1002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 71
[INFO] - Button 3002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Increase lightness, payload: 00200000, zclSeq: 72
[INFO] - Button 4002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Decrease lightness, payload: 01200000, zclSeq: 73
[INFO] - Button 5002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), action: Move to color, payload: CE371C5A00000000, zclSeq: 74
[INFO] - Button 6002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), action: Move to color temperature, payload: B90000000000, zclSeq: 75
[INFO] - Button 2002 - ZG2855-RGB, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Off, payload: None, zclSeq: 76
```

Smanar, 2022-10-16:
Yep, I think now you are able to see button event.
I don't remeber your third app, but you can see them in phoscon/help/API Information/event/sensor/....

olivier6931, 2022-10-16:
Yes, I use Phoscon and I can see it & map it :

[image]

Thank you very much! :)

olivier6931, 2022-10-16:
Another last question :
why using 2 buttons for on & off like I see in your code  ?

```
[1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
[1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"]
```

Is it linked to zigbee functionnality ? or is it because I push the "stop" button each time I gave you an example ? 😄 
because on the switch, it's physically one button which is use to toggle the light :
[image]

Smanar, 2022-10-16:
Pr made here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6425

If you want to compare DDF, to be sure I haven't forget something.

Smanar, 2022-10-16:
Yep, your button is an automatic toogle.
I m using 2 different return for you will be able to use them in automation too.
I can convert it to a toogle, but you will have only 1 return, you can have 1 "toggle" or 1 "on" + 1 "off", as you want. But will be not same for automation.

The return is X00Y
X is the button number and Y the action.
I m cheating to create a fake button.

olivier6931, 2022-10-16:
Oh ok, I understand!
I'll check the DDF a bit latter this morning.

thanks again !

olivier6931, 2022-10-16:
The DDF seems OK in your PR, I only notice this difference with the file edited at home :
`  "path": "/devices/ZG2855-RGB.json",`

But I'm not sure it's really important...

Smanar, 2022-10-16:
Nice, thx.

And nope, it's the editor that add this line itslef, it's useless, as users can change the file name.

closed 2022-10-16 by olivier6931