dresden-elektronik/deconz-rest-plugin #6031: TS004F _TZ3000_4fjiwweb

vvbogdanov87, 2022-05-08:
## Device

- Product name: Tuya ZigBee Smart Knob
- Manufacturer: _TZ3000_4fjiwweb
- Model identifier: TS004F
- Device type : 
  - Switch
  - Dimmer

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
#### Power Configuration
[image]

tomky, 2022-05-29:
Also have this device, by default recognize as a switch and a light (no effects). Will be great if it can be supported.

Smanar, 2022-05-29:
It's a battery switch ?
They are easy to add if they haven't special feature https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6068

Else can you share the DDF skeleton for I finish it ?

tomky, 2022-05-29:
Not sure what DDF skeleton is. After comparing the above screenshots with others I think maybe it is this one?
[image]

Smanar, 2022-05-29:
Here you have the procedure to create a DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

You can finish it yourself, you will see it s realy easy, but on my side I need a starting point to avoid typo (and you a base to replace the file) . some part can be autofilled

- Fill the "basic cluster identification" 
- Drag and drop a ZHAswitch
- Set the status to "gold"
- Save the file (remember the path for later)
- make "hot relaod" to test your changes, but you will miss stuff.

Share the DDF (the file contain and copy/paste the third tab.

tomky, 2022-05-30:
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_4fjiwweb",
  "modelid": "TS004F",
  "product": "TS004F",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0104",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x1000"
        ],
        "out": [
          "0x0005",
          "0x0006",
          "0x0008"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "description": "The current device battery level in 0&ndash;100&thinsp;%."
        },
        {
          "name": "config/group",
          "description": "comma seperated list of groups which a device controls."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_DIMMER_SWITCH",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

Smanar, 2022-05-30:
Arf, lol, have just see what kind of device it's, not a simple switch ^^

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_4fjiwweb",
  "modelid": "TS004F",
  "product": "TS004F",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
  ]
}
```

For the moment have put a bind only for the cluster on/off so the setlevel will probably don't work (I think it's the rotation)

Can you try this DDF, then on deconz / help / debug view, enable flag "info", "info_l2" and use the device, you will have error message like this one

`23:50:37:152 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 3`

Can you try wich one commands are working (if you have the error message it mean it can work, if you don't have the error message we miss a binding)

vvbogdanov87, 2022-06-01:
Button push
```
[INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 9
```
Rotation clockwise(one step):
```
[INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 000D0100, zclSeq: 3
```
Rotation counterclockwise(one step):
```
[INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 010D0100, zclSeq: 3
```
The rotation payload is different depending on the direction and number of steps in one rotation. For example, if I rotate the knob pretty fast for several steps in one action the payload could be "006D0300". Steps here I mean kinda mechanical feedback during rotation. The rotation is not completely smooth, it has small tactile bumps. I hope it is clear what I mean.

tomky, 2022-06-01:
> Button push
>
> ```
> [INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 9
> ```
>
> Rotation clockwise(one [...]

> Button push
>
> ```
> [INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 9
> ```
>
> Rotation clockwise(one [...]

Sorry didn’t face time to test it this week. I wonder if you can also try “rotate while holding button”? It is also a valid button combination for this dimmer (to change the ambiance of the light).

Smanar, 2022-06-01:
It seem fine.

In the same folder you have found the DDF (the "devices" folder) you have a file called button_maps.json
It s the json file used for button, so just add for exemple

```
        "TuyaSmartKnob": {
            "vendor": "Tuya",
            "doc": "Smart Knob",
            "modelids": ["_TZ3000_4fjiwweb"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Clockwise"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Anticlockwise"]
            ]
        },
```

>I hope it is clear what I mean.

Yep but unfortunately, the basic ZHAswitch only return value like X00Y, need to create a special sensor like for the cube if you want to use the "force used" for the rotation.

And as the cluster 0x0008 is working, I think you can remove the the part for the 0x0006

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    },
```

But we will see at the end, can need it to prevent broadcast request.

vvbogdanov87, 2022-06-01:
Double click
```
[INFO] - No button map for: TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: ONOF (0x0006), command: LIDL (0xFD), payload: 01, zclSeq: 9
```
push(hold) and rotate clockwise(1 STEP)
```
[INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), command: STEP_COLOR_TEMPERATURE (0x4C), payload: 011120001009900F401, zclSeq: 47
```
push(hold) and rotate counterclockwise(1 STEP)
```
[INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: COLOR_CONTROL (0x0300), command: STEP_COLOR_TEMPERATURE (0x4C), payload: 03120001009900F401, zclSeq: 47
```
Again payload depends on the direction and how many "bumps" are done during one continuous rotation.

> ATM I m using the first byte so 0x00 or 0x01 to reconize the direction, other bytes are probably the force.
Looks like it is not a force(how fast I rotate), but a number of steps done in one rotation. If this number is not delivered, then it means for example that to deem lights by several levels you need to pause after each rotation. Because when I rotate the knob continuously it produces one event in the log instead of multiple.

Smanar, 2022-06-01:
With last features

```
        "TuyaSmartKnob": {
            "vendor": "Tuya",
            "doc": "Smart Knob",
            "modelids": ["_TZ3000_4fjiwweb"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Level Clockwise"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Level Anticlockwise"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Color Clockwise"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "3", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Color Anticlockwise"]
            ]
        },
```

But the problem is the same all third app wait for a X00Y return, I don't have idea to exploit the "step number"

vvbogdanov87, 2022-06-01:
I updated DDF file and added to the button_maps.json file. `Info` messages are gone in the log.
I have several zigbee bulbs and when I push the button or rotate the knob it adjusts all the bulbs simultaneously. I only added the devices in Phoscon App. No other configuration was done.
The second issue I still have is that I don't see events from the knob in openHAB. I have other buttons which are added to openHAB via deCONZ and I can confirm that I see events from them(click, double click etc.).

vvbogdanov87, 2022-06-01:
Here is my scenario:
I preferably want to control everything from openHAB. If I have to bound knob rotation to the light via Phoscon App it is OK. But I need to have toggle events from the knob only in openHAB. I don't want to have toggle event bound to the bulb directly and event in openHAB, I need the event only in openHAB.
So the question is how to bound knob rotation to only one bulb and how to send toggle events only to openHAB?

tomky, 2022-06-02:
> I updated DDF file and added to the button_maps.json file. `Info` messages are gone in the log.
> I have several zigbee bulbs and when I push the button or rotate the knob it adjusts all the bulbs [...]

If you add the dimmer (light) into the same group as the lights you want to control, the dimmer will only control those lights instead of all lights. Just a workaround and how I use the dimmer at this moment. But yes there is no event occurred outside of deCONz (such as homebridge-hue)

Smanar, 2022-06-03:
But you don't have the state/buttonevent updated with the button value ?
You can see it on third app, or direclty in the API on phoscon / help / API Information

This value need to be updated according to value used in the last json.

vvbogdanov87, 2022-06-03:
@Smanar  API section for the switch and for the dimmer:
```
"4": {
            "config": {
                "battery": 100,
                "on": true,
                "reachable": true
            },
            "etag": "aedc393ea04403d0d009ab8885ad7444",
            "lastannounced": "2022-06-02T05:05:41Z",
            "lastseen": "2022-06-03T22:35Z",
            "manufacturername": "_TZ3000_4fjiwweb",
            "mode": 1,
            "modelid": "TS004F",
            "name": "viktor_bed_switch",
            "state": {
                "buttonevent": null,
                "lastupdated": "none"
            },
            "type": "ZHASwitch",
            "uniqueid": "84:ba:20:ff:fe:8d:5c:1a-01-1000"
        },
```
```
"4": {
            "etag": "2015f04a12d7ca982d98340dd058b169",
            "hascolor": false,
            "lastannounced": "2022-06-02T05:05:41Z",
            "lastseen": "2022-06-03T22:35Z",
            "manufacturername": "_TZ3000_4fjiwweb",
            "modelid": "TS004F",
            "name": "viktor_bed_dimmer",
            "state": {
                "alert": "none",
                "on": false,
                "reachable": true
            },
            "swversion": null,
            "type": "Dimmer switch",
            "uniqueid": "84:ba:20:ff:fe:8d:5c:1a-01"
        },
```
The value is `null`, so we have a problem...What problem is that? Why is it null? :)

Smanar, 2022-06-04:
Are you sure using the good DDF ? I have removed the "Dimmer switch", mine create only "ZHASwitch"

vvbogdanov87, 2022-06-05:
I deleted the switch and reconnected it. I still see 2 devices in phoscon app. In deCONZ app I now see 3(!) devices each has DDF with one sub-device.
[image]

vvbogdanov87, 2022-06-05:
I deleted the devices and added it again. I still see 2 devices in Phoscon APP, but events are propagated properly under `lights` section
```
"state": {
                "buttonevent": 3002,
                "lastupdated": "2022-06-05T20:39:05.745"
            },
```

Smanar, 2022-06-06:
>under lights section

But "light" haven't state/buttonevent ? It s the "sensor" entry that have this field from your previous c/c. You are looking event in phoscon or openHAB ?

What is your deconz version ? The code part that prevent useless "light" entry creation is just some month old.
Else can just ignore the light entry for the moment, but openHAB need to see them for the sensor entry.

Smanar, 2022-06-06:
Ok so you have a recent version, there is definitively a problem.
Can you show again the 2 json with the new DDF like here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6031#issuecomment-1146434501

When you have deleted the device, you have do that using phoscon ?

vvbogdanov87, 2022-06-06:
```json
{
    "config": {
        "battery": 0,
        "group": "20000",
        "on": true,
        "reachable": true
    },
    "etag": "622453a41a2b0f100202c34f507e6ad5",
    "lastannounced": "2022-06-06T05:33:31Z",
    "lastseen": "2022-06-06T16:20Z",
    "manufacturername": "_TZ3000_4fjiwweb",
    "mode": 1,
    "modelid": "TS004F",
    "name": "viktor_bed_switch",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2022-06-05T22:12:19.010"
    },
    "type": "ZHASwitch",
    "uniqueid": "84:ba:20:ff:fe:8d:5c:1a-01-1000"
}
```
```json
{
    "etag": "622453a41a2b0f100202c34f507e6ad5",
    "hascolor": false,
    "lastannounced": "2022-06-06T05:33:31Z",
    "lastseen": "2022-06-06T16:20Z",
    "manufacturername": "_TZ3000_4fjiwweb",
    "modelid": "TS004F",
    "name": "viktor_bed_dimmer",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Dimmer switch",
    "uniqueid": "84:ba:20:ff:fe:8d:5c:1a-01"
}
```

Smanar, 2022-06-06:
Ok so both entries are new and used, recent lastannounced/lastseen
But why you still have the "light" entry ....

But we can see the "buttonevent" is now updated in the "sensor", so openHAB is able to see then too.

vvbogdanov87, 2022-06-12:
```You are sure not having 2 time a DDF for the same device ?```
@Smanar I double-checked it. I have only one DDF for the device.

tomky, 2022-07-06:
So, will this one be supported officially in next release?

jjansen85, 2022-07-08:
Don't know is this is useful...
I applied the DDF and button_maps.json as created by Smanar.

Present as light and sensor (switch)
```
{
    "etag": "3b92bc2e3ea38d1d696883e2eb62aed6",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-07-08T19:44Z",
    "manufacturername": "_TZ3000_4fjiwweb",
    "modelid": "TS004F",
    "name": "Dimmer switch 21",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Dimmer switch",
    "uniqueid": "38:5b:44:ff:fe:36:9e:eb-01"
}

{
    "config": {
        "battery": 100,
        "group": "20000",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "eaff92759484b1a2c19e7045ad7b0b97",
    "lastannounced": null,
    "lastseen": "2022-07-08T19:44Z",
    "manufacturername": "_TZ3000_4fjiwweb",
    "mode": 1,
    "modelid": "TS004F",
    "name": "TS004F 36",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2022-07-08T19:44:46.142"
    },
    "type": "ZHASwitch",
    "uniqueid": "38:5b:44:ff:fe:36:9e:eb-01-1000"
}
```

When using the knob, the log gives results like
```
21:41:51:366 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 000D0100, zclSeq: 108
21:41:54:214 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 109
21:41:55:759 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 000D0100, zclSeq: 110
21:41:57:120 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 112
21:41:58:410 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 000D0100, zclSeq: 114
21:41:59:812 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 115
21:42:01:107 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 116
21:42:02:054 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 117
21:42:02:142 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 116
21:42:02:143 [INFO] - Button 3002 Level Anticlockwise, discard too fast event (dt = 89) TS004F
21:42:02:901 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 118
21:42:22:111 [INFO] - Button 1002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Toggle, payload: None, zclSeq: 119
21:44:01:169 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 000D0100, zclSeq: 120
21:44:36:166 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 00190100, zclSeq: 121
21:44:37:710 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 000D0100, zclSeq: 122
21:44:41:148 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 123
21:44:42:172 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Anticlockwise, payload: 010D0100, zclSeq: 124
21:44:46:141 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Level Clockwise, payload: 000D0100, zclSeq: 126
```

Home Assistant also sees these events.
[image]

Smanar, 2022-07-09:
So you are able to retreive buttonevent trought the API, even with a bad detection (having both entry, sensor + light) ?

If it work like that, I can make the PR waiting for the code be updated to block the "light" ressource ?

jjansen85, 2022-07-09:
I think you are correct.

First I added without DDF.
Nothing happened, except it controlled all my lights and plugs. No data via API.
Fixed via instructions on: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/What-you-can-do-when-a-switch-turns-(almost)-all-devices-on-off

Then I added this DDF:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_4fjiwweb",
  "modelid": "TS004F",
  "product": "TS004F",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/group",
          "default": "auto"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0300"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
  ]
}
```

and added this to the `buttons_maps.json`
```
        "TuyaSmartKnob": {
            "vendor": "Tuya",
            "doc": "Smart Knob",
            "modelids": ["_TZ3000_4fjiwweb"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Level Clockwise"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Level Anticlockwise"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Color Clockwise"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "3", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Color Anticlockwise"]
            ]
        }, 
```

Smanar, 2022-07-11:
Arf, missed ^^.
The DDF is here to prevent user having to use the GUI for make unbind/bind.
This part is here for that

```
  "bindings": [
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0300"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
  ]
```

If the DDF is finished, you just need to add it.

manup, 2022-07-17:
I've just tested the DDF, looks like this is yet another Tuya f*ck the standards device ;)
The group bindings are added and also returned in the binding table request, but they are completely ignored.

By default the device sends to the group 0x0000, this can only be changed by using the *Group Cluster / Add Group* command.

**Note:** With the current deCONZ version there is wrongly a light resource added for the device and the 0xfff0 "group0" is applied, hence it did send to that group, controlling all lights. Locally I'm working on the fix so that DDF devices don't create bogus devices not in the DDF file.

So this brings the todo that we need an alternative way to apply our "auto" groups via Group Cluster instead of bindings, I need to think about this a bit more.. this isn't the only device which requires it :/ As workaround we could just leave group 0x0000 as configured one and ditch the `config/group` however I'd like to get the proper control.

Another side note, I also joined and sniffed the smart knob to the Lidl gateway, where it shows up properly. They don't do any bindings or group configuration so it keeps sending to group 0x0000. But a test automation to control a light didn't work there either :thinking:

Smanar, 2022-07-18:
@manup 

>As workaround we could just leave group 0x0000 as configured one and ditch the config/group however I'd like to get the proper control

Mean it can work using bind 
```
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0300"
    },
```
And with don't using 
```
        {
          "name": "config/group",
          "default": "auto"
        },
```

Because yes, have thoses issues on others devices too, and if I have understand, it can be worse if the code don't create the useless "light " entry ?

closed 2022-08-07 by manup