dresden-elektronik/deconz-rest-plugin #6032: MOES Smart Brightness Thermometer

TheRealElgardo, 2022-05-08:
## Device

- Product name: Smart Brightness Thermometer
- Manufacturer: _TYZB01 ftdkanlj
- Model identifier: TS0222
- Device type : Please remove all unrelated device types. 
   - Sensor
    - Other: Sensor for Light, Humidity, Temperature

## Screenshots

### Basic
[image]

### Identify
[image]
[image]
[image]
### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering
[image]
[image]
[image]

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-05-08:
Can you make sure the basic cluster is read?

Smanar, 2022-05-08:
In fact need to press the "read" button for all cluser ^^
The more important for me it s the power configuration Cluster, the device is probably sleeping and you need to wake it in same time.

Another question is why 2 cluster for illuminance measurement ? all are used ?

Else this device realy look classic, can you at least start a DDF to have a squeletton ?

TheRealElgardo, 2022-05-08:
The conbee is connected.
I pressed the button several times. There were no changes in either the power or the basic cluster compared to the hardcopy above.
Do you have a link to a description for executing a DDF?

closed 2022-06-08 by github-actions[bot]