dresden-elektronik/deconz-rest-plugin #6034: ts0201 tz3000_yd2e749y

emorigi, 2022-05-08:
## Device

- Product name: Tuya Temperature & humidity sensor
- Model identifier: ts0201
- Device type :  sensor

## Screenshots

[image]

### Basic

[image]

### Identify

### Alarms

### Device Temperature
[image]

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

Smanar, 2022-05-08:
Hello, can you try this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_yd2e749y",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

emorigi, 2022-05-09:
Great,
I re-added the device after deleting it on deconz and after putting the ddf file.
While the join procedure was not ending, I saw the device appear under the join-window and now I can also see it on the home assistant.
The only problem is that the battery stays at 0%.
Is there the possibility of making the reading of the battery charge also work?
Thank you

emorigi, 2022-05-10:
yes it works!
thank you!

closed 2022-05-16 by manup

Koal4, 2022-05-19:
Hi, having the same problem
But stupid question where is the ddf location on a hassio installation? Can't find anything on google about it :s

Smanar, 2022-05-19:
Good question if you find it, I m curious to knw it ^^
It s not always the same path, but https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/2

Koal4, 2022-05-19:
@Smanar As you already mentioned. I don't have that path at all in /usr/share. Not a deconz folder or a dresdenxxx folder

Koal4, 2022-05-19:
I managed to get in via docker bash

_docker container exec -it addon_core_deconz bash_

created the file locally on my machine and saved it on /home folder

Afterwards I copied the file from my local installation into the docker container

_docker cp ts0201.json addon_core_deconz:/home_

Afterwards I copied the file into the /usr/share/deConz/devices/tuya folder

And then I opened the DDF file via 'edit ddf'.  Go to the above location and read and saved and hot reloaded. I now have the device visible in phoscon.

Not sure if everything is working fine I will find that out in next couple of hours but so far looking good!

Smanar, 2022-05-19:
Sure, tell me if it work, I will copy your post on the phoscon forum.

Koal4, 2022-05-19:
Well I have it visible now in Phoscon and it seems it's providing data.

[image]

But somehow I don't find it in Home assistant.

Have to look another time

Smanar, 2022-05-20:
>But I can't seem to open the sensor like any other sensor

I m not sure to understand ...

If you can see it in phoson you can see in in HA (but not the reverse, phoscon is less powerfull than HA), perhaps a synchronisation issue.

Koal4, 2022-05-21:
I mean if you open sensors in phoscon you can see something like this.

[image]

It doesn't work for me with that sensor I added through that way

Smanar, 2022-05-21:
It can be phoscon magic, but HA use direclty the API.
You can see it in HA now ?
If no, are you sure it's the good one visible on phoscon ? you can take a look in phoscon / help / API Information / sensor

Koal4, 2022-05-21:
Yes I have it visible now in HA
But somehow it stopped working (giving temps). Had it updating in Phoscon for a while but it stopped. Not sure what has caused this. I will reset it later and try again when I have the time.