dresden-elektronik/deconz-rest-plugin #6037: Zemismart Tuya _TZ3000_vjhcenzo, cannot control switches individually

alahdal, 2022-05-09:
## Describe the bug
After pair , can't control each socket of device.
When I toggle the first socket, it's change all 3.

## Steps to reproduce the behavior
If the problem is reproducable, list the steps here:

After pairing, I used the HomeKit to switch on or off one switch but all three will be switched. If I switch manually, it works fine.

If the problem can't be reproduced and is sporadic, please provide some details
on how often and when the issue happens.

## Expected behavior
control the 3 switch individual

## Screenshots

## Environment
 - Host system: (Raspberry)
 - Running method: (Raspbian)
 - Firmware version: (26400500)
 - deCONZ version: (2.15.3)
 - Device: ( RaspBee I )

## deCONZ Logs

## Additional context
Similar problem is available on this link: https://github.com/Koenkk/zigbee2mqtt/issues/9564

Exact device with some advice is available in this comment: https://github.com/Koenkk/zigbee2mqtt/issues/9564#issuecomment-1051270586

Smanar, 2022-05-11:
Another device that need the tuya unlock sequence.

alahdal, 2022-05-12:
> Another device that need the tuya unlock sequence.

Thanks for your help. 
Is there any data required from my side?

Mimiix, 2022-05-16:
So this is a not supported device @Smanar ? I suggest @alahdal  opens an device request with appropriate screenshots so we can track the integration there.

Smanar, 2022-05-16:
No, and was not the only one, there is too this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4958
But IDK to do, not possible using only DDF.

closed 2022-06-14 by github-actions[bot]