dresden-elektronik/deconz-rest-plugin #6041: SML001 Lux=0 in API broken since v2.15.3

Markkuuss, 2022-05-10:
Since the update from v2.14.1 to v2.15.3, the lux value of the "1 Gen Hue motion sensor (SML001)" no longer works in complete darkness.

All lux values >0 work.

When it is completely dark, the web interface displays correctly with lux=0, but in the API the lux value remains at the previous value; only the lightlevel value is displayed correctly with 0.

[image]

ebaauw, 2022-05-11:
The sensor only reports `lightlevel`; `lux` is computed from that.  Likely a division by 0 in the DDF?

Markkuuss, 2022-05-11:
I think I have found the problem. `lux` is only set when `lightlevel` is greater than 0:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/46ccf39a17723bdc7db950e6df1e3a6280565cbb/devices/philips/sml_light_level.js#L7-L10

When it is completely dark, `lightlevel = 0`, but `lux` is no longer set.

closed 2022-05-16 by manup