dresden-elektronik/deconz-rest-plugin #6043: GLEDOPTO RGB+CTT Ledstrip Controller

MaxGramser, 2022-05-11:
## Describe the bug

When I'm making scenes in the Phoscon app, im able to fully control the RGB+CTT ledstrip, but when I save that scenes and then recall it, it only recalls the RGB values and doesn't do anything when a white color was in the scene.
## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.15.03
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: nope
 - Ledstrip controller: GLEDOPTO GL-C-008

## deCONZ Logs

## Additional context

Mimiix, 2022-05-11:
Known issue since 2021:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4444

@manup Can we fix this with DDF;s>?

manup, 2022-06-04:
Unfortunately currently not via DDF. This will be doable (in a proper way) when the whole scene machinery is refactored as roughly summarized in https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/5 

So as the workaround I'm looking forward to include https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4444 
I need to check a few things in the PR, aiming for v2.17.2.

closed 2022-07-04 by github-actions[bot]

Galapagon, 2022-10-23:
@manup Since https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4444 was merged in, does that mean that this could be revisited? Thank you for your hard work either way!