dresden-elektronik/deconz-rest-plugin #6044: Eglo Fueva-Z

cameo77, 2022-05-11:
## Device

- Product name: Fueva-Z
- Manufacturer: AwoX
- Model identifier: TLSR82xx
- Device type:
  - Light

## Screenshots

### Basic

[image]
[image]

### Identify

[image]

### Groups

[image]
[image]

### Scenes

[image]
[image]
[image]
[image]

### On/Off

[image]
[image]
[image]

### Level Control

[image]
[image]

### Color Control

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### ZLL commissioning

[image]
[image]

### Unknown

All Unknown Cluster look the same.

[image]

Mimiix, 2022-05-11:
Is there an issue with this light?

cameo77, 2022-05-11:
Sorry, I should have explain it. My Problem is way simpler than you think. I added this light through the Phoscon Web UI in the Home Assistant Add-On. If I want to control the color temperature, the value range of the slider is larger than the value range of the device. So the slider has a  "dead" zone at the bottom and at the top. As I found out just now, it can be solved by reading in the value range manually in the deconz UI. But is it possible that this works right away through the Phoscon UI? Or is reading in manually the normal way to go? (I'm pretty new to Phoscon/deconz)
Otherwise the device works very well.
Thanks

Smanar, 2022-05-11:
And you can't use HA to control the temperature ? (will be faster to have a correction on HA than phoscon)

cameo77, 2022-05-12:
I tried it again more carefully. With the Phoscon UI I can use the full range of the slider. However, no absolute values are displayed there. Only in Home Assistant UI, the slider has two dead zones. I can move the slider to the lowest position, but then it jumps to the value 250, see here:
[image]

If I slide it to the highest position, it jumps to the value 454. The lights there are acutally warm white and not neutral, as the slider shows:
[image]

Anyway this can be fixed if I read in the attributes manually via the deconz UI. That would be okay for me too, I know it now.

Mimiix, 2022-05-12:
@kane610 Would this be something on the device or on the implementation at HA? As the thing manually  goes back.

Kane610, 2022-05-12:
It depends on if the ct max and min values are reported properly in the rest API. Based on what is described here it sounds like perhaps the range is either wrong or not reported (just a guess though)

The Integration bases its range on what is reported if it exist (else it falls back to default values)

How pydeconz handles ct max/min https://github.com/Kane610/deconz/blob/abfceb1d8511f30854fa099cb38bd689abb3df43/pydeconz/models/light/light.py#L116

How the integration handles ct max/min https://github.com/home-assistant/core/blob/3ce19cd6f870b91f162c69d027a52f2f795f899d/homeassistant/components/deconz/light.py#L291

Smanar, 2022-05-12:
Yes can you check for the ct min/maxvalue you have in the API ? For exemple using phoscon / help / API Information / light

cameo77, 2022-05-12:
Okay, I deleted the light from Phoscon and did a reset of the light's hardware. After that I reconnected it through Phoscon. In deconz it looks like this:
[image]

In Phoscon (as you described):
`{
    "colorcapabilities": 16,
    "ctmax": 454,
    "ctmin": 250,
    "etag": "c1ec5ef56a9e9237193569d161827b12",
    "hascolor": true,
    "lastannounced": null,
    "lastseen": "2022-05-12T17:27Z",
    "manufacturername": "AwoX",
    "modelid": "TLSR82xx",
    "name": "Color temperature light 5",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "ct",
        "ct": 454,
        "on": true,
        "reachable": true
    },
    "swversion": "0122052017",
    "type": "Color temperature light",
    "uniqueid": "a4:c1:38:bb:f0:4a:08:52-01"
}`
And in Home Assistant I wasn't able to control it. This error message appeared:
[image]
The manufacturer and the model identifier were unknown in Home Assistant, too.
After a restart of Home Assistant I can control it now, and, however that happened, everything is correct (including the slider in Home Assistant). I don't know what's different this time. I've tried the resetting and reconnecting several times in the last days and never it behaved like that. But I remember, that the restart of Home Assistant wasn't necassary at that tries.
To be complete here is a actual screenshot of the values in deconz:
[image]

By the way, in the last days I had the same problem I described in the last post with another new light. But I think I will test this again soon.

Mimiix, 2022-05-12:
The device wasn't added fully probably.

Either way, seems like it's working fully and there's nothing else.

If you need help with anything, I'd like to ask you to check the forum.

Kind regards,

closed 2022-05-12 by Mimiix