dresden-elektronik/deconz-rest-plugin #6047: Device name

ppptmstr, 2022-05-11:
## Device

- Product name: ZigBee Multi-Sensor - Motion/Temp/Humi https://www.owon-smart.com/uploads/Datasheet-PIR323-ZigBee-Multi-Sensor.pdf
- Manufacturer: OWON
- Model identifier: PIR323-PTH
- Device type : 
  - Sensor (PIR + Temp. + Humidity)

## Screenshots

### Node Info
[image]

### Basic
[image]

### Temperature
[image]

### Humidity
[image]

### IAS

[image]

### Power configuration
[image]

The device seems quite similar to #5929 or #5777 ?

Smanar, 2022-05-11:
Hello can you try this DDF, you will probably mis the binding for the presence sensor

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR323-PTH",
  "product": "PIR323-PTH",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
            "name": "config/battery",
            "parse": {"cl": "0x0001", "at": "0x0021", "eval": "Item.val = Attr.val"},
            "awake": true
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/tampered"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x21",
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

ppptmstr, 2022-05-11:
OK now it shows up in Phoscon but the readings are weird :

[image]

and 

[image]

Smanar, 2022-05-11:
lol, ha yes, I m not lucky.

Can you check if values are updated if you read them in deconz ?
For exemple the temperature on the endpoint 0x02 cluster 0x0402, attribute 0x0000 ?

If yes try to replace
        {
          "name": "state/temperature",
          "awake": true
        }

by 

```
        {
          "name": "state/temperature",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x0000",
            "cl": "0x0201",
            "ep": 2,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl"
          }
        },
```

ppptmstr, 2022-05-12:
Great ! Now I get in Phoscon temp and humidity values that match the clusters..
[image]

Now about presence binding, is it the right way to do it ?
[image]

This gave me this additional binding :
   

> {

      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 0,
          "max": 0
        }
      ]
    }

And the full ddf preview:
>{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR323-PTH",
  "product": "PIR323-PTH",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/pir323-pth.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/presence",
          "awake": true
        },
        {
          "name": "state/tampered"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "eval": "Item.val = Attr.val / 2;"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 2,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x21",
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}

And I get the proper motion state in Phoscon.

BUT values won't update automatically, I have to press "read" in each cluster to get an update (and/or press the reset button on the device to wake it up).

ppptmstr, 2022-05-12:
>  it still not working after a time ? Or try a re-inclusion.

How long should I wait for ?
I made the change to remove the reporting, then tried a re-inclusion, and rebooted the ConBee II, still not updating.
While fiddling around, I noticed these strange log lines too :

> 17:02:01:373 [IAS ZONE] - Address 0x3C6A2CFFFED10E15, Payload 0000003001010000310d00020000193000100000f00000000000000000, Command 0x01
17:02:01:374 [IAS ZONE] - 0x3C6A2CFFFED10E15 No IAS sensor found for endpoint: 0x01
17:02:01:387 [IAS ZONE] - Address 0x3C6A2CFFFED10E15, Payload 1100002000, Command 0x01
17:02:01:387 [IAS ZONE] - 0x3C6A2CFFFED10E15 No IAS sensor found for endpoint: 0x01

Smanar, 2022-05-12:
```
"type": "$TYPE_PRESENCE_SENSOR",
"restapi": "/sensors",
"uuid": [
"$address.ext",
"0x01",
"0x0500"
],
```

All seem fine too, what is the uniqueid for this sensor ? Something ending by -01-500 ?

Ha I have perhaps found, the code check for fingerprint, try adding to DDF
```

 "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [

```

Nothing on logs about binding  for this device ?

ppptmstr, 2022-05-13:
> All seem fine too, what is the uniqueid for this sensor ? Something ending by -01-500 ?

Where do I find that value ?

So I made the change with the "fingerprint" part, and the logs for IAS zone changed :

> 09:31:40:647 [IAS ZONE] - Address 0x3C6A2CFFFED10E15, Payload 0000003001010000310d00020000193000100000f00000000000000000, Command 0x01
09:31:40:647 [IAS ZONE] - 0x3C6A2CFFFED10E15 Read attributes response:
09:31:40:648 [IAS ZONE] - 0x3C6A2CFFFED10E15   -> IAS Zone State: enrolled.
09:31:40:648 [IAS ZONE] - 0x3C6A2CFFFED10E15   -> IAS CIE address = 0x0000000000000000: NOT written.
09:31:40:649 [IAS ZONE] - 0x3C6A2CFFFED10E15 Sensor (ZHAPresence) is enrolled.
09:31:40:661 [IAS ZONE] - Address 0x3C6A2CFFFED10E15, Payload 1100002000, Command 0x01
09:31:40:661 [IAS ZONE] - 0x3C6A2CFFFED10E15 Read attributes response:
09:31:40:661 [IAS ZONE] - 0x3C6A2CFFFED10E15 Sensor (ZHAPresence) is enrolled.

> Nothing on logs about binding for this device ?

What should I be looking for ?

This time I remembered to copy the logs while adding the device, in case it may help:
 
> 09:40:41:842 APS-DATA.indication from unknown node 0x3C6A2CFFFED10E15
09:40:41:843 ZDP device announce: 0x3C6A2CFFFED10E15, 0x2068, 0x80
09:40:41:844 new node - ext: 0x3c6a2cfffed10e15, nwk: 0x2068
09:40:41:845 device announce 0x3C6A2CFFFED10E15 (0x2068) mac capabilities 0x80
09:40:41:845 set fast probe address to 0x3C6A2CFFFED10E15 (0x2068)
09:40:41:846 FP indication 0x0000 / 0x0013 (0x3C6A2CFFFED10E15 / 0x2068)
09:40:41:846                       ...     (0x3C6A2CFFFED10E15 / 0x2068)
09:40:41:846 device announce 0x3C6A2CFFFED10E15 (0x2068) mac capabilities 0x80
09:40:41:849 ZDP get node descriptor for 0x2068
09:40:41:849 ZDP get node descriptor for 0x2068
09:40:42:309 skip device discovery while end devices is added
09:40:42:717 [1] get node descriptor for 0x3c6a2cfffed10e15
09:40:42:718 ZDP get node descriptor for 0x2068
09:40:42:789 skip device discovery while end devices is added
09:40:42:917 ZDP status = 0x00 -> SUCCESS
09:40:42:918 ZDP Node_Descriptor_rsp 0x3C6A2CFFFED10E15 - 0x2068
09:40:42:939 DB UPDATE device_descriptors SET data = x'0240803c11525200002c520000', timestamp = 1652427642 WHERE device_id = (SELECT id FROM devices WHERE mac = '3c:6a:2c:ff:fe:d1:0e:15') AND endpoint = 0 AND type = 2
09:40:42:940 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'0240803c11525200002c520000', 1652427642 FROM devices WHERE mac = '3c:6a:2c:ff:fe:d1:0e:15'
09:40:42:953 FP indication 0x0000 / 0x8002 (0x3C6A2CFFFED10E15 / 0x2068)
09:40:42:953                       ...     (0x3C6A2CFFFED10E15 / 0x2068)
09:40:42:954 ZDP indication search sensors 0x3C6A2CFFFED10E15 (0x2068) cluster 0x8002
09:40:42:954 ZDP indication search sensors 0x3C6A2CFFFED10E15 (0x2068) clear timeout on cluster 0x8002
09:40:42:962 [2] get active endpoints for 0x3c6a2cfffed10e15
09:40:42:962 ZDP get active endpoints for 0x2068
09:40:42:970 ZDP status = 0x00 -> SUCCESS
09:40:42:971 ZDP Node_Descriptor_rsp 0x3C6A2CFFFED10E15 - 0x2068
09:40:42:971 FP indication 0x0000 / 0x8002 (0x3C6A2CFFFED10E15 / 0x2068)
09:40:42:972                       ...     (0x3C6A2CFFFED10E15 / 0x2068)
09:40:42:972 ZDP indication search sensors 0x3C6A2CFFFED10E15 (0x2068) cluster 0x8002
09:40:42:993 wait response fastEnddeviceProbe() 0x3C6A2CFFFED10E15, elapsed 30 ms
09:40:43:269 skip device discovery while end devices is added
09:40:43:649 54:ef:44:10:00:1c:7d:70-15-000c/state/power expression: Item.val = Math.round(Attr.val); --> 0
09:40:43:749 skip device discovery while end devices is added
09:40:43:882 ZDP status = 0x00 -> SUCCESS
09:40:43:883 ZDP active ep response for 0x3c6a2cfffed10e15
09:40:43:883 	ep: 0x01
09:40:43:883 	ep: 0x02
09:40:43:883 FP indication 0x0000 / 0x8005 (0x3C6A2CFFFED10E15 / 0x2068)
09:40:43:884                       ...     (0x3C6A2CFFFED10E15 / 0x2068)
09:40:43:884 ZDP indication search sensors 0x3C6A2CFFFED10E15 (0x2068) cluster 0x8005
09:40:43:884 ZDP indication search sensors 0x3C6A2CFFFED10E15 (0x2068) clear timeout on cluster 0x8005
09:40:43:891 [3] get simple descriptor 0x01 for 0x3c6a2cfffed10e15
09:40:43:891 ZDP get simple descriptor 0x01 for 0x2068
09:40:43:894 ZDP status = 0x00 -> SUCCESS
09:40:43:895 ZDP active ep response for 0x3c6a2cfffed10e15
09:40:43:895 	ep: 0x01
09:40:43:895 	ep: 0x02
09:40:43:896 FP indication 0x0000 / 0x8005 (0x3C6A2CFFFED10E15 / 0x2068)
09:40:43:896                       ...     (0x3C6A2CFFFED10E15 / 0x2068)
[... 101 lines cut ...]
09:40:49:728 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, has item: attr/type
09:40:49:729 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, has item: attr/uniqueid
09:40:49:729 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, create item: config/battery
09:40:49:729 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, create item: config/offset
09:40:49:730 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, has item: config/on
09:40:49:730 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, has item: config/reachable
09:40:49:730 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, create item: state/humidity
09:40:49:731 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0405, has item: state/lastupdated
09:40:49:731 DEV found DDF for 0x3C6A2CFFFED10E15, path: /opt/deCONZ/devices/pir323-pth-9.json
09:40:50:077 skip device discovery while end devices is added
09:40:50:557 skip device discovery while end devices is added
09:40:51:037 skip device discovery while end devices is added
[... cut ...]
09:40:53:441 skip device discovery while end devices is added
09:40:53:917 skip device discovery while end devices is added
09:40:54:397 skip device discovery while end devices is added
09:40:54:877 skip device discovery while end devices is added
09:40:55:357 skip device discovery while end devices is added
09:40:55:633 saved node state in 0 ms
09:40:55:633 sync() in 0 ms

Also, after adding the device, I still need to enroll it manually and read clusters to get it to show in Phoscon, and then it won't auto update.

Smanar, 2022-05-13:
Thoses device are sleeper, setting are not immediate.

But this trick is not usefull for temperature and humidity, I suspect a endpoint problem on the core because all sem fine

>09:40:49:698 sub-device: 3c:6a:2c:ff:fe:d1:0e:15-02-0402, has item: attr/id

```
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x00000014"
        }
      ]
    },
```

And the other Owon device is working with same setting https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/owon/THS317-ET_temperature_sensor.json

>What should I be looking for ?

Just enable loging with "info" info_l2" "error" and "error_l2" during the normal working mode (too hard to have them during the pairing) I hope something with the word "bind".
Or perhaps if you make "hot reload" in the DDF editor.

Stim04, 2022-05-13:
Hey,

I'm seeing these sensors are still tricky to add to the gateways. I personnaly suspect that the pairing is tricky as some sleepers are (eq. Aqara) and that you can end with an incomplete pairing even if your ddf is fine. The worst part is that when you pair it right it takes ages to report everything (in my case like 40mn). Then it works fine.

Smanar, 2022-05-14:
Hello, ha yes effectively I think there is a typo, it's not the same attribute used on 2 parts for battery

 "parse": {"cl": "0x0001", "**at": "0x0021"**, "eval": "Item.val = Attr.val"},

and

"report": [ {  **"at": "0x0020",**

For the status, as it's a new sensor and as we can't broke something, can change it to "Gold" if you want.
Sorry, it's more to not break network on user that already have this device but used with legacy code.

For the temperature I realy have no idea, I hope after a time deconz succed to set the binding ....

Stim04, 2022-05-14:
Thanks,
Some of them who have battery report problem already tried to tweak the DDF without success so I think we're guessing right with a pairing problem similar to some Aqara sleepers.
I have no clue why for me it works with the at '0x0020' in the bindings but i was thinking it must not be the problem because I succeded to pair it (more than once) on my gateway and that it s now working fine with me DDF without any issues.

I remember complaining that all the info wasn't available on my Jeedom more than once before I finally succeded.

So what do I tell them ? change that part of bindings or let it be and just reset module until it finally decide to work like I did ?
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0001",
      "report": [
        {
          **"at": "0x0020",   -> change to 0x0020 ?
          "dt": "0x21",    -> change to 0x20 or leave it ?**
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]

Smanar, 2022-05-14:
???
How you have 0x21 as DT ? Have juste made a try on my side to be sure, impossible to have it by defaut. You haven't edited the value ?
Perhaps an old deconz version ...
But I don't understand this part on the jeedom Forum (I m french too)

> On glisse tout le power config et le logiciel prend tout seul les valeurs qui sont dispo

On my side when I drag and drop a cluster, I haven't attributes pre-filled, I need to drag and drop the attribute after.

>Some of them who have battery report problem already tried to tweak the DDF without success so I think we're guessing right with a pairing problem similar to some Aqara sleepers.

I think too, but one of them use the lastest version and from the changelog it improve pairing https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5983

Can you say them to use the aqara tips ? If it's possible, I don't know this device, press shortly the reset button to prevend the device going to sleep.

And for the DDF it's as you prefer, 0x0020 or 0x0021 (we generaly use the 0x0021), but I m sure the "dt" is bad, IDK what happen on it.

Stim04, 2022-05-14:
what dt format should be ? I'm confused.

`On my side when I drag and drop a cluster, I haven't attributes pre-filled, I need to drag and drop the attribute after.`
I followed the tutorial for DDF creation and dragged and dropped temperature cluster and power config cluster (drag from left to right). I didn't have to fil the right part of the windows except for the refresh times that we discussed together

Ah cool you're french so you can see that I already adviced them the aqara trick and also warned them that pairing was tidious.
They incriminate my work (with your help) but don't question the (chinese) hardware quality when it comes to respecting zigbee standards

Smanar, 2022-05-14:
Ok, so you had a different version, or found a new issue (with cached DDF ???)
Because I m doing the same on the last one, and when I drag and drop the power config cluster, the right panel is still empty, I need to drag and drop the attribute too, and on the editor it's "u8", and when I go on DDF preview, the value is 0x20

And I m sure at 90% it s the correct behaviour, I realy have no clue what happen during your tests.

And it s just a battery reporting problem ^^, no so terrific, I'm more worried by the temperature reporting, and I have another owon device without reporting at all, and I don't see anything bad in temperature DDF part (I m using the same than you, just changed the endpoint)

Stim04, 2022-05-14:
That s what I told them.. who care of battery % it's not like it was very accurate anyway..

Smanar, 2022-05-14:
Your device is working, you can't do more, need a not working one.

Say them to make tries with 0x0021 or 0x0020, but with "dt" = 0x20. To test what is the better working mode for their device.
On jeedom they don't have GUI if I remember, so they need to re-start the deamon or re-include the device to go faster.

And BTW yes, don't forget to set the status to "Gold", I m waiting for this story is finished to see if we need to edit more, but I will switch the status to "Gold", I think there is enought testers to be sure "bronze" is useless ^^

Stim04, 2022-05-14:
Ok thanks. I'm going to translate them

Smanar, 2022-05-14:
Only this one

    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]
    },

I will check for temperature, but this one seem fine.

Stim04, 2022-05-14:
Yea me too.. but I still remember complaining to you that it stayed at zero... until it decided to work

Smanar, 2022-05-14:
Loool, the user had same reaction for the other owon device ^^
Ha have made an issue to say it's not working and another 9h after to say it s working (but only for dectection yet)

Stim04, 2022-05-14:
I don't see any problem either on the ddf on this post concerning temperature cluster

Smanar, 2022-05-14:
This one have some stuff usefull yes https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.16.0-beta

- Fix processing rules referencing DDF devices
- Improve loading DDF sensor data from legacy DB table
- Improve detection NWK address changes and ZDP discovery

But not sure it will help for this device ...

ppptmstr, 2022-05-14:
> Hello, ha yes effectively I think there is a typo, it's not the same attribute used on 2 parts for battery
> 
> "parse": {"cl": "0x0001", "**at": "0x0021"**, "eval": "Item.val = Attr.val"},
> 
> and
> 
> "report": [ { **"at": "0x0020",**
> 

You're right ! In this cluster, attribute 20 is voltage and 21 is battery percentage, so I think it should be 21 for both...
And here is the ddf file with the various changes you recommended.
[pir323-pth.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8693509/pir323-pth.json.txt)

When hot reloading the ddf just now, I got one T°/H% update, but nothing since my last post.
@Stim04 , I'm going to try your pairing+spam method and then leave the device quiet for a while to see if it's an incomplete pairing/binding...

Stim04, 2022-05-15:
Hello,

Apparently I have someone in Jeedim that used the spam pairing technique and managed to get battery % reported (his temp cluster was already working). I'm asking him what exactly he done.

ppptmstr, 2022-05-16:
> This one have some stuff usefull yes https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.16.0-beta
>
> * Fix processing rules referencing DDF devices
> * Improve loading DDF sensor [...]

I tried the latest beta but no luck.
@Stim04 As for the spam pairing thingy, any caveat I should be aware of ? I tried twice to pair while hitting the button until the device appeared in phoscon and continued until the end of the discovery: I  only get presence, still no Batt/T°/H%.

One funny thing : I do get the refresh time to update in Phoson once in a while (without anything moving around the sensor) but the values won't change
[image]

Stim04, 2022-05-16:
> @Stim04 As for the spam pairing thingy, any caveat I should be aware of ? I tried twice to pair while hitting the button until the device appeared in phoscon and continued until the end of the discovery: I only get presence, still no Batt/T°/H%.

Hey,
Only 2 inclusion for this device is not what I call spam pairing. The guy that got his battery % report yesterday really spammed inclusion more than that I think. At least that's what I did in february to get this mofo on the network with all the info.
You can reset the device and pair it again without deleting it. Gateway will handle it no problem and replace the old one. 

The best is doing that while looking at Phoscon interface opened so you can see what is reported or not.
If after a dozen of tries if you don't get anything you can try to wait an hour to see if it decides to work.

ppptmstr, 2022-05-18:
> Can you try this DDF ?

>If after a dozen of tries if you don't get anything you can try to wait an hour to see if it decides to work.

OK I tried both your suggestions, and the best I can get is move and battery.

> But you had temperature previosuly ?

I think I got tricked.
- in deconz select the temp cluster and hit "read" -> Phoscon updates
- re-pair the device (without deleting in Phoscon)
- wait a while and the old value will be "restored" in Phoscon. But that is not because reporting works, just Phoscon kept it cached somehow
So yeah, gateway can handle re-pairing without deleting the sensor, but no me :)

I spent enough time on this, I'll make up do with no temp/H%, that's good enough for me.

closed 2022-06-16 by github-actions[bot]