dresden-elektronik/deconz-rest-plugin #6051: _TZE200_9cxuhakf Dimmer Switch

Dazwah, 2022-05-13:
### **Device**
Product name: Dimmer Switch Mechanism https://www.ikuu.com.au/product/switch-mechanism-3/
Manufacturer: Mercator
Model identifier: SSWM-DIMZ
Device type :
Dimmer Switch
### **Screenshots**
**Node Info**
[image]
[image]
[image]
[image]

**Phoscon**
 {
    "etag": "736925467fbd4d9965f9b9bd47707b4e",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-05-13T12:18Z",
    "manufacturername": "_TZE200_9cxuhakf",
    "modelid": "TS0601",
    "name": "Smart plug 24",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "cc:86:ec:ff:fe:9a:10:78-01"
}

Smanar, 2022-05-17:
Can make a try to check if it use classic tuya request
Can you try this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_9cxuhakf",
  "modelid": "TS0601",
  "vendor": "Mercator",
  "product": "Mercator SSWM-DIMZ",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"},
          "refresh.interval": 84000
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

closed 2022-06-15 by github-actions[bot]

DubbyMcDubs, 2022-09-13:
I tried this today and it appears to be working correctly. Thanks!

Smanar, 2022-09-13:
Thx, haved added it here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

DubbyMcDubs, 2022-09-13:
One thing that is probably the only thing not correct though, is that in the Phoscon App it appears as product type "Fairy lights".

Smanar, 2022-09-14:
Yeah this problem is from phoscon ^^, don't worry you will not have it on other third app.