dresden-elektronik/deconz-rest-plugin #6053: Namron Thermostat Touch Zigbee 16A

norgezigbee, 2022-05-14:
Hi,

I have problems getting my new Namron Thermostat appear in Phoscon. I saw somewhere else that it has been added already but after numerous tries with reading descriptors while connecting it is still not available also not in Home Assistant. In contrast if I use ZHA instead of phoscon it appears in Home Assistant.
I would like to stay with Phoscon. t would be great if you could help me there

## Device

- Product name: Namron Thermostat Touch Zigbee 16A
- Manufacturer: Namron AS
- Model identifier: 4512737
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots
[image]

[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-05-14:
Have mimic this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5848

Can you try this DDF (have removed sinope specific stuff)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMROM AS",
  "modelid": "4512737",
  "product": "Thermostat Touch Zigbee 16A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
[... 250 lines cut ...]
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

norgezigbee, 2022-05-14:
Hi,

thanks for checking on a Saturday!! 
Adding the ddf didnt make any difference unfortunately.
I'm not sure if I do it right though. I save the ddf as Json. Then click on edit ddf and open it. Is that correct?

Smanar, 2022-05-14:
You just need to put it in the good folder and restart deconz (can retry an inclusion) 

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

But yes, you can make "edit DDF" then file / open and select this DDF, then save and make "hot reload" on the menu, the node name need to change from 0x4223 to a real name.

Deconz check  your device using 
```

  "manufacturername": "NAMROM AS",
  "modelid": "4512737",
```

norgezigbee, 2022-05-14:
Strange I tried now the method with the home folder and restart. It still shows this ddf
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON AS",
  "modelid": "4512737",
  "product": "4512737",
  "sleeper": false,
  "status": "Draft",
  "subdevices": []
}
If I do the manual file open alternative. It populates the ddf editor. Then I press Hot reload and close. But if I go afterwards on edit ddf the ddf above is again shown

norgezigbee, 2022-05-14:
Ah I found the problem in the ddf. The manufacturer name was wrong its Namron not Namrom :)
Now it appears in phoscon.

Thanks a lot!!

norgezigbee, 2022-05-14:
Now it reads the set heating temperature but the room temperature and voltage appear as 0. 
I also cannot yet control it from HA. Setting heating temperature gives an attribute error and turning on doesnt work

Smanar, 2022-05-14:
Have corrected the manufacture name, and have added some bind (from cluster thermostat, so you will have the room temperature now)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON AS",
  "modelid": "4512737",
  "product": "Thermostat Touch Zigbee 16A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
[... 275 lines cut ...]
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Can you show the error message ?
For information if the device was added in HA, before the device was correcly added in deconz, HA miss some information and add it partially.
So try to synchronise it, or just delete it in HA for it will be full reconised.

norgezigbee, 2022-05-14:
Phoscon now shows some voltage reading (0.225V unit error?). The room temperature is still 0 both in Phoscon and HA. HA shows power and consumption.
I can now control the temperature from HA but I cannot turn it off or on

Smanar, 2022-05-14:
>Phoscon now shows some voltage reading (0.225V unit error?)

Yep, edit your DDF like that 

```
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val * 10 ; }"
          },
          "default": 0
        }
```

>I cannot turn it off or on

You are using "preset' or "mode" ?
If I m right if "mode = "null" HA consider the "mode" is not working, so you can use manualy the device to have a first value.
 (Can set a defaut value in the DDF too)

>The room temperature is still 0

Oups my bad, the sensor I have mimic have a temperature measurement cluster, not your

Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON AS",
  "modelid": "4512737",
  "product": "Thermostat Touch Zigbee 16A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
[... 236 lines cut ...]
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Have correcty too the voltage, unfortunately, you need to delete the previous sensor (to remove the useless temperature sensor), and remake the inclusion.

norgezigbee, 2022-05-15:
Yes that's it!! I deleted everything from deconz and ha and readded the thermostat with the new ddf.
It's working perfectly now. With the voltage I guess this is the internal circuit voltage and not the heating cable voltage (230V) as I expected. Then time 10 makes sense. Learned something great.

Thanks a lot!

Smanar, 2022-05-15:
So all is fine ? I can make a PR for this device ?

norgezigbee, 2022-05-15:
Yes I think so. I didn't find any problems. It's ready for the PR. When will it be in the regular beta or steady version?

Smanar, 2022-05-15:
I think the next one (a beta one) you can check PR validation here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6046

closed 2022-05-16 by manup

jreidel, 2022-05-18:
Sorry to reply in the closed issue. I have managed to get things working, using the lastest DDF. But room temp is still 0 (HA and Phoscon), and status is allways "Off" (even if the thermostat is on).
[image]

and
[image]

Any idea?

Smanar, 2022-05-18:
It' exactlty issues @norgezigbee have but, before the last DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6053#issuecomment-1126769293

You are sure, you are using the good DDF ?
You can check in phoscon / help / API Information / sensors
If you have a ZHAtemperature sensors, you are using the bad one

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/namron/4512737_thermostat.json

```
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val * 10 ; }"
          },
          "default": 0
        }
```

jreidel, 2022-05-18:
Strange, I'm using the latest DDF. Don't see any ZHAtemperature

This is from the API info:
[image]

`{
    "config": {
        "heatsetpoint": 2350,
        "mode": "off",
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "91c8a9a8b17c72f06aa279b4ea91412d",
    "lastannounced": null,
    "lastseen": "2022-05-18T20:19Z",
    "manufacturername": "NAMRON AS",
    "modelid": "4512737",
    "name": "Thermostat 51",
    "state": {
        "lastupdated": "none",
        "on": null,
        "temperature": 0
    },
    "swversion": "6.9.1.0_r4",
    "type": "ZHAThermostat",
    "uniqueid": "1f:ff:00:01:00:00:03:cb-01-0201"
}`

Smanar, 2022-05-18:
Ok have found the issue.
There is a cluster issue, can you try this DDF ?
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON AS",
  "modelid": "4512737",
  "product": "Thermostat Touch Zigbee 16A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device.",
         "default": "off"
        },
        {
[... 207 lines cut ...]
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

It will probably solve the tension and the temperature issue.
But I have removed the reporting for the "mode" it seem the code don't use it, I will re add it if needed.

```
        {
          "at": "0x0012",
          "dt": "0x29",
          "min": 1,
          "max": 3600,
          "change": "0x00000001"
        }
```

BabaIsYou, 2022-11-28:
I think you can perhaps also control "Display Auto Off" parameter using this part in DDF (to be included in subdevice $TYPE_THERMOSTAT part as a new item 
`       {
          "name": "config/ledindication",
          "description": "Use it for Display Auto Off configuration.\nfalse = deactivated\ntrue = acivated",
          "refresh.interval": 360,
          "read": {
            "at": "0x1001",
            "cl": "0x0201",
            "ep": 1,
            "mf": "0x1224",
            "fn": "zcl"
          },
          "write": {
            "at": "0x1001",
            "cl": "0x0201",
            "dt": "0x30",
            "ep": 1,
            "eval": "Item.val",
            "mf": "0x1224",
            "fn": "zcl"
          },
          "parse": {
            "at": "0x1001",
            "cl": "0x0201",
            "ep": 1,
            "eval": "Item.val = Attr.val;",
            "mf": "0x1224",
            "fn": "zcl"
          },
          "default": false
        },`