dresden-elektronik/deconz-rest-plugin #6058: Device name

adx-git, 2022-05-16:
## Device

- Product name: ZigBee Bewässerungsventil (ZX-7155-675)
- Manufacturer: _TZE200_2wg5qrjy
- Model identifier: TS0601
- Device type : 
  - Switch
  - Other: sprinkler valve

## Screenshots

[image]

[image]

### Basic
[image]
[image]

The other clusters, even after reading, do not contain any data and just zero values.

Mimiix, 2022-05-16:
Hi, Please add all screens, even if they have zero's or other data.

Smanar, 2022-05-17:
Hello can you try this DDF, If I m right it will enable the on/off

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_9cxuhakf",
  "modelid": "TS0601",
  "vendor": "Mercator",
  "product": "Mercator SSWM-DIMZ",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

I m using the command 0x01 0x01, If I m wrong, can you check log (help/debug view with "info" and "info_l2), use the device manualy to check in log the good request ?

adx-git, 2022-05-17:
Hi @Smanar, thanks a lot, it is working, I can open  and close the the valve.

Though, I am not sure about the manufacturername, my readings are saying that it should be "_TZE200_2wg5qrjy", from the json it is "_TZE200_9cxuhakf", which one is the correct one, or does it matter at all?

Smanar, 2022-05-17:
It realy matter, it s like that deconz reconise the device ....
The _TZE200_9cxuhakf is a dimmer, have make mistake with DDF ^^. I have updated the DDF, but you are sure the value in deconz is _TZE200_2wg5qrjy ?
It mean if we use a bad manufacture name, deconz still can choose a bad DDF and change the manufacture name in the device json ?

adx-git, 2022-05-18:
> It realy matter, it s like that deconz reconise the device .... The _TZE200_9cxuhakf is a dimmer, have make mistake with DDF ^^. I have updated the DDF, but you are sure the value in deconz is _TZE200_2wg5qrjy ?

Yea, I am sure about "_TZE200_2wg5qrjy", at least it is working after I used this as manufacturer name 

 > It mean if we use a bad manufacture name, deconz still can choose a bad DDF and change the manufacture name in the device json ?
I changed the DDF content before loading it into deconz, I haven't tried to change the manufacturer name after loading the file.

Smanar, 2022-05-18:
>I changed the DDF content before loading it into deconz, I haven't tried to change the manufacturer name after loading the file.

Ha ok, so it s fine, was my fault, I think if you hadn't changed the DDF it will have not worked.

closed 2022-06-16 by github-actions[bot]

adx-git, 2022-06-16:
In which version was the supported added?

Smanar, 2022-06-16:
The DDF is not in code, no one have made the PR
So you just need a version that support DDF and add the previous DDF

ANierbeck, 2022-07-02:
Hope that pull-request helps :)

vwegert, 2024-08-29:
The DDF that is currently shipped (https://deconz-community.github.io/ddf-tools/#/store/bundle/01604e0f83e273149071c0b5c771c0a8f70fd5328e56d954de1d5034b8733448) does not appear to expose the battery state information - which really would come in handy for a device like this. Would it be possible to add the "usual" battery_level and battery_low channels?

Smanar, 2024-08-29:
Hello, if you can try this DDF 

```
{
  "schema": "devcap1.schema.json",
  "uuid": "1d0e1b23-c414-4ad4-bb8a-1736cb9f977e",
  "manufacturername": ["_TZE200_2wg5qrjy", "_TZE200_akjefhj5", "_TZE200_81isopgh"]
  "modelid": ["TS0601", "TS0601", "TS0601"]
  "vendor": "Royal Gardineer",
  "product": "Water valve (ZX-7155-675)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Have (try to) added the battery level, 2 clones and removed a 300s forced poll, so it can have break something.

To add again the poll

```
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
```