dresden-elektronik/deconz-rest-plugin #6059: Multiple sequential rules triggered by 1 button press (state change)

ldvc86, 2022-05-16:
## Describe the bug
Using the Hue Essentials app, I've configured my Hue Dimmer Switch to activate a scene, dependent on the number of presses on the 'ON' button. 1 press activates scene A, 2 presses activates scene B. For this, 2 rules are present on the conbee; one for each press.
This used to work fine, but starting 2.15.x, both rules are activated one after another, by only one press. Result is that you see the lights transition to scene A and immediatly afterwards transition to scene B.

The logging (if i'm reading this correctly...) seems to confirm that both rules are triggered by the single press (see logging section below)

Relevant rules on Conbee II:
(sensor 20 = dimmer switch, sensor 48 = ClipGenericStatus to keep track of button presses)

 
`  "91": {
        "actions": [
            {
                "address": "/groups/15/scenes/12/recall",
                "body": {},
                "method": "PUT"
            },
            {
                "address": "/sensors/48/state",
                "body": {
                    "status": 1
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/20/state/buttonevent",
                "operator": "eq",
                "value": "1000"
            },
            {
                "address": "/sensors/20/state/lastupdated",
                "operator": "dx"
            },
            {
                "address": "/sensors/48/state/status",
                "operator": "eq",
                "value": "0"
            }
        ],
        "name": "Dimmer switch 20 on-press",
        "periodic": 0,
        "status": "enabled"
    },
    "92": {
        "actions": [
            {
                "address": "/sensors/48/state",
                "body": {
                    "status": 2
                },
                "method": "PUT"
            },
            {
                "address": "/groups/15/scenes/9/recall",
                "body": {
                    "scene": "9"
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/48/state/status",
                "operator": "eq",
                "value": "1"
            },
            {
                "address": "/sensors/20/state/buttonevent",
                "operator": "eq",
                "value": "1000"
            },
            {
                "address": "/sensors/20/state/lastupdated",
                "operator": "dx"
            }
        ],
        "name": "Dimmer switch 20 on-press",
        "periodic": 0,
        "status": "enabled"
    },
`

## Steps to reproduce the behavior
1. Configure Hue Dimmer Switch using rules above
2. press top button of dimmer switch once

## Expected behavior
It seems that the lastupdated property is regarded as changed (dx) for too long, allowing for multiple sequential rules to fire. This should not be the case.

## Screenshots

## Environment

 - Host system: QNAP TS-251D (NAS)
 - Running method: Community docker
 - Firmware version: 26720700
 - deCONZ version: 2.16.00
 - Device: Conbee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

21:13:09:057 expression: Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(4) --> 1000
21:13:09:557 expression: Item.val = ZclFrame.at(6) | ZclFrame.at(7) << 8 --> 0
21:13:09:565 rule event /sensors/20/state/lastupdated: 0 -> 0
21:13:09:565 trigger rule 91 - Dimmer switch 20 on-press
21:13:09:566 rule event /sensors/20/state/lastupdated: 0 -> 0
21:13:09:566 trigger rule 92 - Dimmer switch 20 on-press
21:13:09:567 delayed group sending
21:13:09:570 delayed group sending
21:13:10:016 expression: Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(4) --> 1002
21:13:10:476 expression: Item.val = ZclFrame.at(6) | ZclFrame.at(7) << 8 --> 0
21:13:10:486 rule event /sensors/20/state/lastupdated: 0 -> 0
21:13:10:487 rule event /sensors/20/state/lastupdated: 0 -> 0
21:13:11:020 Skip idle timer callback, too early: elapsed 531 msec

## Additional context
Using hue dimmer configuration created with Hue Essentials app

Thomas-Vos, 2022-05-16:
I am not sure what is going wrong here. The rules you shared look correct. It should not be possible for both to trigger at the same time, so I am not sure why that happens.

What you could try is resetting the smart control configuration in Hue Essentials (under the wizard button). Then set up the multiple presses again and see if the issue persists.

Thomas-Vos, 2022-05-16:
Can you please share all rules that contain `/sensors/20` or `/sensors/48` in the conditions/actions?

manup, 2022-05-16:
I've done a quick test and can reproduce this even with a simple rule. Looks like we emit the lastupdated event twice somewhere in the code path. I'll check tomorrow in the debugger to provide a fix as PR.

popy2k14, 2022-05-17:
Just for cross reference, here is an thread in the community forum on this issue: https://forum.phoscon.de/t/hue-dimmer-switch-scene-toggling-not-working-correctly-since-v2-15-3/1985

closed 2022-05-17 by manup