dresden-elektronik/deconz-rest-plugin #6062: Tuya Temp/Hum sensor TS0201

Ltek, 2022-05-17:
This looks like a common Tuya sensor under multiple different 'brands' and different shells/cases. Anyone had a DDF for it? thx

Screenshots of what I have is included.

- Product name: Tuya Temperature and Humidity Sensor
- Manufacturer: _TZ300_yd2e749y
- Model identifier: TS0201
- Device type: Temperature and Humidity Sensor

screenshots... https://imgur.com/a/Ik9gCUd

Mimiix, 2022-05-17:
Looks like this is a duplicate:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6034

Please make sure to search git first prior opening an request.

Closing.,

closed 2022-05-17 by Mimiix

Ltek, 2022-05-18:
@Mimiix I tried the DDF there and its no working. the Device doesnt show in Phoscon, and consequently in Homeseer. Looks like deCONZ is not "publishing" the device. (or whatever the correct term is) 

I'm having the same issue with the other Tuya Temp/Hum sensor here...
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6063#issuecomment-1128973903

I created JSON files using Windows Notepad and put them in the folder.. C:\Program Files (x86)\deCONZ\devices\tuya

any ideas why this is not working?