dresden-elektronik/deconz-rest-plugin #6063: Tuya Temp/Humi Sensor with Screen TS0601

Ltek, 2022-05-17:
This looks like a common Tuya sensor, with screen, under multiple different 'brands' and different shells/cases.  Anyone had a DDF for it? thx

Screenshots of what I have is included.

- Product name: Tuya Temperature and Humidity Sensor with Screen
- Manufacturer: _TZE200_qoy0ekbd
- Model identifier: TS0601
- Device type: Temperature and Humidity Sensor

screenshots... https://imgur.com/a/0yPt0HI

Smanar, 2022-05-17:
BTW this device have Humidity/temperature cluster AND the tuya cluster.

From that I m reading on Z2m only the tuya cluster is working, but can't be sure.
So it s probably the same working mode than this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5927

Can try this DDF 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_qoy0ekbd",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "description": "The current relative humidity in percent.",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

Ltek, 2022-05-18:
@Smanar thx for the help. I created JSON the using Windows Notepad and put in the folder… C:\Program Files (x86)\deCONZ\devices\tuya  

But while its showing in deCONZ UI, its not showing in Phoscon?

Mimiix, 2022-05-18:
@Smanar we require screenshots for a reason. Please only proceed when they are correctly attached.

Smanar, 2022-05-18:
How look the node under your capture, the name is something like "0xXXXX" or something like "temperature 10" ?
When you make "edit DDF" on the editor title you can see the file path of the DDF ?
Nothing more if you select "hot reload" in the menu ?

Can you make a "Hot reload" but with enabling debug flag " info" and "info"l2" (logs are in help / debug view)

Ltek, 2022-05-18:
> How look the node on top of your capture, the name is something like "0xXXXX" or something like "temperature 10" ? When you make "edit DDF" on the editor title you can see the file path of the DDF ? Nothing more if you select "hot reload" in the menu ?
> 
> Can you make a "Hot reload" but with enabling debug flag " info" and "info_l2" (logs are in help / debug view)

Attached Debug with 'hot reload'

all 3 devices I paired - multiple times (one TS0201 & two TS0601) node are named "0xXXXX"  ... of course, the prior 20+ devices which worked fine have real word names as you pointed out. screenshot attached (I removed/paired them again last night)

[image]
[Debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8718827/Debug.txt)

Smanar, 2022-05-18:
>(one TS0201 & two TS0601) 

The code can work only on this one 

```
  "manufacturername": "_TZE200_qoy0ekbd",
  "modelid": "TS0601",
```

>11:32:18:038 DEV found DDF for 0xA4C13889735D4D20, path: C:/Program Files (x86)/deCONZ/devices/tuya/_TZ3000_yd2e749y.json

This device is not the same one (if the device is realy a _TZ3000_yd2e749y), but I don't see why the DDF is not used for this one, perhaps with flags "error" and "error _l2"

Ltek, 2022-05-18:
Please see the screenshot, you can see all the DDFs (attached) in the Tuya folder.  Are both devices (TS0201 and TS0601) using the wrong DDF the one that comes in the install, not the ones associated with proper manufacturer name?

[image]

[_TZE200_qoy0ekbd.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8721435/_TZE200_qoy0ekbd.json.txt)
[_TZ3000_yd2e749y.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8721436/_TZ3000_yd2e749y.json.txt)
?

Ltek, 2022-05-19:
> You have phoscon in permit join when you make your tests? "add new sensor" ? As the device is not included at all, have you try to make a new inclusion now the DDF is loaded ?

I use Phoscon to add the sensor -- and it adds to deCONZ, thats the only way I know how to add it. 
I dont understand you comment..."device is not included at all". How is it not 'included' if it shows up in deCONZ?

I think we are seeing a bug... 

I have 3 devices (2 different models) currently paired:  one TZE200_qoy0ekbd and two TZ3000_yd2e749y

both correct DDFs are in the Tuya folder

BUT... ALL 3 devices ONLY use the _TZ3000_yd2e749y.json --- two of the devices should be using _TZE200_qoy0ekbd.json, but they do not

If I remove the _TZ3000_yd2e749y.json from the folder (_TZE200_qoy0ekbd.json still in the folder)... NONE of the devices pull in _TZE200_qoy0ekbd.json even though 2 of the currently paired devices match that manufacture.

Smanar, 2022-05-19:
Perhaps a typo in the DDF, I will check on others zigbee project, easier with copy paste than looking in a picture ^^
But I don't understand why all 3 are using the same DDF 

You can see 
` DEV found DDF for device_mac_adress, path: C:/Program Files (x86)/deCONZ/devices/tuya/_TZ3000_yd2e749y.json`

For all 3 devices ?

>I dont understand you comment..."device is not included at all". How is it not 'included' if it shows up in deCONZ?

Deconz include all zigbee devices, it s a zigbee application. But inluded in deconz, don't mean included in the API.
When the device have the name 0xXXXX, it s the zigbee network adress, mean the device is included in the zigbee network. When the API reconise it it use a "API friendly name" like temperature 7.

Ltek, 2022-05-19:
I removed the devices and re-paired one of each device model... They each seem to be grabbing the correct DDF now. BUT they are still showing the 0xXXXX name style.

Any way to turn on debug during deCONZ load? Would be nice since there is not clear button on Debug window (really need one). Its a lot more text to copy since it even persists across close/open of the window.

Smanar, 2022-05-19:
>They each seem to be grabbing the correct DDF now. BUT they are still showing the 0xXXXX name style

Mean they are not reconised by deconz, even with DDF ...

You can send logs to a file too, but need to run deconz with command line.
I just need logs during the pairing so
- start log with "DDF' (few logs) and "info" (realy talkative) and "info_l2"
- in phoscon select add new sensor
- make the reset procedure on the device
- wait 1 mn
- stop logs.

You can too just take a look yourself and copy paster only the part you found usefull.

You can try just with only "error" and "error_l2", just to take a look, no need to copy/paste.

Ltek, 2022-05-19:
I removed both sensors and re-paired them. The DDFs were in the Tuya folder at the same time. Debug log attached
[DEBUG.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8734431/DEBUG.txt)

Smanar, 2022-05-20:
Both are reconised this time
```

15:47:35:395 DEV found DDF for 0xA4C1380395605A60, path: C:/Program Files (x86)/deCONZ/devices/tuya/_TZE200_qoy0ekbd.json
10:15:33:056 DEV found DDF for 0xA4C13889735D4D20, path: C:/Program Files (x86)/deCONZ/devices/tuya/_TZ3000_yd2e749y.json
```

both have made announce

```
15:49:17:297 device announce 0xA4C13889735D4D20 (0x83F5) mac capabilities 0x80
15:46:43:422 device announce 0xA4C1380395605A60 (0x4E2E) mac capabilities 0x80
```

And the 0xa4c1380395605a60 have start a complete inclusion procedure, but was not included ...

I will ask others devs for tips because all seem fine. The device have make a rejoin just after the last step on the inclusion procedure, you have make 2 times the reset ?

And if you try only with "error" and "error_l2" nothing special ?

Smanar, 2022-05-20:
@Ltek sorry what is your deconz version ?

Ltek, 2022-05-20:
> @Ltek sorry what is your deconz version ?

I was on 2.15.3 and then tried 2.16.0 beta...went back to 15.3

> The device have make a rejoin just after the last step on the inclusion procedure, you have make 2 times the reset ?

Not sure what you are asking. I have deleted and re-paired these devices many times (at least 5 or 6 times each device)

I removed and re-paired the TS0201, there's the DDF & Error debug...

16:28:26:954 void zmController::onApsdeDataIndication(const deCONZ::ApsDataIndication&),3950: assertion 'node' failed
16:28:27:788 a4:c1:38:89:73:5d:4d:20/attr/manufacturername expression: Item.val = Attr.val --> _TZ3000_yd2e749y
16:28:28:685 a4:c1:38:89:73:5d:4d:20/attr/modelid expression: Item.val = Attr.val --> TS0201

Smanar, 2022-05-21:
>Not sure what you are asking. I have deleted and re-paired these devices many times (at least 5 or 6 times each device)

The inclusion procedure take lot of step 

```
15:46:44:407 [1] get node descriptor for 0xa4c1380395605a60
15:48:54:238 [2] get active endpoints for 0xa4c1380395605a60
15:47:08:900 [3] get simple descriptor 0x01 for 0xa4c1380395605a60
15:47:10:736 [4.1] Get manufacturer code
15:47:10:736 [4.2] get basic cluster attr 0x0004 for 0xa4c1380395605a60
```

And it the last step, the next one is the addition in the database, but instead we can see the device join again, but with a new network id, 

>I was on 2.15.3 and then tried 2.16.0 beta...went back to 15.3

You have issue with it ?
Because you miss some improvement
- Fix processing rules referencing DDF devices
- Improve loading DDF sensor data from legacy DB tables
- Improve detection NWK address changes and ZDP discovery

Can't make a fast try with this version ?

Ltek, 2022-05-21:
> You have issue with it ? Because you miss some improvement
>
> * Fix processing rules referencing DDF devices
> * Improve loading DDF sensor data from legacy DB tables
> * Improve detection NWK address [...]

Yes...
16.0
-- usually requires two tries (device resets) to pair
-- does not pull in all the data
-- does not associate the DDF ever
-- after reboot (restart deCONZ) ALL cluster information is gone, the device still shows on the UI
---- does not read the device at all even when using Read button: no Error or DDF in debug log at all.

15.3
-- pairs on first try, appears in UI in 2 or 3 seconds
-- pulls in more data from device on pairing
-- associates the DDF immediately
---- installed 15.3 directly over 16.0, deleted the device and re-paired... worked paired immediately, pulling in all info, and associated with DDF (Debug_15.3_pair log attached)

[DEBUG_16.0beta.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8747771/DEBUG_16.0beta.txt)
[DEBUG_15.3pair.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8747773/DEBUG_15.3pair.txt)

Mimiix, 2022-05-23:
Asked @manup to check in.

Ltek, 2022-05-25:
@manup any ideas on this? I cant imagine these are different internally than any of the other Tuya devices?

Ltek, 2022-05-27:
@manup ?

Mimiix, 2022-05-27:
Phoscon won't show them in the first place as it needs seperate support. The rest api should.

I've asked manup again to check

manup, 2022-05-28:
Hmm the temperature and humidity sensors should be shown generically in the Phoscon App when it's in the REST-API.
Do you see the respective sensors in the API information tool?

*"Phoscon App > Menu > Help > API Information > Sensors"*

Is there a product page or Amazon link for this sensor?

Ltek, 2022-05-28:
I've added and removed them at least 20 times each (and tried different physical devices, I have 6 of them). Neither sensor model shows in Phoscon -- I attached the Phoscon info you asked for. 

links to both models...
https://www.aliexpress.com/item/3256802176456322.html
https://www.aliexpress.com/item/3256803451172448.html

[Phoscon_API_Sensors.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8790468/Phoscon_API_Sensors.txt)

Ltek, 2022-05-29:
@manup ?

Mimiix, 2022-05-29:
@Ltek you can't expect an reply within an day, especially in the weekend 😅

Ltek, 2022-05-29:
I'm sure he's busy. I'm just bumping it, it has been 2 days, not 1 ;-) 
I'm eager to get it done, its been 2 weeks since I posted and I feel like I'm banging my head against a wall without progress. Thanks for the help!

manup, 2022-05-31:
The logs show that the sensor resources aren't created, which is the reason the Phoscon App won't show anything and the deCONZ gui shows only the device address as name.

To check if a device picks the right DDF you can select the device in the GUI and press Ctrl+E (or context menu > Edit DDF), but I'd expect it will only show a Draft DDF?

Ltek, 2022-06-02:
@manup ... more testing, still failures...

I removed 16.1 and back to 15.3 = DDF did not load. Did a hot reload and it loaded the DDF properly.
I installed16.1 over the top of 15.3 = DDF did not load. Did a hot reload and it loaded the DDF properly.

So looks like on 15.3, 16.0, and 16.1 it loads the DDF on the initial pairing, loses the DDF on a restart, loads the DDF on Hot Reload. In no case does it ever show in Phoscon.

manup, 2022-06-04:
Interesting, looking at the screenshot there are these duplicates which shouldn't be there, might be related to the problem seen in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5974#issuecomment-1145983104  On Linux this would be indicated if the /usr/share/deCONZ/devices directory contains more files/directories than it should. This is fixed and cleaned up in the upcoming version.

Ltek, 2022-06-06:
> Interesting, looking at the screenshot there are these duplicates which shouldn't be there, might be related to the problem seen in [#5974 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5974#issuecomment-1145983104) On Linux this would be indicated if the /usr/share/deCONZ/devices directory contains more files/directories than it should. This is fixed and cleaned up in the upcoming version.

I'll test and report back when the beta is posted. thx

Ltek, 2022-06-08:
@manup I tested .17 beta and it does not load the DDF ... when device is added new, or with Hot Reload.

this is what the Devices folders looks like...
[image]
[image]

Smanar, 2022-06-09:
>or with Hot Reload

Mean you have done a "edit DDF" and it have open the editor with a DDF ? Or you have tried to make a new one ? (Or was an empty DDF ?)

Ltek, 2022-06-12:
> > or with Hot Reload
> 
> Mean you have done a "edit DDF" and it have open the editor with a DDF ? Or you have tried to make a new one ? (Or was an empty DDF ?)
> 
> Your device is the "_TZE200_qoy0ekbd" ? There is a DDF in your capture for it ?

Please see the screenshots I posted 12 days ago.  The device is the _TZ3000_yd2e749y  (not TZE200_qoy0ekbd) and you can see in 16.1 & 15.3 it pulls in the DDF just fine on initial pair or hot load, but not ever on reboot..

Yes I have opened the Edit DDF function, that is how I see the DDF is not loaded at all on 17. Why would i make a new one if it loads fine in 15.3 & 16.1?  

@manup there seems to be several bugs in 15.3 & 16.1 ... and a new one in 17.x stopping it from loading at all.

manup, 2022-06-15:
Sorry missed that one. Does your DDF editor still show these duplicated entries with v2.17.0-beta?

If yes we need to check the folders containing everything as expected, the installer in v2.17.0 should already force a clean installation of the /devices folder but maybe there's another issue :thinking:

Ltek, 2022-06-16:
> Sorry missed that one. Does your DDF editor still show these duplicated entries with v2.17.0-beta?
> 
> If yes we need to check the folders containing everything as expected, the installer in v2.17.0 should already force a clean installation of the /devices folder but maybe there's another issue 🤔

What duplicate entries - .17's DDF editor for that device is 100% blank. My prior post provided screenshots of the folders and also explains...

```
Please see the screenshots I posted 12 days ago. The device is the _TZ3000_yd2e749y (not TZE200_qoy0ekbd) and you can see in 16.1 & 15.3 it pulls in the DDF just fine on initial pair or hot load, but not ever on reboot..

Yes I have opened the Edit DDF function, that is how I see the DDF is not loaded at all on 17. Why would i make a new one if it loads fine in 15.3 & 16.1?

@manup there seems to be several bugs in 15.3 & 16.1 ... and a new one in 17.x stopping it from loading at all.
```

manup, 2022-06-17:
[image]

The duplicates in your screenshot are in the available items list on the right, it shouldn't look like this as it indicates something wrong with the /devices folder.

Here is an example how it should look like:

[image]

manup, 2022-06-17:
It looks like it doesn't load any DDF relates files, I have no clue why currently. Would you be open for a Teamviewer session so I can check live whats going on, and have a look that the files/directories are correct?

Perhaps on Monday?

Ltek, 2022-06-17:
> It looks like it doesn't load any DDF relates files, I have no clue why currently. Would you be open for a Teamviewer session so I can check live whats going on, and have a look that the files/directories are correct?
> 
> Perhaps on Monday?

Yes for sure... I'd love to get these sensors working! I'll PM you

Ltek, 2022-06-20:
@manup How do you want to arrange this?

Mimiix, 2022-06-20:
Ping him on discord :)

manup, 2022-06-20:
Damn almost 8 o'clock, time flies during coding :D

You can ping me either on Discord or via mail to mpi@dresden-elektronik.de

Ltek, 2022-06-24:
@manup lmk when you want to look at the Tuya device again. I'm posting the new DDF for the other device

Ltek, 2022-06-29:
@manup any thoughts on how to get the Tuya version working?

Ltek, 2022-07-04:
@manup hello?

Ltek, 2022-07-08:
@manup I'm not sure what you are asking in your last msg... you initially asked me, via teamviewer chat, to post the DDF (the one you created for the non-screen device) on github, though you did not specify a specific way or place. So, I did above; the only way I know how. 

You told me you'd work on the 'Tuya screen version' so that is what I'm asking about now.

Ltek, 2022-07-29:
@manup checkin in

manup, 2022-08-10:
Small update here: I have received a package from Nous which contains:
* Temperature and humidity sensor (E5) [link](https://nous.technology/product/smart-humidity-temperature-sensor-nous-e5-zigbee.html)
* Temperature and humidity sensor with display (E6) [link](https://nous.technology/product/e6.html)

Both are branded from Tuya.

[image]

I'll try to get them integrated via DDF, ideally they work the same as your devices due the Tuya firmware, but that needs to be tested next.

Ltek, 2022-09-13:
@manup I installed 18.1 beta and copied the DDF (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6306) into a new file for his TS0601 sensor. Then changed the manufacture info. It read and associates the DDF but the device does not show up in Phoscon.

[image]

Smanar, 2022-09-13:
Strange, the name was changed from 0xXXXX to "Humidity 74", it s the API that do that, can you take a look directlty in the API using phoscon/help/api Information/sensor ?

Ltek, 2022-09-13:
API is FDF8F0FCEC (attached)

I found this for _TZE200_qoy0ekbd

       "74": {
            "config": {
                "battery": 100,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "etag": "73de33aaa50568638ca28e88ddede957",
            "lastannounced": null,
            "lastseen": "2022-09-13T15:51Z",
            "manufacturername": "_TZE200_qoy0ekbd",
            "modelid": "TS0601",
            "name": "Humidity 74",
            "state": {
                "humidity": 3200,
                "lastupdated": "2022-09-13T15:51:04.741"
            },
            "swversion": "1.1.2",
            "type": "ZHAHumidity",
            "uniqueid": "a4:c1:38:03:95:60:5a:60-01-0405"
        },
        "75": {
            "config": {
                "battery": 100,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "etag": "b464895487576eb603eee5a54a507f71",
            "lastannounced": null,
            "lastseen": "2022-09-13T15:51Z",
            "manufacturername": "_TZE200_qoy0ekbd",
            "modelid": "TS0601",
            "name": "Temperature 75",
            "state": {
                "lastupdated": "2022-09-13T15:51:04.421",
                "temperature": 1920
            },
            "swversion": "1.1.2",
            "type": "ZHATemperature",
            "uniqueid": "a4:c1:38:03:95:60:5a:60-01-0402"
        }
    }

[PhosconAPI_FDF8F0FCEC.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9558784/PhosconAPI_FDF8F0FCEC.txt)

Smanar, 2022-09-13:
So the device is in the API and working ? the problem is from Phoscon.

Ltek, 2022-09-13:
It could be. The Homeseer JowiHue plugin pulled the device info/data and I can see it properly. Jowihue goes gets data directly from Deconz.

BTW - these devices (humi/temp with or without screen) something seem temperamental. They pair instantly (show in Deconz within a second) but often the data is not ready at all or only partly. For this one I'm testing now with the screen... 2 out of 5 times it did not load a DDF at all, just left it as 0xXXXX

Smanar, 2022-09-13:
When the name stay 0x0000, it mean the api haven't reconised the device.

Ltek, 2022-09-13:
Yes, I understand that.  I'm saying its not consistent / temperamental. I just added my second Tuya TS0601 with screen and as you can see, it added it as Humidity. It seems to change from Temperature <-> Humidity even within the same device. Its not pairing these devices the same with consistently / predictability.

and Phoscons is not showing them, ever. 

[image]

Ltek, 2022-09-21:
@manup @Smanar with 18.2 these devices still do not show up in Phoscon.

Ltek, 2022-09-27:
hello?

Smanar, 2022-09-27:
Sorry was on holiday ^^.
If I understand correctly, you are not alone, there is some problem with sensor not visible in phoscon  https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.18.2

So if you can see them in the API, will be fine.

For the name, deconz is working with events, so it can depend with one information the device have send first, and yes it's random,

Ltek, 2022-09-28:
18. 2 didn't fix the issue. Those devices are no showing in apps, thus the
API is not providing any data on them. This is why Phoscon doesn't show
them, nor does Jowihue plugin for Homeseer.

Smanar, 2022-09-28:
But your device are in the API no ? we can see them here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6063#issuecomment-1245619482

And when the name change from 0x0000 to "Temperature X" it mean the device is reconized.

Ltek, 2022-09-29:
My attempt at a humorous explanation... 

If Shows_In_Deconz = true
And If Shows_In_Phoscon = false
And If Shows_In_Jowihue-Homeseer = false
Then Deconz_Publishing_Device = false

If Deconz_Publishing_Device = false
Then Bug_in_Deconz = true

[20 more replies]

closed 2023-01-07 by github-actions[bot]