dresden-elektronik/deconz-rest-plugin #6066: _TZ3000_cehuw1lw not reporting power/current/energy

fsecko1, 2022-05-20:
## Device

- Product name: Smart plug Zigbee Plug 20A
- Manufacturer: _TZ3000_cehuw1lw
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 
  - Light
  - Other: plug

## Screenshots

### Basic
[image]
### Identify

### Alarms

### Device Temperature

### Groups

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering
[image]

### Diagnostics

### Other clusters that are not mentioned above
[image]

Mimiix, 2022-05-20:
@Smanar this probsbly is an easy one right?

Smanar, 2022-05-20:
Sure ^^
This device have some issue
- it power itslef alone https://github.com/Koenkk/zigbee2mqtt/issues/11648#issuecomment-1120991469
- it need a minimum consumption to work

Can you try this DDF ?

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3000_cehuw1lw",
    "modelid": "TS011F",
    "product": "Random tuya plug",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/alert",
            "description": "The currently active alert effect.",
            "default": "none"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/consumption"        },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
          {
            "name": "state/power"
          },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 1,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 1,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 1,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 1,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

fsecko1, 2022-05-23:
I´m sorry, but I don´t know, what is DDF and how can I try it. :-(

Smanar, 2022-05-24:
DDF description https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

So you just need to create a file in the "devices" folder and copy/paste this contain.
what is your OS ?

fsecko1, 2022-05-26:
Home Assistant OS 8.1 / Rpi or Windows

Smanar, 2022-05-26:
Ok so bad luck, will be harder ^^

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

closed 2022-06-25 by github-actions[bot]

Purplehazezzzz, 2022-10-23:
This one is still not fixed right?

Purplehazezzzz, 2022-10-23:
[image]
The _TZ3000_cphmq0q7 work fine and show al metrics, the _TZ3000_cehuw1lw only work as a switch no metrics

Smanar, 2022-10-23:
You have a special DDF for them ?
I don't found one of them in code ....

Purplehazezzzz, 2022-10-27:
No still haven't got the _TZ3000_cehuw1lw to work (only as a switch). I tried to use the DDF as mentioned above but that causes the switches to stop working

Purplehazezzzz, 2022-10-29:
Got it working sort of with the above ddf. (didnt place it in the correct folder.) But the measuring seems of. I will have to check this later, to busy next couple of days

Flight777, 2022-12-02:
@Purplehazezzzz Did you already had any luck it getting it to work properly? Struggling with the same issues here.

Smanar, 2022-12-02:
What is your manufacture name ?

maxoss, 2022-12-13:
Same issue no power reporting for the _TZ3000_cehuw1lw. @Smanar how can I help to fix this?

maxoss, 2022-12-14:
I did now. I've duplicated your DDF in the DDF editor in the home assistant addon. First impression looks good. 
Only have a dip to 0 every 4-5min. Not a big deal for me. If I remember correctly I've seen this issue before on Github.
[image]

Smanar, 2022-12-14:
Yes, but idk where this issue is from, perhaps because we are not using the tuya unlock sequence (Z2M do it) but I can't be sure.

maxoss, 2022-12-14:
After some testing it becomes clear that power consumption is a factor 10 too low. Around 30 minutes 2000w gives only 0.1 kwh.

Any idea how I can fix this in the DDF?

[image]

Smanar, 2022-12-15:
Sure

```
        {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 1, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 1, "eval": "Item.val = Attr.val * 10"},
          "refresh.interval": 900
        },
```

jelleklinge, 2023-01-07:
Same issue here, since i am using Deconz and Domoticz i don't know how to resolve this issue. Anyway the _TZ3000_mraovvmm works now after update. But _TZ3000_cehuw1lw can only be set on or off.. Since i don't have a desktop environment installed on Deconz i cannont open the application, only web based or CLI. Please help me out, i have bought 10 pcs.

Smanar, 2023-01-07:
You are on domoticz with a real OS or docker/VM ?
If you have a real OS

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

You can just install the DDF you need on the "devices" folder you prefer.

jelleklinge, 2023-01-07:
> You are on domoticz with a real OS or docker/VM ? If you have a real OS
Thanks i am going to figure this out.
 I am having everything running in Proxmox, Debian 10 VM for Deconz and another VM for Domoticz under Ubuntu.
Its a 3.5 year running system, but i try to keep everything up to date *Latest stable Deconz.

Smanar, 2023-01-07:
VM are more friendly than docker, I think you will found exactly the same folder name.

jelleklinge, 2023-01-07:
I Did thanks, now for first time using DDF, reading the cheat sheet at the moment, its not very clear to me. Might be my way of thinking of some sort. But i copied the DDF from above but cannot find the {
            "name": "state/consumption"        }, to change that file. 
Can you post te working DDF please this would be greatly appreciated.

Smanar, 2023-01-07:
```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3000_cehuw1lw",
    "modelid": "TS011F",
    "product": "Random tuya plug",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/alert",
            "description": "The currently active alert effect.",
            "default": "none"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/consumption",
            "read": {"at": "0x0000", "cl": "0x0702", "ep": 1, "fn": "zcl" },
            "parse": {"at": "0x0000", "cl": "0x0702", "ep": 1, "eval": "Item.val = Attr.val * 10"},
            "refresh.interval": 900
          },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
          {
            "name": "state/power"
          },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 1,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 1,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 1,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 1,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

jelleklinge, 2023-01-07:
TTHX!!! Ii works like a charm!

BabaIsYou, 2023-01-25:
Seems that the same DDF also works for  Mini Smart Plug _TZ3000_gznh2xla #6691 ;-)

Smanar, 2023-01-25:
Yeah it s for that I have called this DDF 

`"product": "Random tuya plug",`

We can use this one as base for all future clones that don't have special feature or real brand ^^

BabaIsYou, 2023-01-26:
@Smanar  Is there already a opened PR that I can join ?

Smanar, 2023-01-26:
Arf, my bad, the DDF name file is not genric, it's this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_cehuw1lw_smartplug_EU.json

And from my memory you can use it ATM, no PR using it.