dresden-elektronik/deconz-rest-plugin #6067: Support for EJLINK EK02 Wireless Switches

isamaru, 2022-05-22:
## Device
(3 devices)
- Product name: EK02 (86/1CH, 86/2CH, 86/3CH) Zigbee Push Button Remote Panel White
- Manufacturer: _TZ3000_tk3s5tyg, _TZ3000_adkvzooy, _TZ3000_qzjcsmar
- Model identifier: TS0041, TS0042, TS0043
- Device type:
  - Switch

## Screenshots
Endpoints and clusters:
[image]

1-gang:
[image]

2-gang:
[image]

3-gang:
[image]

### Basic
Read cluster does not work very well, it's only available after pairing. Later it returns blank values.
1-gang:
[image]

2-gang:
[image]

3-gang:
[image]

### Power Configuration
Same for all 3 devices, only empty data is being returned:
[image]

### Groups
Strange that the group cluster sometimes appears and sometimes not after pairing, even for same device.

### On/Off
Same for all on/off clusters:
[image]

## Description
[image]

A family of wireless switches from EJLINK, 1-gang, 2-gang, and 3-gang. They are all based on the same board (YC-ZS-L04C V1.0)

They can be paired to gateway but work very poorly, especially in the Phoscon app:
- read cluster works strange, I am not getting any useful information from that after pairing
- 1-gang switch does not send button events at all, and can't be linked as a switch in the Phoscon app
- 2-gang and 3-gang switches are somewhat usable: they do send button events (1002, 1003, 1004, 2002, 2003, 2004 for both, plus 3002, 3003, 3004 for the 3-gang), but only the single-press can be associated with light groups in the phoscon app. 
- the switches sometimes "fall asleep", then after pressing a button rejoin the network in a couple of seconds, only after that they send events again
- they drain battery very fast (couple of days). I found this was also reported with a CC2531 gateway, but does not manifest with a Tuya gateway. The seller recommends disabling reporting of battery status to address this problem. 
- 1-gang and 2-gang show up in the phoscon app as lights:

[image]

Mimiix, 2022-05-22:
Normally  we ask users to create a device request per device as they might differ. 

@Smanar can you tell that these devices are different from each other or are they the same?

@isamaru If they are different, I'm going to ask you to create a device request per device.

Smanar, 2022-05-22:
Only _TZ3000_adkvzooy and _TZ3000_qzjcsmar are supported by code ATM (without DDF).
Need to use DDF for the 1 gang

And they create useless "light" entries.
Need to use DDF to prevent that, but last time I have tried was some versions ago, and not working.

Need to use too DDF to set the reporting.

So to resume, choose a device and I can give you a DDF to test, for for information there is so much issue for battery switches on DDF.

Mimiix, 2022-05-22:
@Smanar so it would be better to have 3 separate issues?

Smanar, 2022-05-22:
Can make all on same PR.
In the better situation will have 3 new DDF and a change on the button json file for all 3.

Only the tuya scene button is special, all other are same.

Mimiix, 2022-05-22:
@Smanar I want to know what's best for the process, not what's possible 😅

Smanar, 2022-05-23:
lol, for me it s same.
All will works or none of them, so can group them in the issue.

isamaru, 2022-06-04:
Hi, do you need any more information from me?

Update about the switches: the 3-gang burned through 2 additional batteries in the last 2 weeks, but is working OK while the battery is fresh, I can toggle lights with all 3 buttons. Other than battery drain, the remaining problems are that I can't bind any events other than single-push in the phoscon UI, and that the I see it incorrectly as lights (minor).
The 2-gang doesn't suffer from battery drain so far, otherwise has same minor issues.
The 1-gang doesn't work at all.

Smanar, 2022-06-05:
You have a sample here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6068

For exemple for the 1 Gang, use this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_tk3s5tyg",
  "modelid": "TS0041",
  "vendor": "EJLINK",
  "product": "EK02 86",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

But you need too edit the button_maps.json file, to add the device. like on the sample.

isamaru, 2022-06-11:
I tried to find a way to test this, but could not find any process for deploying my own build as an HA addon.
@Mimiix said on discord that it was not possible as far as he knows.

Smanar, 2022-06-11:
This link can't help you ? https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/2

isamaru, 2022-06-12:
I managed to upload the files (the new DDF + button_maps.json update) inside the docker, then restarted the container from the host system (`docker restard [id]`), but it doesn't seem to be enough - removing and pairing the switch looks exactly the same = no button events.
Not sure if the file changes are picked up properly, is there a way to check?

Smanar, 2022-06-12:
Yes on deconz logs.
On deconz/help/debug view with "info" and "info_l2"

Or can just check the file to see if the changes are still here.

isamaru, 2022-06-12:
It works weird. The DDF is there, and it's picked up:
[image]

But no matter what I do, there is still the light subdevice and it doesn't work. When I push the button and the device wakes I can see this:
[image]

BUT when I retried this for the first time today, it was working fine for a while, registering events. But never managed to repeat that.

Smanar, 2022-06-13:
Ok so same issue than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5889

You need to have ONLY the sensor entry, and I don't understand why you have too the "light" entry, this issue was corrected 2/3 version ago ....

You have correclty deleted the previous entry before making again a new inclusion ?

If I m right you will see the device in "light" and "sensor" ATM.

isamaru, 2022-06-14:
I'm on addon version 6.14.1 (using deconz 2.16.1)
I watched this [video](https://www.youtube.com/watch?v=rgTQpOVg8w0) and it's not behaving like that for me. Even though the log says that the ddf has been found for this device, I'm still stuck with incorrect vendor on the subdevice, unlike in the video:
[image]

Yes, I still see both light and sensor in the phoscon app, and with the REST API as well. Here's the current REST API response for the light (at `http://homeassistant.local:40850/api/{{API_KEY}}/lights/27`):
`{
    "etag": "3249c39c18b227609daf043625ddb0c5",
    "hascolor": false,
    "lastannounced": "2022-06-14T18:38:22Z",
    "lastseen": "2022-06-14T18:39Z",
    "manufacturername": "_TZ3000_tk3s5tyg",
    "modelid": "TS0041",
    "name": "On/Off light 27",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": false
    },
    "swversion": null,
    "type": "On/Off light",
    "uniqueid": "0c:43:14:ff:fe:31:48:71-01"
}`

It is possible that I did not delete the old entries correctly, but not sure how I can do that. I do not think so, though, because the light subdevice (which should not be there) gets a fresh light ID each time it is deleted and paired again (checked via rest API), and so does the sensor.

Smanar, 2022-06-15:
>I'm still stuck with incorrect vendor

It's fine, Phoscon say "Vendor" but it's the manufacture name, so we are fine.

But we are fine for the "vendor", as this device is a battery one, you need to don't have a light entry, so if you have this one, there is a problem.

```
{
  "etag": "3249c39c18b227609daf043625ddb0c5",
  "hascolor": false,
  "lastannounced": "2022-06-14T18:38:22Z",
  "lastseen": "2022-06-14T18:39Z",
  "manufacturername": "_TZ3000_tk3s5tyg",
  "modelid": "TS0041",
  "name": "On/Off light 27",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": false
  },
  "swversion": null,
  "type": "On/Off light",
  "uniqueid": "0c:43:14:ff:fe:31:48:71-01"
}
```

You need to have only the sensor entry, we had previosuly issue with that, but they was corrected by this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6021 and you have it in the deconz version 2.16.1

And It work for some users, so I can't be sure where the issue is from.
I will ask for Manup take a look, but just to be sure you have deleted the device entry before ? (Just to be sure the "light" entry was already here before the inclusion)

isamaru, 2022-06-15:
I've been deleting and re-pairing the switch for some time now. Most of the time it doesn't work at all and ddf isn't registered. But I've had some random successes (2 times out of maybe 50) where after the pairing the device was sending button press events correctly. No idea what was different about those 2 pairings, but they were both associated to a hot reload.

As for deleting, there are 3 ways to delete it, I've been alternating them but couldn't find any way in which they differ: delete light in phoscon, delete switch in phoscon, delete node in deconz. All 3 appear to have the same result: light, switch, and node all disappear.

Smanar, 2022-06-16:
>All 3 appear to have the same result: light, switch, and node all disappear

Yeah, some version ago you can can delete a specific entry, but no more possible.

>No idea what was different about those 2 pairings

I think on thoses one, you don't have the "light" entry created.

But from that @manup say to me, it can be random, can be provoked because the device "start to present itlsef", before the DDF engine bypass the legacy code.
He will check if it's possible to put some securities in the code to prevent that.
But as it will need c++ code modification, it will take more time.

closed 2022-07-15 by github-actions[bot]