dresden-elektronik/deconz-rest-plugin #6072: Missing documentation/funcionality: How to revoke App-Connection

mspbonn, 2022-05-23:
## Describe the bug

There is a button to connect 3rd party apps like 'Hue essential'. But how to revoke the App-Connection ?

## Steps to reproduce the behavior

No functionality to revoke the App-Connection.

## Expected behavior

Button to revoke connection.

## Screenshots

None.

## Environment

 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: 26690700
 - deCONZ version: 2.13.02
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

None

## Additional context

None

Thomas-Vos, 2022-05-25:
In Hue Essentials you can go to the Devices tab, select your bridge, then tap Apps. There you will see a list of all authorized apps and you can remove them.

There is also a Remove All button at the top which removes all apps except the one you are currently using. But you can remove it manually.

I think the only other option is to do this using the API.

mspbonn, 2022-05-25:
I need a possibility to remove the client, because the bridge is installed in a rental apartment. I cannot force a renter to remove the bridge on their on smartphone.

SwoopX, 2022-05-25:
https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#delete-api-key

closed 2022-06-23 by github-actions[bot]