dresden-elektronik/deconz-rest-plugin #6075: /config/localtime in rules does not work since deCONZ 2.13.x

meipat, 2022-05-25:
## Describe the bug

## Steps to reproduce the behavior
Since Deconz v2.13.x, my rules no longer work in which /config/localtime is used. I just tried with v2.16.01. The only thing I can do is to rollback to 2.12.x.

Here is an example.

```
{
    "name": "Bath motion detection at day",
    "actions": [
        {
            "address": "/groups/8/scenes/1/recall",
            "body": {},
            "method": "PUT"
        }
    ],
    "conditions": [
        {
            "address": "/sensors/25/state/presence",
            "operator": "eq",
            "value": "true"
        },
        {
            "address": "/config/localtime",
            "operator": "in",
            "value": "T07:00:01/T20:00:00"
        }
    ]
}
```

Has anything changed regarding /config/localtime? If yes, what?

## Expected behavior
The light goes on.

## Screenshots
None

## Environment

 - Host system: (Raspberry Pi 3 B+)
 - Running method: Raspbian 11.3
 - Firmware version: 26400500
 - deCONZ version: 2.16.01
 - Device: RaspBee I
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system: no

## deCONZ Logs
```
10:14:57:757 rule event /config/localtime: 10:14:56.755 -> 10:14:57.756 (1001ms)
10:14:58:756 rule event /config/localtime: 10:14:57.756 -> 10:14:58.755 (999ms)
10:14:59:443 rule event /sensors/25/state/presence: 0 -> 0
10:14:59:443 rule event /sensors/25/state/lastupdated: 0 -> 0
10:14:59:443 rule event /sensors/25/state/presence: 1 -> 1
10:14:59:443 rule event /sensors/25/state/lastupdated: 0 -> 0
10:14:59:756 rule event /config/localtime: 10:14:58.755 -> 10:14:59.755 (1000ms)
10:15:00:772 rule event /config/localtime: 10:14:59.755 -> 10:15:00.766 (1011ms)
10:15:01:756 rule event /config/localtime: 10:15:00.766 -> 10:15:01.755 (989ms)
10:15:02:756 rule event /config/localtime: 10:15:01.755 -> 10:15:02.755 (1000ms)
10:15:03:757 rule event /config/localtime: 10:15:02.755 -> 10:15:03.755 (1000ms)
10:15:04:756 rule event /config/localtime: 10:15:03.755 -> 10:15:04.755 (1000ms)
```

## Additional context
None

Mimiix, 2022-05-25:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5806

That has something to do with it.

meipat, 2022-06-02:
@Mimiix Thanks for your feedback. If I see it correctly, this should be fixed since v2.14.1. I see the changes from #5806 in the current code. But it does not work in any version since v2.14.x.

closed 2022-07-01 by github-actions[bot]