dresden-elektronik/deconz-rest-plugin #6078: Devices randomly change reported State while beeing off all the Time

dafunkydan, 2022-05-26:
## Describe the bug
After a Tradfri Motion Sensor gets triggered, some Lights change their reported States, while beeing physically off all the Time.
Some Lights change back to report their true State after 2 Mins, some after 4 Mins, some stay on for longer.
I observe that Problem since yesterday. 
What i suspect: Yesterday i removed one Motion-Sensor, repaired and renamed it (first Friendly Name, Entity after a Restart. This Sensor isn't bound to any group in Phoscon. In HA it is to _some_ of the Devices. But at the Time the errors occur, it doesn't trigger any Light-Changes HA. Furthermore, some Device in no way related to the Sensor get reported wrong.

Additional Information: Yesterday, in HomeAssistant i removed the zha-Addon, which only had one (old) Device, the Addon was disabled for month. So it really shouldn't be related, but regarding the Time the Error first observed, it is.

## Steps to reproduce the behavior

  1. Restart HomeAssistant
  2. Make sure all Devices are reported as "off"
  3. Trigger the Motion Sensor mentioned above
  4. Watch the Devices report wrong States, while they stay physically "off"
  5. Wait a Couple of Minutes, til the reported States start to turn back to "off"

## Expected behavior
Only report true States

## Screenshots

## Environment

 - Host system: Raspberry Pi4
 - Running method: Home Assistent OS, deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.15.03 / 21.4.2022
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Any other USB or serial devices connected to the host system: No

## deCONZ Logs
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8777911/log.txt)

## Additional context
After renaming that Sensor (+entity), the Battery Level (witch should be an additional entity) can't be found.
Yesterday, for testing purpose i only renamed the Motion Sensor and one Light. The other Lights, which are the ones affected, haven't been changed.
Though diggin around, i couldn't find any straight information about how to Clean Rename Device Names and their belonging Entities.

wvuyk, 2022-05-26:
I am seeing similar behavior, mainly on plugs and lights. I don't know what is causing it, I did not rename anything here lately. I think of the 27 lights and plugs here present on my Raspbee there are 5 devices reporting wrong states at random times. A very rough estimation seems to be every half an hour, but it also skips now and then.
The devices are 2 INRR plugs, one Philips Hue White light, a IKEA Floalt light and a LIDL Silvercrest outlet device.
I did check if my JowiHue plugin might be causing this, but the state is coming from the webhook send by deCONZ.

The issue started for me with 2.15.03 and continues in 2.16.1

johndoewashere2, 2022-06-10:
Having a similar issue with my setup as well, unsure when exactly this occurred however it's probably since the last two releases. 

Setup is current version of HA Supervised install (2022.6.5), with the latest deCONZ add-on (6.14.1). 

Using a a mix of Philips Hue lights, sensors and plugs combined with Wi-Fi lights in HA's light groups, I can verify this does occur with lights connected to my Conbee II USB dongle rather than my Wi-Fi lights in the HA light groups. 

[image]

As per the above screen shot showing the status of the light group, it's a random set of Zigbee entities changing states for ~15 seconds, despite all of them being off in real life. 

What's odd is the last two entries (on the top) of the screen shot, neither of the lights were on however it's exactly the same time when I walked into the room, triggering the Hue motion sensors.

closed 2022-07-09 by github-actions[bot]