dresden-elektronik/deconz-rest-plugin #6080: Remote TUYA

sebsteph51, 2022-05-27:
## Device

- Product name: Remote TUYA.
- Manufacturer: _TZ3000_ygvf9xzp
- Model identifier: TS0044
- Device type : Please remove all unrelated device types. 
   - Switch
 

## Screenshots
[image]
[image]
[image]
[image]

Mimiix, 2022-05-27:
Missing screenshots of all clusters.

sebsteph51, 2022-05-27:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Sorry, I have take a picture of all cluster, the name of picture explain the cluster of device.

Smanar, 2022-05-27:
Lol, you make same screenshoot than my boss at work ^^

Can try this DDF 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ygvf9xzp",
  "modelid": "TS0044",
  "product": "Tuya remote 4 gangs",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
} 
```

And you need too edit the button_maps.json file (in the "devices" folder too)

```
       "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0"],
```

Just add your model in the list "_TZ3000_ygvf9xzp"

You will miss the battery, but it s a test because I m not sure it s possible only with DDF

sebsteph51, 2022-05-28:
Lol I haven't visual access on my odroid. 

Thanks a lot the remote is pair.

Thanks Thanks Thanks.

Smanar, 2022-05-28:
Seriously on first try ?
All is working ? Can make a PR ?

sebsteph51, 2022-05-28:
a simple click it's ok
a double click it's ok
a long click isn't ok
i haven't battery but i don't now if i can obtain it.

Smanar, 2022-05-28:
You haven't pressed the "read" button on your capture, so I m not sure for battery and have used defaut code

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ygvf9xzp",
  "modelid": "TS0044",
  "product": "Tuya remote 4 gangs",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

For the long press, can you go in deconz / help / debug view, enable flags "info" and "info_l2" and make a long press.
You probably will see an error message, if you can share it ?

sebsteph51, 2022-05-30:
I have change json for the battery, but i haven't the battery in domoticz. I have look info and info L2 and action = B4 short or B4 long but in domoticz Bx = simple click, Lx = long click, Dx = double click.

[image]
[image]
[image]

Thanks a lot for your kelp.

Smanar, 2022-05-30:
The problem can be from domoticz too (I can make correction too)
Just to be sure, button event from deconz are X00Y with X is the button number and Y the button state (2 = release after short press, 3 = release after long press, 4 release after double press)

Can you check values send by deconz by the websocket ?

You can see them too in domoticz with enabling "debug log only" in the plugin tab.

Still no battery (I m seing 0 value on your previous capture) ?

sebsteph51, 2022-06-07:
Sorry for this long time but i was in work. I have test with web socket and i have :
	print(str(data["sensors"]["28"]["state"]["buttonevent"]))
3002 for simple click on button 3 and 3004 for double click on bouton 3, 
4002 for simple click on button 4 and 4004 for double click on bouton 4, ...

[image]

long press is not detect.

sebsteph51, 2022-06-09:
Sorry, but it's ok for a long press. it was me the problem. The time for a long press is 4 or 5s but i not click long enough. Sorry

sebsteph51, 2022-06-10:
I has 0 % battery for websocket. I have click on all button for test if the battery is send with info buttton. The battery is stay at 0% for web socket

On Deconz i have click on power configuration, Read cluster and value of battery 0x0020 = 30 and 0x0021 = 200.

[image]

i have test after manipulation on deconz and i have 100 on websocket

[image]

Smanar, 2022-06-10:
Perhaps a binding issue, but all is almost working.
PR done, thx https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6125

Mimiix, 2022-07-02:
I'd prefer to keep it open until it's merged.

closed 2022-07-06 by manup