dresden-elektronik/deconz-rest-plugin #6081: Legrand 0 677 55 Wireless Scenes Command

Didier3L, 2022-05-28:
## Device

- Product name: 4 scenario wireless control
- Manufacturer: Legrand
- Model identifier: Wireless Scenes Command
- Device type : Wall switch Light
  

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

## Poll Control
[image]

## Power configuration
[image]

## Node info
[image]

Smanar, 2022-05-28:
Ok so will be the same than this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5021

Unfortunately IDK if it will be possible using DDF.

Can you share at least a DDF skeleton ?
- fill "basic identification cluster" is not auto filled
- set the status to "Gold"
- drag and drop a ZHAswitch
- save the file, remember the path for later
- share the result.

If you make a "hot reload" the device will be reconised, but not working
Can you share the log when using it (on deconz / help / debug view with "info" and "info_l2") , you will have error message about button or mao.

Didier3L, 2022-05-28:
[image]

Didier3L, 2022-05-28:
[image]
[image]

Didier3L, 2022-05-28:
<html><body>

21:57:09:056 ZCL attribute report 0x0C4314FFFECB72C2 for cluster: 0x0201, ep: 0x01, frame control: 0x08, mfcode: 0x0000
--

<pre class="allowDefCtxMenu selectabletext" ext:qtip="21:57:20:832 Poll control check-in from 0x000474000092B22E
">21:57:20:832 Poll control check-in from 0x000474000092B22E</pre>
</body>
</html>

Didier3L, 2022-05-28:
Debug ?
           - DEBUG_INFO=1
           - DEBUG_APS=0
           - DEBUG_ZCL=0
           - DEBUG_ZDP=0
           - DEBUG_OTAU=0

Smanar, 2022-05-28:
Thoses debug flag are for the command line.
As you use the GUI, it s easier, just on on the menu on help / debug view and enable "Info" and "info_l2" then press a button.

BTW can you copy paste the DDF (on the third tab on the editor)
I haven't asked what is your third app ? To know if you are able to edit the DDF with a text editor later ?

Didier3L, 2022-05-28:
23:22:22:961 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 4

Didier3L, 2022-05-28:
'
23:22:21:321 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 3

Didier3L, 2022-05-28:
23:22:22:961 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 4

23:38:40:136 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEB, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EBFF010000, zclSeq: 8

23:38:38:318 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 7

23:38:36:870 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 6

23:38:34:385 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EDFF010000, zclSeq: 5

23:38:31:874 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEB, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EBFF010000, zclSeq: 4

23:38:29:330 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 3

23:38:26:534 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 2

23:38:24:025 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EDFF010000, zclSeq: 0

23:50:37:152 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 3
23:50:37:153 Websocket 192.168.1.243:49772 send message: {"attr":{"id":"47","lastannounced":"2022-05-28T21:49:43Z","lastseen":"2022-05-28T21:50Z","manufacturername":"Legrand","modelid":"Wireless Scenes Command","name":"Wireless Scenes Command","swversion":"001c","type":"ZHASwitch","uniqueid":"00:04:74:00:00:92:b2:2e-01"},"e":"changed","id":"47","r":"sensors","t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 356)

23:54:15:455 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 0
23:54:15:782 APS-DATA.indication from child 0x63EE
23:54:15:783 DeviceAnnce of SensorNode: 0x000474000092B22E [1]
23:54:15:785 Websocket 192.168.1.243:49772 send message: {"attr":{"id":"47","lastannounced":"2022-05-28T21:54:15Z","lastseen":"2022-05-28T21:54Z","manufacturername":"Legrand","modelid":"Wireless Scenes Command","name":"Wireless Scenes Command","swversion":"001c","type":"ZHASwitch","uniqueid":"00:04:74:00:00:92:b2:2e-01"},"e":"changed","id":"47","r":"sensors","t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 356

Smanar, 2022-05-29:
Ok so seem good.
But as you haven't copy / paste the DDF, you will need to make change yourself.

For information Legrand stuff use a special method for payload, just need to edit a line in c++ code, but as now we can't use this way will try with DDF

In the DDF editor, select the state/buttonevent on the parse part try

- endpoint 0x01 (or just 1)
- cluster": "0x0005"
- "cmd": "0x05"
- "eval": "Item.val = ZclFrame.at(1)"

On the menu make "hot reload" and retry to see if it s better on log.

Didier3L, 2022-05-29:
> Mais comme vous n'avez pas copié/collé le DDF, vous devrez faire le changement vous-même.

Bonjour, Désolé
voici wireless_scenes_command.json

 `{
  "schema": "devcap1.schema.json",
  "manufacturername": "Legrand",
  "modelid": "Wireless Scenes Command",
  "product": "Wireless Scenes Command",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/wireless_scenes_command.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}`

Smanar, 2022-05-29:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Legrand",
  "modelid": "Wireless Scenes Command",
  "product": "Wireless Scenes Command",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "parse": {
            "cl": "0x0001",
            "at": "0x0021",
            "eval": "Item.val = Attr.val / 2"
          },
          "awake": true
        },
        {
          "name": "state/buttonevent",
          "parse": {
            "cl": "0x0005",
            "cmd": "0x05",
            "eval": "Item.val = ZclFrame.at(0)"
          },
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
  ]
}
```

If you can replace your contain by this one ?
Have added the battery (but not sure it will work Legrand have something special for battery)
And the previous tips, If I m right you will have new usefull debug line with it. Can just make a "hot relaod", ne need to re-include the device or restart deconz (You have jeedom I presume ?)

Smanar, 2022-05-29:
And I can't edit my post, but you can too set "Sleeper" to true.

Didier3L, 2022-05-29:
> Vous avez jeedom je présume ?
Yes

Didier3L, 2022-05-29:
Log
```
16:30:27:546 expression: Item.val = ZclFrame.at(0) --> -19
16:30:27:789 Websocket 192.168.1.243:49342 send message: {"e":"scene-called","gid":"65517","r":"scenes","scid":"1","t":"event"} (ret = 70)
16:30:27:790 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EDFF010000, zclSeq: 2
16:30:27:797 Websocket 192.168.1.243:49342 send message: {"e":"changed","id":"47","r":"sensors","state":{"buttonevent":-19,"lastupdated":"2022-05-29T14:30:27.546"},"t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 159)

16:30:27:854 Force read attributes for ZHASwitch SensorNode Wireless Scenes Command

16:30:29:859 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('47', 'Wireless Scenes Command', 'ZHASwitch', 'Wireless Scenes Command', 'Legrand', '00:04:74:00:00:92:b2:2e-01', '001c', '{"buttonevent":-19,"lastupdated":"2022-05-29T14:30:27.546"}', '{"battery":0,"on":true,"reachable":true}', '', 'normal', '1', '2022-05-29T14:29Z', '2022-05-29T14:29:34Z')

16:30:30:584 expression: Item.val = ZclFrame.at(0) --> -20
16:30:30:812 Websocket 192.168.1.243:49342 send message: {"e":"scene-called","gid":"65516","r":"scenes","scid":"1","t":"event"} (ret = 70)
16:30:30:812 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 3
16:30:30:819 Websocket 192.168.1.243:49342 send message: {"e":"changed","id":"47","r":"sensors","state":{"buttonevent":-20,"lastupdated":"2022-05-29T14:30:30.584"},"t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 159)

16:30:33:406 expression: Item.val = ZclFrame.at(0) --> -21
16:30:33:634 Websocket 192.168.1.243:49342 send message: {"e":"scene-called","gid":"65515","r":"scenes","scid":"1","t":"event"} (ret = 70)
16:30:33:634 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEB, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EBFF010000, zclSeq: 4
16:30:33:643 Websocket 192.168.1.243:49342 send message: {"e":"changed","id":"47","r":"sensors","state":{"buttonevent":-21,"lastupdated":"2022-05-29T14:30:33.406"},"t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 159)

16:30:34:668 Websocket 192.168.1.243:49342 send message: {"attr":{"id":"47","lastannounced":"2022-05-29T14:29:34Z","lastseen":"2022-05-29T14:30Z","manufacturername":"Legrand","modelid":"Wireless Scenes Command","name":"Wireless Scenes Command","swversion":"001c","type":"ZHASwitch","uniqueid":"00:04:74:00:00:92:b2:2e-01"},"e":"changed","id":"47","r":"sensors","t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 356)

16:30:36:625 expression: Item.val = ZclFrame.at(0) --> -22
16:30:36:878 Websocket 192.168.1.243:49342 send message: {"e":"scene-called","gid":"65514","r":"scenes","scid":"1","t":"event"} (ret = 70)
16:30:36:879 [INFO] - No button map for: Wireless Scenes Command, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 5
16:30:36:886 Websocket 192.168.1.243:49342 send message: {"e":"changed","id":"47","r":"sensors","state":{"buttonevent":-22,"lastupdated":"2022-05-29T14:30:36.625"},"t":"event","uniqueid":"00:04:74:00:00:92:b2:2e-01"} (ret = 159)

```

Didier3L, 2022-05-29:
Génial 😊😉
[image]

Smanar, 2022-05-29:
Nice, but to respect Deconz rules, event need to be X002 for "release button" with X is the button number, so you need to have 1002, 2002, 3002, 4002

So you can edit direclty the DDF file with replacing

`"eval": "Item.val = ZclFrame.at(0)"`

by 

`"eval": "Item.val = (ZclFrame.at(0) + 22) * 1000 + 2"`

Or can do same using the editor.
I m not sure for the typo (and convertion) but I think you will understand what I m trying to do.

If all is fine you will have as button event 1002 2002 3002 and 4002.

Now the bad news.
This device try to use new group every time you press a button, so we need to block them, 2 thing to check in logs:
- You need to not see something like
```
10:22:04:211 delete old group 65516 of sensor Pocket remote 2
10:22:04:212 reanimate group 65515 for sensor Pocket remote 2
```
- you need to not have the "group" field in "info brutes" in jeedom, for exemple this one is bad ("group" : 13 )
```
{
  "e": "added",
  "id": "2",
  "r": "sensors",
  "sensor": {
    "config": {
      "battery": null,
      "group": "13",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "ff33f9691376122c64c6815d213b0034",
    "id": "2",
    "lastseen": "2021-07-08T10:21Z",
    "manufacturername": "Legrand",
    "mode": 1,
    "modelid": "Pocket remote",
    "name": "Pocket remote 2",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:04:74:00:00:83:40:f2-01-0005"
  },
  "t": "event",
  "uniqueid": "00:04:74:00:00:83:40:f2-01-0005"
}
```

Didier3L, 2022-05-30:
For the battery here is the code

```
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "eval": "const vmin = 20; const vmax = 30; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "default": 0
        },
```
[image]

[wireless_scenes_command.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8799566/wireless_scenes_command.json.txt)

**Many thanks for your help** 😎

Smanar, 2022-05-30:
Ha nice thx a lot, so even your device support it, it have problem on the attribute 0x0021 but it work fine on the 0x0020, good to know

Do you want to make a PR to add the device officialy or do you want I make it ?

Didier3L, 2022-05-30:
yes I want you to. thank you

Smanar, 2022-05-30:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6096
Have just edited the binding to use the same attribute you are using, the 0x0020

closed 2022-06-04 by manup

Didier3L, 2023-01-01:
hello
I just updated deconz with version 2.19.03 / 04/21/2022

When I included my Legrand 0 677 55 Wireless Scene Command, it was still not recognized!

Didier3L, 2023-01-03:
Good evening
I modified the file directly in /usr/share/deCONZ/devices/wireless_scenes_command.json

Smanar, 2023-01-04:
I have made a PR for this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6642

But yes for the moment, can use DDF.