dresden-elektronik/deconz-rest-plugin #6086: Unstable connection with WOOX 7060 irrigation Valve

nielsvlaarhoven, 2022-05-29:
## Describe the bug

## Steps to reproduce the behavior
Pair the Woox R7060 with your home automation system using deCONZ (in my case Home Assistant) and then reboot deCONZ or Home Assistant (for example to make the entity available in Home Assistant)
I have also tried to skip deCONZ and pair it in ZHA instead of deCONZ, but this show the same behavior.

## Expected behavior
I would expect the Woox R7060 to stay paired with my zigbee network, and i would expect it to join the network just like my other zigbee devices, and pair with multiple devices for a stable connection.

## Screenshots
Woox R7060 lost connection to the mesh network
[image]
Woox R7060 is paired but it only connects to one zigbee device (quite far away from it in comparison to other zigbee devices)
[image]

## Environment
Running 
 - Host system: PC / Server with ESX host
Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1
Kernel version 5.15.41
Agent version 1.2.1
 - Running method: Virtual Machine
 - Firmware version: 26660700 (firmware of Conbee II )
 - deCONZ version: 6.13.0
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: Just USB for power supply to a ESP-01 board.

## deCONZ Logs

Debug log won't show much since a reboot is needed for the device to lose it's connection the the mesh network. 

## Additional context
Using a WOOX R7060 for the water irrigation of my garden, the rest of the devices are mostly IKEA Tradfri bulbs or GU10 lights. 
Also use some xiaomi stuff like the movement sensors and some ICASA build in and NIKO wall light switches

Smanar, 2022-05-29:
Unfortunately this device is an end device like a simple temperature sensor, and can have only 1 connection to a router.

Mimiix, 2022-05-29:
Hi,

As this is not a bug report as described in #5113 I am closing this issue. Feel free to open an topic on the forum.

Kind regards,
Dennis

closed 2022-05-29 by Mimiix