dresden-elektronik/deconz-rest-plugin #6089: hue motion sensor: config/delay is ignored and reset after some time

daniel1v, 2022-05-29:
## Describe the bug
changing the config/delay value of my hue motion sensors (v1 / v2), via REST API (PUT) does not have any effect. although the value changes for some time (validated over GET), the time between "motion detected" and "no motion detected" is always about one minute.

after some time / and some motion detections, the value resets itself to the default value (62 seconds in my case)

## Steps to reproduce the behavior
1. set config/delay value over PUT
2. validete over GET
3. trigger some motion
4. watch the sensor to reset the "presence" state

## Expected behavior
the value should be respected and also should not be reset by itself

## Screenshots
not available

## Environment
Raspberry Pi / bullseye
home assistant / deconz-community docker container
deCONZ 2.16.01
Conbee II 26780700
## Describe the bug

## deCONZ Logs

## Additional context

ebaauw, 2022-05-29:
The Hue motion sensor sends a message when it no longer detects motion, and the REST API plugin shouldn't interfere with that.  Setting `config.delay` should write the _PIR Occupied to Unoccupied Delay_ attribute in the _Occupancy Sensing_ cluster, causing the sensor to delay the message that it no longer detects motion.  Via the C++ code, that attribute is written through `config.pending`, but I don't know if that still happens through the DDF file.

Could you double-check:
- That setiing `config.delay` updates the _PIR Occupied to Unoccupied Delay_, once `config.pending` has cleared;
- That updating _PIR Occupied to Unoccupied Delay_ in the deCONZ GUI does in fact change the delay before the sensor reports Occupance as 0x00.

daniel1v, 2022-05-31:
@ebaauw thank you for your reply and sorry for my late reply. as I see, in the meantime a pull request was opened, which will solve the issue, so I think there are no further investigations necessary.

SwoopX, 2022-05-31:
Yeah, the issue is/was that the respective write functions were previously not included in the DDF. This became necessary when the approach of writing data to the device was shifted from legacy to new approach.

closed 2022-06-02 by manup