dresden-elektronik/deconz-rest-plugin #6090: ZM79E-DT

vlude, 2022-05-29:
## Device

- Product name: Zemismart ZM79E-DT
- Manufacturer:  _TZE200_3i3exuay
- Model identifier: TS0601
- Device type : 
  - Curtain

## Screenshots
### Basic
[image]

### Node info
[image]

### Groups
[image]

### Scenes
[image]

vlude, 2022-06-20:
Hi, is there any update on this one?

Mimiix, 2022-06-20:
Nope, a Dev needs to pick it up. They do that when they please. Perhaps @Smanar wants to help you out.

vlude, 2022-06-20:
Ok, no problem at all. But after 7 days of inactivity, this issue will be closed. Is that a problem or can it remain on the radar of the devs (when they have the time)?

Mimiix, 2022-06-20:
It gets unstaled tonight :)

Smanar, 2022-06-20:
Bad news for you.
ATM there is a blocked PR to add tuya cluster to be used with DDF for TRV.
I m waiting for this one was validated before making same for covering.

So for the moment not possible adding new tuya covering.

vlude, 2022-06-22:
Thanks for the information, I will follow that one

closed 2022-07-22 by github-actions[bot]