dresden-elektronik/deconz-rest-plugin #6093: LK Wiser outlet does not create energy sensors

DurgNomis-drol, 2022-05-30:
## Describe the bug

My LK Wiser outlet (LK/OUTLET/1) by Schneider Electric does not create energy sensors when added to deconz hub. Looking at the the recent PR's, it looks like it should work, but I am confused as to why they are not created. It can toggle it and everything, but no energy data. Am I missing something. I have tried to remove and re-add it a couple of times without any success.

## Steps to reproduce the behavior

Add the device, and then choose API Information, select sensors and then see that there is no ZHAPower and ZHAConsumption sensor for this device.

## Expected behavior

An insane amount of energy data XD

## Screenshots

[image]

[image]

[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS) RPI 4B 4GB
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine) Docker container
 - Firmware version: (26xxyy00) 26660700
 - deCONZ version: (2.xx.yy) 2.16.01
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II) Conbee II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Aeotec Zwave 5 dongle

## deCONZ Logs

## Additional context

Mimiix, 2022-05-30:
Can you re pair it and see again? Perhaps when there's a load on it.

DurgNomis-drol, 2022-05-30:
Just tried it again, and the same result as before. No energy data and no sensors created in webUI

Mimiix, 2022-05-30:
Can you share a screen of the basic cluster?

Mimiix, 2022-05-30:
Also, can you read the power clusters you already screenshotted but then when there's a load on it?

DurgNomis-drol, 2022-05-30:
@Mimiix The values didn't not change until I did what @Smanar suggested

@Smanar It worked!

Thanks for the help! 😃

closed 2022-05-30 by DurgNomis-drol

Mimiix, 2022-05-30:
In that case, you should change it back to bronze and enable the bronze setting.

DurgNomis-drol, 2022-05-30:
@Mimiix How do you enable the bronze setting?

DurgNomis-drol, 2022-05-30:
It seems to have reset itself to bronze anyway

Mimiix, 2022-05-30:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-change-DDF-modes-and-enable-or-disable-DDF-levels