dresden-elektronik/deconz-rest-plugin #6099: Sengled outlet (E1C-NB7) reported cumulative power consumption incorrect

kimmeld, 2022-06-01:
## Describe the bug

The cumulative consumption values reported via the REST API are overstated by a factor of 10.

## Steps to reproduce the behavior

Using a Sengled E1C-NB7, hook up a known load and watch the cumulative power consumption values.  They will be 10x what they should be.

## Expected behavior

Correct cumulative power consumption values are reported.

## Screenshots

[image]

This value is correct - 21074 / 10000 = 2.1074 kWh, however it is reported as 21.074 kWh to Home Assistant via the REST API.  I think that something expects this to be reported in Watts, but it's actually reported in Watts/10.

I've worked around this by adding the following to devices/sengled/e1c-nb7.json:

```
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "eval": "if (Attr.val != -32768) { Item.val = Attr.val / 10; }"
          }
        },
```
(added the `read` and `parse` pieces, copied from the code that does the same to the state/power attribute)

## Environment

 - Host system: PC
 - Running method: Windows 10 Pro
 - Firmware version: (26580700)
 - deCONZ version: (2.16.01)
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?  No.

## deCONZ Logs

## Additional context

kimmeld, 2022-06-30:
Reopen this please, the issue has not been resolved as far as I know.

kimmeld, 2022-06-30:
Hey @Mimiix,

That's correct, I've modified the DDF and placed the modified version in ....\AppData\Local\dresden-elektronik\deCONZ\devices so that it doesn't get overwritten during upgrades.

I've attached my modified DDF:

[e1c-nb7.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9021093/e1c-nb7.txt)

Thanks,
-- Dave

Mimiix, 2022-06-30:
In that case, I suggest you do a pr to this repository and we can close this one :)

kimmeld, 2022-06-30:
Sounds good, I’ll try to throw something together next week.

SwoopX, 2022-07-15:
Closing this one as the respective PR got already merged.

closed 2022-07-15 by SwoopX