dresden-elektronik/deconz-rest-plugin #6100: IH-K009

EvertJob, 2022-06-02:
## Device

- Product name: IH-K009
- Manufacturer: Unknown
- Model identifier: Unknown
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

### Basic
[image]
[image]

### Identify
[image]

### Device Temperature
[image]
[image]

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-06-02:
Hello, your device relay look standard, can do like this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6034

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_dowj6gyi",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

gforums, 2022-06-03:
Dear all, good morning,

hope this is not off topic, but if anyone could help me here or by direct contact, I would much appreciate.

I bought such a tuya TS0201 device with the name "_TZ3000_dowj6gyi".

I am new to all that DDF file stuff and gave it a try. Usually using a rasperry with openhabian, I wasn't able to get the device linked. I tried to copy the ts0201.json file to /usr/share/deCONZ/devices but wasn't able to link the device.

Now I put the conbee II stick to my windows device, used the DDF file editor to load the template, I activated permit join and afterwards the device was successfully shown in the phoson app incl. temp/hum values.

My question now is - what can I do within my rasperry device to also get the device connected?
Any short howto/steps I can try?

Thank you and best regards,
Christian

EvertJob, 2022-06-03:
> Hello, your device relay look standard, can do like this issue #6034
> 
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZ3000_dowj6gyi",
>   "modelid": "TS0201",
>   "vendor": "Tuya",
>   "product": "IH-K009",
>   "sleeper": true,
>   "status": "Gold",
>   "subdevices": [
>     {
>       "type": "$TYPE_TEMPERATURE_SENSOR",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0402"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/battery",
>           "awake": true,
>           "parse": {
>             "at": "0x0021",
>             "cl": "0x0001",
>             "ep": 1,
>             "eval": "Item.val = Attr.val / 2;",
>             "fn": "zcl"
>           }
>         },
>         {
>           "name": "config/offset"
>         },
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/lastupdated"
>         },
>         {
>           "name": "state/temperature",
>           "awake": true
>         }
>       ]
>     },
>     {
>       "type": "$TYPE_HUMIDITY_SENSOR",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0405"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/battery",
>           "awake": true,
>           "parse": {
>             "at": "0x0021",
>             "cl": "0x0001",
>             "ep": 1,
>             "eval": "Item.val = Attr.val / 2;",
>             "fn": "zcl"
>           }
>         },
>         {
>           "name": "config/offset"
>         },
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/humidity",
>           "awake": true
>         },
>         {
>           "name": "state/lastupdated"
>         }
>       ]
>     }
>   ],
>   "bindings": [
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0001",
>       "report": [
>         {
>           "at": "0x0021",
>           "dt": "0x20",
>           "min": 60,
>           "max": 3600,
>           "change": "0x00000001"
>         }
>       ]
>     },
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0402",
>       "report": [
>         {
>           "at": "0x0000",
>           "dt": "0x29",
>           "min": 60,
>           "max": 300,
>           "change": "0x00000064"
>         }
>       ]
>     },
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0405",
>       "report": [
>         {
>           "at": "0x0000",
>           "dt": "0x21",
>           "min": 60,
>           "max": 300,
>           "change": "0x00000064"
>         }
>       ]
>     }
>   ]
> }
> ```

I've deleted the node, import the DDF file to the Tuya folder, restart deconz, add the sensor but nothing is happening.

gforums, 2022-06-10:
> @gforums if it have worked on windows no reason for it don't work on raspian with the same method
> 
> First thing to check, the "devices" folder, check the contain, if it s empty, it s a wrong one. Them check the DDF status and set it to "Gold" to be sure it will be used. If you have the GUI, select the node them "edit DDF", you need to see on the DDF editor window title the same path and file name. You can see some usefull logs with "info" "info_l2" and "DDF" It's possible there is a typo in the DDF too, I m using a Test editor to make them, if the file have typo you will have error message when trying to load them using the editor. Check your deconz version too.
> 
> If you are using the editor, there is an option "hot relaod".
> 
> If you are using HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Thanks Smanar!
Finally found some time and installed VNC plus deconz on my rasperry. That way I could enter DDF editor, copy_paste same file as I did within windows and now having the sensor connected succesfully with Phoscon :)

Will see if it works stable.

Again thanks a lot!
Christian

Smanar, 2022-06-11:
You mean the temperature never change ?

gforums, 2022-06-11:
> You mean the temperature never change ?

Exactly. Read a little bit of may be Osram plugs may cause this? This is where I wanted to gather more experience.
I just received one value after pairing, that's it.

Smanar, 2022-06-11:
Osram can have issue with Xiaomi/aqara but tuya ....
You still have access to GUI ? to see if the device is still connected ?
You haven't return at all (temp, battaery, humidity) ? what is the "lastseen" date ?

gforums, 2022-06-11:
I tried removing and pairing again.
Now having sleeper=false in the DDF, that was on false previously.

Currently it's updating since roughly 4 hours...maybe it's solved for now? :)

Smanar, 2022-06-11:
>Now having sleeper=false in the DDF

Are you sure it s the good DDF ? Because on the file you have 
```
  "product": "IH-K009",
  "sleeper": true,
```

But yes, when you re-include it, deconz remake all bind, so can solve the issue.

gforums, 2022-06-11:
I had true in the DDF file, yes.
But for some reason today as I paired the device, it was described in the
file with the value = false.

But I am a bloody beginner so I am just trialing and erroring :D

Will let you know here whether it stays working.

Smanar, 2022-06-12:
Sure, and tell me if I need to make the PR to add it officialy.

gforums, 2022-06-16:
> Sure, and tell me if I need to make the PR to add it officialy.

It was a little bit tricky at the beginning as I couldn't get the tuya really into the mesh network. It didn't want to connect to any of the router devices and therefore I needed to relocate the raspi with conbee a little bit.

But now since 3 days the sensor works and updates values actively.

Happy if you could bring it officially to one of the next updates.

The final ddf file that works for me:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_dowj6gyi",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "IH-K009",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}

Best regards and thanks a lot!

Christian

Smanar, 2022-06-16:
DDF added, thx.

closed 2022-07-06 by manup