dresden-elektronik/deconz-rest-plugin #6102: Paulmann Zigbee Switch 501.34

carstenph1, 2022-06-02:
## Device

- Product name:  Zigbee Wandschalter On/Off/Dimm 
- Manufacturer: Paulman LichtGmbH
- Model identifier: 501.34
- Device type : Switch

## Screenshots

### Basic

[image]

### Node Info

[image]

Mimiix, 2022-06-03:
Missing screenshots.

carstenph1, 2022-06-03:
I added the complete available information on cluster information.

carstenph1, 2022-06-14:
@Mimiix are you missing any further details?

Mimiix, 2022-06-14:
Woops, sorry! Missed your comment. Looks good :)

LilYea, 2022-07-07:
Is there progress yet?

carstenph1, 2022-08-01:
it would be awesome to see this feature in one of the next releases :-)

carstenph1, 2022-08-28:
pushing it again!

Artur-Fedjukevits, 2022-09-01:
I have the same switch. I would also be very grateful to see this device as supported.

carstenph1, 2022-09-26:
@Mimiix is it still planned to include this switch compatibility in one of the next releases?

Mimiix, 2022-09-26:
A dev needs to pick it up. Perhaps @smanar or @SwoopX  wants to help out. Otherwise, we are kinda stuck...

Smanar, 2022-09-27:
>Product name: Zigbee Wandschalter On/Off/Dimm

So this device don't have color ?

carstenph1, 2022-09-27:
no. its a switch to turn on/off/dim the light bulbs. there is nothing available to control the color.

Smanar, 2022-09-27:
So can try with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Paulman LichtGmbH",
  "modelid": "501.34",
  "product": "Paulmann Zigbee Switch",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008"
    },
  ]
}
```

With it the device will be reconized but not working.
Just enable deconz log with "info_l2" in the application / help / debug view
Press all buttons and pls share the result, you will error like

```
	Line 4570:         DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",
	Line 5607:         DBG_Printf(DBG_INFO_L2, "[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",
```

We will use them to edit another json file.

carstenph1, 2022-10-01:
thanks Smanar. 
I hope I made the import correct. I had to remove a comma in the json, otherwise it did not load.
I imported the json in the DDF Editor of Deconz and ran the debug view with info_l2. but I cant see any output of "no button ...". just "APS-DATA.indication from child 0x32E3" every time I clicked the switch buttons.

here you can find the full debug log
https://pastebin.com/Z9ayZkyy

Smanar, 2022-10-02:
Ha yes, have removed the comma too.

But it s strange there is still nothing. The node have changed from 0xXXXX to a better name ? To check if the device is now in the API.
You have made a "hot reload" (in the editor menu) after having modified the DDF ?

Else can enable log "aps" to see what is this aps request.

carstenph1, 2022-10-03:
well I was playing around in the Editor and did a hot reload without knowing what it does.
the device ID is still the same like at the beginngin 0x32E3.

enabling the API logs while pressing the switch, dont seem to bring more light in the dark....
https://pastebin.com/ZubC12R1

carstenph1, 2022-10-03:
mhh you are right, there was a typo in the manufacturername. but nothing changed right now...
https://pastebin.com/nDdZYCUx

carstenph1, 2022-10-03:
ok. good news, I restarted deconz as the DDF editor still had the manufacterer Description still wrong even after correcting it.
now I see the name "Switch 4" as a DB entry. please have a look

https://pastebin.com/yL7EPR2A

Smanar, 2022-10-03:
```
18:20:26:992 [INFO] - No button map for: 501.34, broadcast to: 0xE311, endpoint: 0x02, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 74
18:20:27:598 [INFO] - No button map for: 501.34, broadcast to: 0xE311, endpoint: 0x02, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 75
18:22:29:825 [INFO] - No button map for: 501.34, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 77
```

So with thoses logs, in the folder "devices" (probably where you have found the DDF) , you can found a file button_maps.json.
Just edit it to add

```
        "PaulmannMap": {
            "vendor": "Paulmann LichtGmbH",
            "doc": "On off remote",
            "modelids": ["501.34"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", " B1 On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 Off"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2 On"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2O ff"]
            ]
        },
```

If you have more logs (with long press for exemple), pls share them too.

carstenph1, 2022-10-03:
ok I see the switch in the phoscon web gui. but I cant edit the switch or connect it to a group of lights. I added the lines you mentioned in the /usr/share/deCONZ/devices/button_maps.json
is this intended?

I will share more logs (long press for dimm) tomorrow.

Smanar, 2022-10-03:
>I cant edit the switch

You mean in phoscon ? So I can't do something, I haven't impact on phoscon. But you can use it in third app, every time you use the device you will have a new notification with state/buttonevent.

>connect it to a group of lights

You mean using the old web app ? To use "direct bind" without automation, using groupcast features ? For this I can do something, need to add the config/group to the DDF

>is this intended?

Yep it s the good one, you will, you will have another message in log now, visible with "info" or "info_l2", not an error one.

carstenph1, 2022-10-11:
Good Morning, 
I have new logs from today using long press button.
https://pastebin.com/kmFNvJtB

do you have any clue why I cant discover the second Paulmann Switch (same model)?

thanks
Carsten

Smanar, 2022-10-11:
So new button map to try

```
        "PaulmannMap": {
            "vendor": "Paulmann LichtGmbH",
            "doc": "On off remote",
            "modelids": ["501.34"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", " B1"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "B5"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "B6"]
            ]
        },
```

But I have some question.
- On your logs you have a second endpoint, but on your capture you have only the endpoint "01" (it's the number on the node title)
- I don't know how look visualy your device, so I m using B1 > B6, but you can have B6 = B1 long press, if you can correct me ?
- I m not sure you will have all action, only 1 endpoint used, will found the missing by elimination.

baaaron, 2022-10-27:
Bought the same switch, just now I realized it isn‘t supported yet :( 
Let me know if anything’s still needed to get it supported

Smanar, 2022-10-28:
Sure can use the same DDF and the button map from the previous post.
Press button and tell me the result.

And BTW if you can confirm me how many endpoint have this device ?

carstenph1, 2022-11-02:
Hello Smanar,
the switch can control two separate endpoints/devices. each side on/off (short press) or dimm brighter/darker (long press). with your current configuration the button map.json I have 6 buttons visible in phoscon html gui.

is this the information you required?

[image]

Smanar, 2022-11-03:
>I have 6 buttons visible in phoscon html gui.

This is "phoscon magic" I haven't impact on it ^^

But I think I miss at least 2 buttons return, as only the on/off use the 2 endpoint

```
        "PaulmannMap": {
            "vendor": "Paulmann LichtGmbH",
            "doc": "On off remote",
            "modelids": ["501.34"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", " B1"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "B5"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "B6"]
            ]
        }
```

You can see button return in Phoscon/help/API Information/events, and I think you will miss 2 of them.

closed 2022-12-09 by github-actions[bot]

Millhouse273, 2022-12-22:
Hi,
I call this button my own and are very thankful for your great work here. After some problems I got it working and can link the buttons to my lights. There is only one thing missing and I didn't figured out fixing it by working me through the button_maps.json. I tried to adopt the dim function, but haven't enough skills to transfer it to the paulmann button.

Now the missing function: I'd like to dimm down with the off-button. dim up is possible with the posted map and the on buttons. Thanks for any response

Millhouse273, 2022-12-22:
Is it possible to get a real dim button? Like the Ikea on/off button with long press controlling the level constantly? Like mentioned before I have no idea off programming all the stuff...

Smanar, 2022-12-23:
Hello, first you are asking with phoscon or third app ?
From that you are saying you are using direct link with remote and light ?

You can make a long press on a button and check logs if the device support this action, with info_l2

```
	Line 4570:         DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",
	Line 5607:         DBG_Printf(DBG_INFO_L2, "[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",
```

Millhouse273, 2023-01-02:
Hi, thx for your answer.

I'm using the web interface of phoscon. The light is grouped in phoscon an the switch is grouped within this group. I can set an action like on or off to the buttons.
Their is no automation in any other app or phoscon.

I check, if the switch supports long press. In the description is mentioned on/off/dimm...

Thanks a lot for your help so far

Smanar, 2023-01-03:
Ok so you are not using automation but direct zigbee connexion.

Nothing to programme, the device support it or not ^^. Ha have the good cluster and support group feature

Just try this DDF to enable the group feature on the DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Paulman LichtGmbH",
  "modelid": "501.34",
  "product": "Paulmann Zigbee Switch",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "config/battery"
        },
        {
            "name": "config/group",
            "default": "auto"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
        "bind": "groupcast",
        "src.ep": 1,
        "cl": "0x0006",
        "config.group": 0
    },
    {
        "bind": "groupcast",
        "src.ep": 1,
        "cl": "0x0008",
        "config.group": 0
    }
  ]
}
```

IDK if phoscon will support it natively but the old webapp (in phoscon /help ) will support it.

Millhouse273, 2023-01-04:
Hi,

this is the log from l2.

{"id":"106","lastannounced":"2022-12-31T14:13:04Z","lastseen":"2023-01-04T19:07Z","manufacturername":"Paulmann LichtGmbH","modelid":"501.34","name":"Switch 106","swversion":"2.7.6_r25","type":"ZHASwitch","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"},"e":"changed","id":"106","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"} (ret = 2125021208)
20:07:36:681 Websocket 192.168.: send message: {"e":"changed","id":"106","r":"sensors","state":{"buttonevent":7002,"lastupdated":"2023-01-04T19:07:36.676"},"t":"event","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"} (ret = 2125021208)
20:07:36:682 Websocket 192.168.: send message: {"e":"changed","id":"106","r":"sensors","state":{"buttonevent":7002,"lastupdated":"2023-01-04T19:07:36.676"},"t":"event","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"} (ret = 2125021208)
20:07:36:686 Websocket 192.168.: send message: {"e":"changed","id":"16","r":"lights","state":{"alert":null,"bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"00:17:88:01:08:a4:01:8d-0b"} (ret = 150)
______________

{"id":"106","lastannounced":"2022-12-31T14:13:04Z","lastseen":"2023-01-04T19:09Z","manufacturername":"Paulmann LichtGmbH","modelid":"501.34","name":"Switch 106","swversion":"2.7.6_r25","type":"ZHASwitch","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"},"e":"changed","id":"106","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"} (ret = 2125021208)
20:09:22:369 Websocket 192.168.: send message: {"attr":{"id":"106","lastannounced":"2022-12-31T14:13:04Z","lastseen":"2023-01-04T19:09Z","manufacturername":"Paulmann LichtGmbH","modelid":"501.34","name":"Switch 106","swversion":"2.7.6_r25","type":"ZHASwitch","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"},"e":"changed","id":"106","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:14:0d:a5-01-0006"} (ret = 2125021208)

Long press on the off Button did no log. Don't know why...

Smanar, 2023-01-05:
`{"buttonevent":7002`

??? The button map go only to button 6 maximum

```
        "PaulmannMap": {
            "vendor": "Paulmann LichtGmbH",
            "doc": "On off remote",
            "modelids": ["501.34"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", " B1"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "B5"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "B6"]
            ]
```

>Long press on the off Button did no log. Don't know why...

Generaly this button don't have long press, its on or off, no utility to use a long press.

But on your side you want to use direct link, so you will not need buttonevent return, just need to check if the device create a "switch group" to be able to put light on it.

NurNano, 2023-07-03:
Hey, 
thank you for your work. I looked over it with my new Switch and found out, some Keybindings were missing. 
I did some try/error and optimized the button_maps.

Here for you to test:

`"PaulmannMap": {
            "vendor": "Paulmann LichtGmbH",
            "doc": "On off remote",
            "modelids": ["501.34"],
			"buttons": [
              {"S_BUTTON_1": "ON Links"},
              {"S_BUTTON_2": "OFF Links"},
			  {"S_BUTTON_3": "ON Rechts"},
              {"S_BUTTON_4": "OFF Rechts"}
            ],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "ON Links"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "OFF Links"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "ON Rechts"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "OFF Rechts"],
				[1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim Up Links"],
		        [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Stop Dim"],
				[1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim Down Links"],
		        [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Stop Dim"],
				[1, "0x02", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim Up Rechts"],
				[1, "0x02", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Stop Dim"],
				[1, "0x02", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim Down Rechts"],
                [1, "0x02", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Stop Dim"]
			]
		}`

With this i got all Keys recognized.

Smanar, 2023-07-03:
>After some testing i got stuck in my openhab and it seems to be needed to add some more Buttons to access all of the Keys in Openhab due to only get the Pressed Button ID delivered.

Ha nice to know, I never use the field "buttons" on my side.

So you have finished the device inclusion ? do you want to make the PR to add it ?

NurNano, 2023-07-04:
I really don't know how this things work. Just made the ddf working and liked to share the results with the community. If anyone is able to make the PR so i really would like it.

Smanar, 2023-07-04:
I have made it here  https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7087

If you can confirm all is fine ? (long time we have started this device)

Smanar, 2023-07-05:
Nice, i's same DDF (except you still have the typo in the manufacture name, you still missing the "n".

I remove the draft state on the PR

NurNano, 2023-07-06:
Wait a sec....I was going to change the typo and opened the DDF, finding out my DDF looks different.

Here is my DDF. 
Seems like I changed something while testing with my Windows PC and switching to Linux headless.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Paulmann LichtGmbH",
  "modelid": "501.34",
  "product": "501.34",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008"
    }
  ]
}
```

Smanar, 2023-07-06:
So there is no big change.
The only one is the "slepper" state, you have changed it to "false", but I m sure at 99% all battery remote are sleeper.

AndreasHauber, 2024-09-30:
Hello,
I bought some new 501.34 as the old one was working perfectly. The new ones have a different firmware version and are not shown after pairing. Here are my readings:

Basic 
[image]
Power Configuration
[image]
Node Info
[image]

NurNano, 2024-09-30:
Seems like they just added a space in the manufacturer filed between the Licht and GmbH.
Can you try to copy the 501_34_battery_switch.json to 501_34_battery_switch_new.json and add the space in the "manufacturername" field?
Then reload the Deconz and try to add it again. It should now be there like the old switches.
If it works the button_maps.json has to be edited too.

NurNano, 2024-09-30:
@Smanar can u do an other commit and add a second json and change the button_maps.json by just copy and pasting and adding the space?
That would help all future users of the new Buttons :)

Smanar, 2024-09-30:
Sure https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7952
I haven't modified the buttonmap file as deconz use only the ModelID for it.