dresden-elektronik/deconz-rest-plugin #6107: SilcerCrest Smart Button (LIDL) _TZ3000_rco1yzb1 TS004F

markusmauch, 2022-06-04:
## Device

- Product name: SilcerCrest Smart Button (LIDL)
- Manufacturer: _TZ3000_rco1yzb1
- Model identifier: TS004F
- Device type : Switch
 
## Screenshots

[image]
[image]
[image]
[image]

The device shows up as a connected switch but is not available in the device editor.

steff75, 2022-06-07:
I also got this today. Could it possibly be the same as this one, just in a different case?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6031

Anyway, I paired it in with Phoscon and it creates a switch and a light. When I press the button, all the lights that exist in Phoscon are turned on without me even linking a light.

oli737, 2022-06-20:
Same Problem here. All devices turning on or off on button press.

LarschOakheart, 2022-06-21:
jep i have the same problem with two of these switches

oli737, 2022-06-21:
It does not only switch lights on all at once.
Also electric outlets switch on too on button press.

traschke, 2022-06-23:
Just a little feedback while I was playing around with these switches (it's my first time I got into adding own DDFs):

I am using the docker container *deconzcommunity/deconz* on armv7.
I mounted the DDF to `/usr/share/deCONZ/devices/tuya/_TZ3000_rco1yzb1_switch.json` and mounted a modified `button_maps.json` to `/usr/share/deCONZ/devices/button_maps.json`, where `_TZ3000_rco1yzb1` was added to the `modelids` array of the `Tuya3gangMap` object.

Afterwards I restarted the deCONZ container and added the first switch (out of 3) via the Phoscon WebUI. It seemed to work and surprisingly the switch didn't turn on/off all devices. I could also connect the switch to OpenHAB and see the `buttonevent` being updated.

I added the second switch and it behaved like others mentioned before, turning all devices on/off. The `buttonevent` also didn't update on the second one (not in openHAB and not in the REST API). After that, the first switch also stopped working (doing literally nothing, no turning on/off all devices and `buttonevent` also not updated anymore). I removed both switches and re-added them and they behaved like the second one did, controlling all devices and no `buttonevent` updates.

I always get 2 devices/sensors per switch in the REST API (from `GET api/<apikey>/sensors`):
```json
  "32": {
    "config": {
      "battery": null,
      "group": "65520",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "32c5f268b3f58453a3b9297e4a33bf83",
    "lastannounced": null,
    "lastseen": "2022-06-23T20:31Z",
    "manufacturername": "_TZ3000_rco1yzb1",
    "mode": 1,
    "modelid": "TS004F",
    "name": "TS004F",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "2c:11:65:ff:fe:c7:22:3c-01-1000"
  },
  "33": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "63eead54a5881bba7838b0f0157c286f",
    "lastannounced": null,
    "lastseen": "2022-06-23T20:31Z",
    "manufacturername": "_TZ3000_rco1yzb1",
    "mode": 1,
    "modelid": "TS004F",
    "name": "Switch 33",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "2c:11:65:ff:fe:c7:22:3c-01-0006"
  }
```
Hope this helps in some way. If something needs to be tested, I'm glad to help.

Smanar, 2022-06-24:
You have a "core mix" ....

`"uniqueid": "2c:11:65:ff:fe:c7:22:3c-01-1000"`

This one is created with the legacy code, and it use the group feature `"group": "65520",`

`"uniqueid": "2c:11:65:ff:fe:c7:22:3c-01-0006"`

This one is created by the DDF code and don't use the group feature.

I can't explain what happen :
- The first one is probably created before the DDF was added ?
- The first one can be created because the legacy code is too fast (If it s that, can be usefull, we have this issue on another device too)
- On your last try nothing is working, but I can add the group feature on the DDF if this kind of working mode is better for you (can disable broadcast request)
- Can check too if you don't have a bad entry created in "light" section.

traschke, 2022-06-26:
I had some more time to tinker around with the switch.

I'll try to answer your questions first:

> The first one is probably created before the DDF was added ?

I deliberately checked the Github Issues before I've done anything with the switch. I found this issue and added your DDF and button mappings before adding the device to my network. Today I double checked if the files are in the correct locations and also checked their filesystem permissions. Seems fine on my side.

> The first one can be created because the legacy code is too fast (If it s that, can be usefull, we have this issue on another device too)

As I have no deep insights in the internal functionality of deCONZ, I cannot answer this.

> On your last try nothing is working, but I can add the group feature on the DDF if this kind of working mode is better for you (can disable broadcast request)

Same as above. I could never replicate this state. The only state I get now is the switch turning on all devices on single press and turning them off again on double press. There are still two switch devices present und `/sensors` and also a entry in the light section (See below)

> Can check too if you don't have a bad entry created in "light" section.

Today I started a "fresh" run. I made sure all switches are removed and only worked with one switch for now. I indeed discoved an odd looking entry in the Phoscon light section called *Dimmer switch 11*.

[image]

The REST API returns this:

```json
  "11": {
    "etag": "2d46612476b0191fd423f93a6aa82932",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-06-26T18:39Z",
    "manufacturername": "_TZ3000_rco1yzb1",
    "modelid": "TS004F",
    "name": "Dimmer switch 11",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Dimmer switch",
    "uniqueid": "2c:11:65:ff:fe:c7:22:3c-01"
  }
```

In the PhosconUI section for switches there is one entry called *TS004F*.

[image]

The REST API on the other hand still returns two entries und the `/sensors` endpoint:

```json
  "22": {
    "config": {
      "battery": null,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "57102ed75a55a901072f4f4850ecead1",
    "lastannounced": null,
    "lastseen": "2022-06-26T18:39Z",
    "manufacturername": "_TZ3000_rco1yzb1",
    "mode": 1,
    "modelid": "TS004F",
    "name": "TS004F",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
[... 259 lines cut ...]
20:38:20:238 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:20:238 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 6952 ms
20:38:20:239 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:20:239 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 6953 ms
20:38:20:670 saved node state in 0 ms
20:38:20:671 sync() in 0 ms
20:38:20:736 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:20:736 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 7450 ms
20:38:20:737 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:20:737 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 7451 ms
20:38:21:234 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:21:235 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 7948 ms
20:38:21:235 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:21:235 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 7949 ms
20:38:21:649 send permit join, duration: 0
20:38:21:734 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:21:735 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 8449 ms
20:38:21:735 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:21:736 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 8449 ms
20:38:21:751 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:21:752 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 8466 ms
20:38:21:752 MAC poll fastEnddeviceProbe() 0x2C1165FFFEC7223C
20:38:21:753 wait response fastEnddeviceProbe() 0x2C1165FFFEC7223C, elapsed 8466 ms
20:38:21:828 Search sensors done
```

markusmauch, 2022-06-26:
Maybe it's a bit off topic but I've switched to Zigbee2MQTT, which supports this button quite well as well as all my other devices that gave me a lot of trouble when using decONZ

Mimiix, 2022-06-26:
Reopened as development is still in progress.

Smanar, 2022-06-27:
I will give this link at Manup, he already have some idea for this issue.
There is the issue with the legacy code faster than the DDF core.

And we have
- 1 sensor created with DDF code
- 1 sensor created by Legacy code
- 1 light created by Legacy code

And ofc only the first one is needed.

nanniknr1, 2022-06-29:
The problem with switching ALL devices on/off is a known problem and already discussed here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/What-you-can-do-when-a-switch-turns-(almost)-all-devices-on-off

This solution worked for me.

Smanar, 2022-06-30:
But he have another issue.
The state/buttonevent is never updated, because of "fake" entry in the API. Only the first sensor need to be created.
Fo me it happen because the legacy code is faster than the DDF one and start to create its entry before the DDF one block it.

traschke, 2022-07-23:
@Smanar Are there any updates on the ddf/legacy code issue? I tried repairing the device with the latest update, but unfortunately it hasn't worked and still produced 3 items.

Smanar, 2022-07-23:
Yep https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6196
Will be on next one, but I have somes user with the same issue, and it work for them, something you can try is making the DDF code use the same unqueid than for the legacy code.
```

      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
```

With using 0x1000 instead of 0x0006, but better to wait for next version, I think it can solve so much issue.

traschke, 2022-08-15:
Just wanted to leave a qick update:
I tried the Smart Buttons with the latest beta release [v2.18.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/tree/v2.18.0-beta) and the pairing issue seems to be resolved. I only see one device under sensors with a UUID ending with `0006`, so it seems ok to me. I tried pairing 4 of these buttons and all paired flawlessly.

One thing I noticed upon setup is, that initially the button still somehow triggers the state of all devices (ON/OFF), but not actually turning them ON (single-press) or OFF (double-press). This means Phoscon reports that all lights were turned on in browser UI, but in fact the real light is still off (or on). In this state the button does not report any `buttonevent` via rest api. By accident I discovered, that by pressing the button 3 times it somehow switches the mode (?). After pressing the button 3 times, everything works as expected, no ghost turning on/off of lights in phoscon and `buttonevent` is populated correct with single and double press. Is there perhabs a mapping issue with 3 clicks? Honestly I have absolutely no clue how much actions this button supports (single, double, triple press?).

jjansen85, 2022-08-15:
I think you can compare it with the Moes smart knob _TZ3000_4fjiwweb. See manual page 2 and 12.
[Smart_knob_manual.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9339031/Smart_knob_manual.pdf)

traschke, 2022-08-15:
I searched up the manual online. Unfortunately it seems only to be available in German: [Manual (German)](https://stesbintegrationprod.blob.core.windows.net/public/articlemanual/9b920ca2-c4f8-478e-b5dc-30e44768af45.pdf).

It mentions the following (Page 13):

Functions:

| Button | Function |
| ------- | ---------- |
| Press 1x | Switch on |
| Press 2x (quick press) | Power off |
| Press and hold | Dim lighting devices up/down |

Additionally it mentions the following about the 3x press: (Page 14)

> Switching between remote dimmer and scene linking function
>
>1. Make sure the Smart Button is in remote dimmer mode. Quickly press the smart button 3x. The LED display flashes 3x if successful.
>NOTE: The button must be pressed 3x quickly and evenly.
>2. You can control the devices in scene link mode.
>3. To return to remote dimmer mode, quickly press the smart button 3x. The LED display flashes 3x if successful.

Smanar, 2022-08-16:
ha ok, it s the scene mode.
On the remote dimmer mode we are looking for 

```
            "map": [
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 short"],
                [1, "0x01", "ONOFF", "0xfd", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "B1 double"],
                [1, "0x01", "ONOFF", "0xfd", "2", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "B1 long"],
                [1, "0x02", "ONOFF", "0xfd", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2 short"],
                [1, "0x02", "ONOFF", "0xfd", "1", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "B2 double"],
                [1, "0x02", "ONOFF", "0xfd", "2", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "B2 long"],
                [1, "0x03", "ONOFF", "0xfd", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3 short"],
                [1, "0x03", "ONOFF", "0xfd", "1", "S_BUTTON_3", "S_BUTTON_ACTION_DOUBLE_PRESS", "B3 double"],
                [1, "0x03", "ONOFF", "0xfd", "2", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "B3 long"],
                [1, "0x04", "ONOFF", "0xfd", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4 short"],
                [1, "0x04", "ONOFF", "0xfd", "1", "S_BUTTON_4", "S_BUTTON_ACTION_DOUBLE_PRESS", "B4 double"],
                [1, "0x04", "ONOFF", "0xfd", "2", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "B4 long"]
            ]
```

So we are using only the onoff cluster, I think in the other mode it use the Scene cluster 0x0005, it s for that deconz is not detecting buttonevent.

traschke, 2022-08-18:
I still use the ddf and buttonmapping from earlier btw. The button mapping seems to me the same one as in "Tuya3gangaMap" as in your comment earlier (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6107#issuecomment-1162089492). Short and double press are working (buttonevent 1002 and 1004), long press does not emit a new buttonevent.

In Deconz I the switch looks like this:
[image]

Smanar, 2022-08-18:
>Short and double press are working

But you already have added the model id "_TZ3000_rco1yzb1" in the xml file for "Tuya3gangaMap" ? Else you can't have buttonevent at all.

If all is fine for missing buttonevent, you will have in logs an error message with all missing value to add in the xml file.

If you have buttonevent 1002 and 1004, it mean all event arent using the manufacture cluster

traschke, 2022-08-19:
> Nothing in logs with "info" and "info_l2" during the long press ?

No, not that I'm aware of. I only see log entries in Deconz Debug View with `INFO`, `INFO_L2`, `ERROR` and `ERROR_L2` for single and double press, e.g.:
```log
18:09:21:142 [INFO] - Button 1002 - TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 90
18:26:13:868 [INFO] - Button 1004 - TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 double, payload: 01, zclSeq: 103
```

I do see `No button map for:` entries for a different device (Smart Plug), so I assume I should have the correct log levels selected:

```log
18:27:38:755 [INFO] - No button map for: lumi.plug.maeu01, unicast to: 0x0000, endpoint: 0x01, cluster: XIAOMI (0xFCC0), command: ATTRIBUTE_REPORT (0x0A), payload: DC00410CC173922A9F50000083A4FFE9, zclSeq: 92
```

Would it be helpful to temporary remove all button mappings for `_TZ3000_rco1yzb1`, re-pair the button and check the log for all `No button map for: <devicename>`?

Smanar, 2022-08-19:
As the long press is the dimming, try adding the setlevel cluster binding

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008"
    },
```

traschke, 2022-08-20:
I don't know if I understand this right, but as far as I can tell this entry already exists in your proposed DDF from this comment: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6107#issuecomment-1162089492

Smanar, 2022-08-20:
Ha yes right.

I m seing "sleeper" need to be true, but it will not help use for this issue

traschke, 2022-09-06:
Yes, it's working for single and double press as long as I don't switch accidentally to the scene mode (3x press). Long press is the only issue. I've been using it for the last couple weeks with no further issues on single and double press.

Smanar, 2022-09-07:
Thx , have made a PR (some others users ask for this device)

Smanar, 2022-09-08:
BTW, I have added the missing battery support, if you test it ? 
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6304

traschke, 2022-09-08:
Battery seems to work after re-pairing the switch:
```json
"48": {
    "config": {
      "battery": 90,
      "on": true,
      "reachable": true
    },
    "etag": "1dd0a1451df6241cbf838bb31dcf9691",
    "lastannounced": null,
    "lastseen": "2022-09-08T16:50Z",
    "manufacturername": "_TZ3000_rco1yzb1",
    "mode": 1,
    "modelid": "TS004F",
    "name": "Switch Livingroom",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "2c:11:65:ff:fe:c7:22:3c-01-0006"
  }
```

Some comments above you mentioned that `sleeper` should be set to `true`, but in the PR it's set to `false`. Did you miss that in the PR by any chance?

Smanar, 2022-09-08:
Yep, I will correct it, better to use the good value.

Sting2saw, 2022-09-30:
I have only Battery entry after connecting it with phoscon Web App, anyone else has this Problem?
[image]

jeinwag, 2022-10-07:
> I have only Battery entry after connecting it with phoscon Web App, anyone else has this Problem?

There is no entity for the button, please read https://www.home-assistant.io/integrations/deconz/#remote-control-devices.

thealkly, 2022-10-09:
I am new to github, i assume if the PR is merged it will be release someday?
Right now the switch is displayed like in the screenshot of @Sting2saw 
Just bought a couple of switches as they are available in the store and compared to the shortcut button from ikea cheaper. 
Thanks!!

jeinwag, 2022-10-10:
@thealkly I guess it will be released with one of the next HA releases. Until then, you can still use the button with the deconz_event trigger as explained in the link I posted above.

FHEMUser, 2022-11-06:
Hello.
I tested this button and have the same problem. I used the DDF that will be posted here and my other two plug devices will switch on/off if i hit the button.

I have no file caled "buton_maps.json" so i create one and add the text to it.
No changes.

So i edit the group entry on cluster info from 0x0000 to 0x0002 an hit exec.
Then the button will no more switch the other two devices.
But is this the correct way ?

I use FHEM for home automation and here this button will not work if i edit the group.
I delete, use autocreate and so on.
I don´t know why, but now it will work on FHEM but the goup in the cluster is back to 0x0000

On/off will work, dimmer will not work.

Smanar, 2022-11-06:
Hello, what is your deconz version ?
The corrective is on the official, so you have nothing to do.

And I m not sure to understand what you are doing, the issue is about a switch, so you are puting light in the device group ? You are using phoscon or FHEM to do that ?

Switch group are special, and can't be deleted, you can add a light in a switch group, but the reverse will not work.

There is no dimmer feature on plugs ?

FHEMUser, 2022-11-07:
Hello.
Version
2.13.01 / 6.10.2021
26660700

I think my Version is to old. I can not update because my system is old too  (Stretch I think?)
So I must switch to an newer Raspbian Version.

I was thinking the group numbers are only numbers.

Good to hear that the new version will work.
Thanks for your response

FHEMUser, 2022-11-10:
On thing, please look on the title:
SilcerCrest Smart Button

The right Name is
SilverCrest

BooWseR, 2023-01-20:
Unfortunately, the "problem" with the different modes has not yet been fixed, or it has not been merged?!

I noticed that when I change the mode, a single press "turns on" all the lamps, a double press turns all the lamps "off" again. The lamps themselves do not react, the status only changes in Phoscon / deConz. 

I'm not sure about this comment: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6107#issuecomment-1216838881
Should i add this map? Or is it already?

And what's about this comment: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6107#issuecomment-1193132227
I see that this change did not make it into the PR. On purpose?

Smanar, 2023-01-20:
Hello, If I m right the device work only in one mode, and not the scene one.

About the first comment yep the code is here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/button_maps.json#L1293
But only for thoses devices

> 
"modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq"],

About the second comment, was because on older deconz version, the core create more than entry for this device. If I m right long time it was solved.