dresden-elektronik/deconz-rest-plugin #6112: Neo Siren Alarm - NAS-AB02

gmitch64, 2022-06-05:
## Device

- Product name: Neo Siren Alarm
- Manufacturer: _TZE200_t1blo2bj
- Model identifier: TS601
- Device type : Please remove all unrelated device types. 
  - Siren

## Screenshots

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

JonLaliberte, 2022-06-23:
Looks similar to [this one](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5248), where modifying the DDF may work. But I'm entirely out of my element here.

Smanar, 2022-06-23:
I don't think.
It miss the cluster list, but from the capture I think this device use the tuya cluster instead of classic cluster.
For siren we use state/alert and this field is not compatible yet with tuya cluster.

gmitch64, 2022-07-23:
Can we have this reopened please?

mateuszroth, 2022-07-28:
@Smanar  same issue here. Seems like the options are not writable, can't turn on the siren at all. Let me know how can I help.
[image]
[image]

Smanar, 2022-07-28:
I haven't the cluster list (the blue number when you select the second circle in the node title)
But from I remember this device use the tuya cluster, and it s not yet possible to use state/alert for DDF (with the tuya cluster)

mateuszroth, 2022-07-28:
@Smanar  there is the list, Tuya is on the list:
[image]

Do you know how many weeks/months will last until use of state/alert for DDF with tuya cluster is possible?

Smanar, 2022-07-28:
Here you have a PR to use them for a TRV https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054
And I don't want to start to do same for other item before this one will be finished.

JonLaliberte, 2022-09-13:
Device still isn’t supported. Should be left open.

gmitch64, 2022-10-06:
Any update on this? I'd like to keep it open please.

Spyder2005, 2022-12-16:
Can we open the topic again. I am trying to integrate exactly this sensor and unfortunately it is not working. It would be great if someone here could take care of it. Thank you!

Spyder2005, 2022-12-16:
Unfortunately, I am completely new to the world of DDF, or to put it better. I have only been involved with deconz/phoscon for a few days. Is there a tutorial on how I can provide you with a DDF?

Spyder2005, 2022-12-17:
Sorry stupid question. Here is the required DDF file. Thanks in advance!!
[_TZE200_t1blo2bj_ts0601.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10251688/_TZE200_t1blo2bj_ts0601.zip)

Spyder2005, 2022-12-17:
@Mimiix : Sorry stupid question, but you don't mean just save as in the DDF editor? The json file looks relatively empty and contains hardly any entries.

Smanar, 2022-12-17:
I think you can have return with DDF, but you will be not able to trigger the siren.

Spyder2005, 2022-12-21:
Isn't that the main task of a siren? I have tried something with DDF, but unfortunately the device cannot be controlled. Is there anything else needed for the siren to be added and supported?

Smanar, 2022-12-21:
>Isn't that the main task of a siren?

Yes it is ^^
But I know some device where the siren work only if the device detect something, you have setting but can trigger the siren.

For this one the problem is from deconz, it's possible to trigger the siren, but it's not yet possible with DDF.

kulmegil, 2023-01-19:
*Bump*
I have one so I'm joining the waiting queue...

gmitch64, 2023-01-20:
I obviously have one, so I'm still here too :)

gmitch64, 2023-02-11:
Bump to keep this issue open.

gmitch64, 2023-03-06:
Can we keep this open please.

LukeDefeo, 2023-03-12:
Plus one from me, let me know if I can help in any way.

gmitch64, 2023-04-03:
Please keep open

BabaIsYou, 2023-04-05:
Well, I don't know if what I propose has sense but will try to gain a minimum of control on this device, even if not completely attach in legacy code.

I based my detection for tuya DPIDs on https://github.com/blakadder/zigbee/blob/6e80b114b0e21779b531450532709d4d566ca30a/_zigbee/Neo_NAS-AB02B2.md?plain=1#L8 and https://github.com/zigpy/zha-device-handlers/blob/db3288dd0387331178b6b7d02a5558424aee6124/zhaquirks/tuya/ts0601_siren.py#L358

As we don't have a `volume` or `melody` attribute, I "reuse" for that purpose, respectively, `controlsequence` and `mode`. I know that it's not the best way but that's to test for the moment.

Don't really know if it's also clear to have a `/light` and a `/sensor` ... wil  se later.

If someone wnat to test this DDF : 
```
{
	"schema": "devcap1.schema.json",
	"manufacturername": [
		"_TZE200_t1blo2bj",
		"_TZE204_t1blo2bj"
	],
	"modelid": [
		"TS0601",
		"TS0601"
	],
	"product": "Neo Smart siren",
	"sleeper": true,
	"status": "Silver",
	"subdevices": [{
			"type": "$TYPE_WARNING_DEVICE",
			"restapi": "/lights",
			"uuid": [
				"$address.ext",
				"0x01"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "state/alert",
					"default": "none"
				},
				{
					"name": "state/reachable"
				}
			]
		},
		{
			"type": "ZHAAlarm",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0500"
			],
			"fingerprint": {
				"profile": "0x0104",
				"device": "0x0403",
				"endpoint": "0x01",
				"in": [
					"0x0000",
					"0x0004",
					"0x0005",
					"0xEF00"
				]
			},
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/enrolled",
					"public": false
				},
				{
					"name": "config/on",
					"description": "On/Off Alarm",
					"write": {
						"dpid": 13,
						"dt": "0x10",
						"eval": "Item.val == 1 ? 1 : 0;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 13,
						"eval": "Item.val == 1 ? 0 : 1;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/pending"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/alarm"
				},
				{
					"name": "config/battery",
					"parse": {
						"fn": "tuya",
						"dpid": 15,
						"eval": "Item.val = Attr.val;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "config/duration",
					"description": "Alarm duration in seconds",
					"write": {
						"dpid": 7,
						"dt": "0x2b",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 7,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "config/mode",
					"description": "Melody used by siren (between 1 to 18)",
					"range": [0, 18],
					"write": {
						"dpid": 21,
						"dt": "0x30",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 21,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "config/controlsequence",
					"description": "Volume of the siren, 0:low, 1:medium, 2:High",
					"range": [0, 2],
					"write": {
						"dpid": 5,
						"dt": "0x30",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 5,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/lastupdated"
				}
			]
		}
	],
	"bindings": [{
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0xEF00"
	}]
}
```

May be logic for `config/on` has to be inversed, and may be binding is not necessary.

BabaIsYou, 2023-05-02:
I was wrong, there are a `config/volume` and a `config/melody` items !
For sure I'll make a different DDF in the future ;-)

mateuszroth, 2023-05-29:
@BabaIsYou  noob question but how to change the melody from Deconz UI?

BabaIsYou, 2023-06-02:
Most part of your modified DDF sensor won't work because don't relay on Tuya cluster, or make reference to sensors (temp and hum) that are non existent on this device AFAIK !

So begin by this DDF, with same disclaimer that I made above : 
```
{
	"schema": "devcap1.schema.json",
	"manufacturername": [
		"_TZE200_t1blo2bj",
		"_TZE204_t1blo2bj"
	],
	"modelid": [
		"TS0601",
		"TS0601"
	],
	"product": "Neo Smart siren",
	"sleeper": true,
	"status": "Silver",
	"subdevices": [{
			"type": "$TYPE_WARNING_DEVICE",
			"restapi": "/lights",
			"uuid": [
				"$address.ext",
				"0x01"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "state/alert",
					"default": "none"
				},
				{
					"name": "state/reachable"
				}
			]
		},
		{
			"type": "ZHAAlarm",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0500"
			],
			"fingerprint": {
				"profile": "0x0104",
				"device": "0x0403",
				"endpoint": "0x01",
				"in": [
					"0x0000",
					"0x0004",
					"0x0005",
					"0xEF00"
				]
			},
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/enrolled",
					"public": false
				},
				{
					"name": "config/on",
					"description": "On/Off Alarm",
					"write": {
						"dpid": 13,
						"dt": "0x10",
						"eval": "Item.val == 1 ? 1 : 0;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 13,
						"eval": "Item.val == 1 ? 0 : 1;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/pending"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/alarm"
				},
				{
					"name": "config/battery",
					"parse": {
						"fn": "tuya",
						"dpid": 15,
						"eval": "Item.val = Attr.val;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "config/duration",
					"description": "Alarm duration in seconds",
					"write": {
						"dpid": 7,
						"dt": "0x2b",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 7,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "config/melody",
					"description": "Melody used by siren (between 1 to 18)",
					"range": [0, 18],
					"write": {
						"dpid": 21,
						"dt": "0x30",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 21,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "config/volume",
					"description": "Volume of the siren, 0:low, 1:medium, 2:High",
					"range": [0, 2],
					"write": {
						"dpid": 5,
						"dt": "0x30",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 5,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/lastupdated"
				}
			]
		}
	],
	"bindings": [{
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0xEF00"
	}]
}
```

closed 2023-07-01 by github-actions[bot]

theflyer08, 2023-07-20:
I tried the DDF from @BabaIsYou. It did not work for me. I have the Neo "_TZE204_t1blo2bj", "TS0601". The sirene has been recognized as light. But it does nothing. Even via IoBroker deconz-Adapter. 

I would be great to keep this alive.

BabaIsYou, 2023-08-23:
How does the device appears from REST-API point of view ?

theflyer08, 2023-08-23:
How can I get the rest API appearance on a headless Linux system?

BabaIsYou, 2023-08-23:
Using Phoscon help/information/api ?

theflyer08, 2023-08-24:
In Phoscon it looks like this:
[image]
In IOBroker I see the device in objects. However, I can't control the device by setting the values.

str31ok, 2023-08-24:
> How does the device appears from REST-API point of view ?

In the REST-API there is only one device
[image]

But i don't see it in the "lights" in phoscon
[image]

And when i open "plugs" tab - it is empty
[image]
 I am not sure, but think this happened after adding siren

As deconz shows it in this way
[image]
there is a DDF file for this siren in device folder, am i right? But i can't find it

BabaIsYou, 2023-08-24:
> In Phoscon it looks like this:
> [image]
> In IOBroker I see the device in objects. However, I can't control the device by setting the values.

What does REST-API see ? Into Phoscon choose help and API nformation

BabaIsYou, 2023-08-24:
> there is a DDF file for this siren in device folder, am i right? But i can't find it
Did you put the proposed DDF into your system ? Into which folder ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

str31ok, 2023-08-24:
> > Did you put the proposed DDF into your system ? Into which folder ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

No, i didn't. I thought that if deconz defined and signed it "correctly" (i mean the meaningful name, not the letter/digit sequence, the plug surely isn't the right definition)), then the file is already available in the system and there is no need to add anything else. I tried to find it in /usr/share/deCONZ/devices but unsuccessfully. 
I'll try to add DDF you are proposing and write about the result little later

Smanar, 2023-08-24:
Yep.
Deconz reconize the device as plug by defaut.
You need the DDF for it will be reconized as "warning device"

`"type:"Smart Plug"`

str31ok, 2023-08-25:
After adding proposed DDF Phoscon sees it like this

[image]

```
{
    "config": {
        "battery": 0,
        "duration": null,
        "melody": 0,
        "on": true,
        "pending": [],
        "reachable": true,
        "volume": 0
    },
    "ep": 1,
    "etag": "f2bdaf2c6f9c88299026b42daf6d02f4",
    "lastannounced": null,
    "lastseen": "2023-08-25T07:41Z",
    "manufacturername": "_TZE204_t1blo2bj",
    "modelid": "TS0601",
    "name": "Alarm 23",
    "state": {
        "alarm": null,
        "lastupdated": "none"
    },
    "type": "ZHAAlarm",
    "uniqueid": "a4:c1:38:94:b6:1b:e4:04-01-0500"
}
```
[image]

```
{
    "capabilities": {
        "alerts": [
            "none",
            "select",
            "lselect"
        ]
    },
    "config": {
        "groups": [
            "0"
        ]
    },
    "etag": "1c8e4c70f36e1f60cf2425b85c37e6e0",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2023-08-25T07:51Z",
    "manufacturername": "_TZE204_t1blo2bj",
    "modelid": "TS0601",
    "name": "Smart plug 9",
    "state": {
        "alert": "none",
        "reachable": true
    },
    "swversion": null,
    "type": "Warning device",
    "uniqueid": "a4:c1:38:94:b6:1b:e4:04-01"
}
```
I added siren to openhab as 2 "things", but i can't control it. I can't figure out where the failure occurs

Mimiix, 2023-08-25:
Please use the rest api to control it.

str31ok, 2023-08-25:
> Please use the rest api to control it.

i can change config parametrs using the rest api, but can't change its state

```
{
    "config": {
        "battery": 0,
        "duration": 10,
        "melody": 7,
        "on": true,
        "pending": [],
        "reachable": true,
        "volume": 1
    },
    "ep": 1,
    "etag": "0a6c2b58c4e06f6918e940719a6a3eb3",
    "lastannounced": null,
    "lastseen": "2023-08-25T11:18Z",
    "manufacturername": "_TZE204_t1blo2bj",
    "modelid": "TS0601",
    "name": "Сирена",
    "state": {
        "alarm": null,
        "lastupdated": "none"
    },
    "type": "ZHAAlarm",
    "uniqueid": "a4:c1:38:94:b6:1b:e4:04-01-0500"
}
```
What am i doing wrong? (sorry if this offtopic)
[image]

BabaIsYou, 2023-08-25:
By design a `state/...` item is most of the time readonly. There are some exception BTW.
Here we are in presence of a Tuya base device, then standard Zigbee logic is a bit disturbed.

Have you try `config/on` item ?

str31ok, 2023-08-25:
> Have you try `config/on` item ?

I've tried - the only thing what happens is that i can't read attributes in basic cluster

Smanar, 2023-08-25:
It's not possible to trigger a tuya siren with the actual DDF core.

hofvanta, 2023-10-26:
I have a similar issue _TZE204_t1blo2bj TS0601 Neo Smart Sirene. I have the DDF loaded but still not able to management it via HA and Deconz. The Sirene is recognized by Deconz as Light, but show a Smart Plug. Was you able to manage it via HA or REST API? Is there any update?

wischnu, 2023-12-02:
Did anybody get this device (_TZE204_t1blo2bj) to work with the Rest API or through Phoscon or homebridge? I tried all parameters on the REST API, but I was unable to switch it on or change the state. I have used the DDF as provided before, so it is shown as Alarm device in the API and phoscon

BabaIsYou, 2023-12-02:
You mean Tuya siren ? Because non Tuya are working with DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/e2413b85ba4c487bd457e9c5ceb720962e5c5538/devices/heiman/hs2wd-e_smart_siren.json#L4

Smanar, 2023-12-02:
Yep, it's because "state/alert" can't be "customised" to work with tuya device, it can work  only natively, so only for normal device.

wischnu, 2023-12-02:
Somehow they were able to get this working 

Tuya Smart Zigbee Siren Alarm Waterproof Outdoor With Solar (_TZE200_nlrfgpny)
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6941#issuecomment-1535313094

Smanar, 2023-12-03:
Probably because on this DDF, they are using state/on, but to avoid issue, siren need to use state/alert.
You can use state/on as workaround, but the DDF will be never valided by deconz.

hofvanta, 2023-12-06:
> Probably because on this DDF, they are using state/on, but to avoid issue, siren need to use state/alert. You can use state/on as workaround, but the DDF will be never valided by deconz.

So there is not solution to get this work with Deconz you say? With REST-API is not working with standard state/alert (atleast I dont get it to work) and state/on via DDF modification is also not working. Any one got it work with Deconz at all of Ziggbee2MQTT is the only option?

bauwfa, 2023-12-09:
please reopen this device request

Smanar, 2023-12-10:
I can, but will be useless, for the moment not possible having an official support for it, so you can just use the actual DDF.

bauwfa, 2023-12-10:
I wasn't able to pair my device, how to use the DDF ?

Smanar, 2023-12-10:
Depend of OS, to make short you need to create a text file with the DDF contain in a file called whatyouwant.json in a "devices" folder.

hofvanta, 2023-12-10:
But what is the correct & working DDF file? And works this in combination with Home Assistant, even with or without API? Did anyone get it to work with Deconz for the _TZE204_t1blo2bj? Cause some say that it is working with DDF and other say it never going to work.

Smanar, 2023-12-10:
Both are right, you can use it only for return, not to send state.
HA use only deconz API
Try the last DDF in this issue.

To add a DDF in HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

hofvanta, 2023-12-10:
> Both are right, you can use it only for return, not to send state. HA use only deconz API Try the last DDF in this issue.
> 
> To add a DDF in HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

@Smanar Tx for your reply. Which one you referring to? Because the last full DDF was not working for me. Posted by BabaIsYou.

Smanar, 2023-12-11:
Your device have the same modelID and same Manufacture name ?
Nothing appear in the API ? Visilbe in Phoscon/help/API Information/sensor.

hofvanta, 2023-12-11:
> _TZE204_t1blo2bj?

Yes (model ID _TZE204_t1blo2bj). Phoscon see siren as a bulb and sensor with siren type. Deconz sees the model (also tried several times to read the node (simple) descriptors. But maybe the DDF is not correct. Can you share again?

Smanar, 2023-12-12:
It seem  fine, you need to have a ZHAAlarm in sensor and a "Warning device" in light but this one can't work.
Ony the ZHAAlarm can work

ThiemeNL, 2024-06-30:
@Smanar when do you think it will be possible?

I loaded the DDF from the comment above. Now I have 2 'devices', a `Warning device` and a `ZHAAlarm`. I can succesfully change the config of the `ZHAAlarm`. For example update the melody field. 

But I'm not able to update the `state/alarm` to `true` for the `ZHAAlarm` or `state/alert`  to `select` for the `Warning device`. How can I trigger the alarm to go on?

[8 more replies]