dresden-elektronik/deconz-rest-plugin #6113: zigbee smart curtain motor _TZE200_r0jdjrvi TS0601

danidask, 2022-06-05:
## Device

- Product name: zigbee smart curtain motor
- Manufacturer: _TZE200_r0jdjrvi
- Model identifier: TS0601
- Device type : 
  - Other: curtain motor

## Screenshots

[image]

### Basic

[image]

### Identify

[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
tuya specific slusters is empty

Mimiix, 2022-06-05:
Can you please add a screen of the cluster list and also the other clusters if applicable.

danidask, 2022-06-05:
> Can you please add a screen of the cluster list and also the other clusters if applicable.

I added that info. Thanks for the quick reply!

Mimiix, 2022-06-05:
Please add all screens, even if they are zero.

danidask, 2022-06-05:
Done! BTW an "export data" option in deCONZ will save a lot of time, and github storage space IMO

danidask, 2022-06-09:
Please remove the "imcomplete issue" tag, or let me know if you need more information

Mimiix, 2022-08-07:
I'll reopen.

Mimiix, 2022-08-07:
@Smanar can you help?

Smanar, 2022-08-07:
Nope sorry, I m still waiting for PR for TRV using tuya cluster was merged to start Covering using the same cluster.

omegasxxi, 2022-08-23:
Will Thais decide be Support ex?

closed 2022-09-21 by github-actions[bot]

Smanar, 2022-11-29:
IDK, will be a surpise, we will perhaps a new code version for DDF, but I can't say when and if this version will be able tu support this device.