dresden-elektronik/deconz-rest-plugin #6115: Deconz overwriting LAN Settings

beatpaul, 2022-06-05:
## Describe the bug

 I want to set a Static IP adress (interface eth0) and some custom dnsmsq Settings for using Pihole on the same Raspberry pi, but settings get overwritten by deconz on reboot or restarting deconz.

## Steps to reproduce the behavior
 I Manualy edited:
- dhcpcd.conf
- dnsmasq.conf

After rebooting or restarting deconz Settings get overwritten by default deconz settings

## Expected behavior
It would be great, that manual changes of the interface settings / dhcpcd / dnsmasq stay persistent

## Screenshots

## Environment

 - Host system: Raspberry 
 - Running method: Raspbian
 - Firmware version:  21.4.2022
 - deCONZ version: 2.16.01
 - Device: RaspBee II
 - Do you use an USB extension cable: no -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? -> no

## deCONZ Logs

non

## Additional context
I'm using the latest SD-Card Image (and updated / upgraded all components) with homebridge-deconz by @ebaauw   

thx for your help and great work!

beatpaul, 2022-06-05:
Seems that this happened before:
https://github.com/dresden-elektronik/phoscon-app-beta/issues/128

Smanar, 2022-06-06:
To disable wifi, if I remember

`sudo systemctl unmask deconz-wifi`

beatpaul, 2022-06-06:
@Smanar 
do i have to mask or unmask the service? 

will it persist after reboot ?

Smanar, 2022-06-06:
Oups, sorry it's "mask" and yes, with this method the wifi is not overwritted again after reboot.
But I don't understand what you are trying to do (we can set a static ip using deconz config file) it 's just to react on your previous link.

beatpaul, 2022-06-06:
I added several other settings ussing dhcpcd, and they got overwritten.

Can u tell me where is the config file of deconz is placed?

Thx for your help!

Smanar, 2022-06-06:
Arf, bad luck you are right
https://github.com/dresden-elektronik/phoscon-app-beta/issues/26

We can change the port, but not DHCP stuff..

I have found this comment to

> The Accesspoint is only intended to access the Phoscon App when no other infrastructure is available e.g. in industry halls. Perhaps some other users or raspberry pi web tutorials can help how to forward internet to devices connected via raspbery pi access point. Perhaps like this one: https://www.raspberrypi.org/documentation/configuration/wireless/access-point-routed.md
> Phoscon also uses the files hostapd.conf, dnsmasq.conf and dhcpcd.conf to configure the accesspoint. Just delete the lines #created by deconz in these files, else Phoscon will restore the default configuration.
> I think net.ipv4.ip_forward=1 and sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE like describted in that tutorial would do the trick.

Do you know which one deconz service overwrite the config files ?

beatpaul, 2022-06-06:
This did the trick for me:
`sudo systemctl mask deconz-wifi`

Settings are not overwritten any more so far, i hope this works for other users too, but i‘m not sure if this all you have to do, since i tried out some other stuff too.

closed 2022-07-05 by github-actions[bot]