dresden-elektronik/deconz-rest-plugin #6123: RTX ZTH1

kozga, 2022-06-08:
## Device
- Product name: ZTH1
- Manufacturer: RTX
- Model identifier: TS0201
- Device type:  Sensor

It's a strange device,  on the sensor packaging it is described as RTX ZTH1, while in the manual the model is ZC-W1.

## Screenshots
* packaging
[image]

* manual
[image]

* attributes
[image]

* node info
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-06-08:
Hello so first the manufacture name is "_tz3000_i8jfiezr"

There is a PR to support it with c++ code here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5299
But as this one will be never merged can you try this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_i8jfiezr",
  "modelid": "TS0201",
  "vendor": "RTX",
  "product": "TS0201",
  "sleeper": true,
  "status": "Bronze",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/battery"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 1,
          "max": 6000,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 10,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 10,
          "max": 9999,
          "change": "0x00000014"
        }
      ]
    }
  ]
}
```

LukaszGaraKrk, 2022-06-08:
Here are the screenshots that were missing:
[image]
[image]
[image]
[image]

Mimiix, 2022-06-08:
Please add them to the original post.

Also missing the full list.

kozga, 2022-06-09:
We moved the screenshots to original post. After adding the DDF file, the sensor was detected correctly. Thanks for your help.

Smanar, 2022-06-09:
So all is working (battery too ?), I can make a PR for it ?

kozga, 2022-06-10:
It looks like the battery is working, response from api:

```
{
      "etag" : "d4953035b71b93321ccd0a724a581e8d",
      "modelid" : "TS0201",
      "manufacturername" : "_TZ3000_i8jfiezr",
      "type" : "ZHATemperature",
      "uniqueid" : "a4:c1:38:b5:b9:25:27:d0-01-0402",
      "name" : "Temperature 10",
      "config" : {
         "on" : true,
         "reachable" : true,
         "offset" : 0
      },
      "lastannounced" : null,
      "state" : {
         "temperature" : 2485,
         "battery" : 79,
         "lastupdated" : "2022-06-10T06:37:46.048"
      },
      "lastseen" : "2022-06-10T06:37Z"
}
```

```
{
      "manufacturername" : "_TZ3000_i8jfiezr",
      "modelid" : "TS0201",
      "etag" : "d1372e04179b092345fa3d7fb5ff50ff",
      "name" : "Humidity 11",
      "uniqueid" : "a4:c1:38:b5:b9:25:27:d0-01-0405",
      "type" : "ZHAHumidity",
      "lastseen" : "2022-06-10T06:37Z",
      "state" : {
         "humidity" : 4882,
         "lastupdated" : "2022-06-10T06:37:46.302",
         "battery" : 79
      },
      "lastannounced" : null,
      "config" : {
         "on" : true,
         "offset" : 0,
         "reachable" : true
      }
}
```

Smanar, 2022-06-10:
Thx, PR done.

closed 2022-07-06 by manup