dresden-elektronik/deconz-rest-plugin #6132: TUYA TS0012

imeSven, 2022-06-14:
## Device tuya ts0012

- Product name: SMART HOME ZT-201 ZIGBEE 2G1W TOUCH SWITCH 
- Manufacturer: TZ_3000_E98_KRVVK
- Model identifier: TS0012
- Device type : Please remove all unrelated device types. 
  - Switch

14:14:36:304 APS-DATA.request id: 32, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
14:14:36:304 	asdu (length: 3): 1a4101
14:14:36:305 send permit join, duration: 65
14:14:36:306 APS-DATA.request id: 33, addrmode: 0x02, addr: 0xfffc, profile: 0xA1E0, cluster: 0x0021, ep: 0xF2 -> 0xF2 queue: 1 len: 6 tx.options 0x00
14:14:36:306 	asdu (length: 6): 194e020b2800
14:14:36:307 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:357 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:420 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:481 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:555 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:606 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:670 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:731 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:796 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:869 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:931 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:985 Delay APS request id: 33 to 0xFFFC, profile: 0xA1E0 cluster: 0x0021 node already has busy 1
14:14:36:987 APS-DATA.confirm id: 32, status: 0x00 SUCCESS
14:14:37:029 aps request id: 32 finished, erase from queue
14:14:37:644 APS-DATA.confirm id: 33, status: 0x00 SUCCESS
14:14:37:692 aps request id: 33 finished, erase from queue
14:14:42:324 Idle timer triggered
14:14:48:036 APS-DATA.indication srcAddr: 0x22db, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 255, rssi: -34
14:14:48:037 	asdu: 00db22febd14665a38c1a480
14:14:48:038 unknown node 0xA4C1385A6614BDFE (0x22DB), lqi: 255
14:14:48:039 APS-DATA.indication from unknown node 0xA4C1385A6614BDFE
14:14:48:040 new node - ext: 0xa4c1385a6614bdfe, nwk: 0x22DB
14:14:48:041 device announce 0xA4C1385A6614BDFE (0x22DB) mac capabilities 0x80
14:14:48:041 set fast probe address to 0xA4C1385A6614BDFE (0x22DB)
14:14:48:042 FP indication 0x0000 / 0x0013 (0xA4C1385A6614BDFE / 0x22DB)
14:14:48:042                       ...     (0xA4C1385A6614BDFE / 0x22DB)
14:14:48:043 device announce 0xA4C1385A6614BDFE (0x22DB) mac capabilities 0x80
14:14:48:044 ZDP get node descriptor for 0x22DB
14:14:48:045 APS-DATA.request id: 81, addrmode: 0x03, addr: 0xa4c1385a6614bdfe, profile: 0x0000, cluster: 0x0002, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
14:14:48:045 	asdu (length: 3): 1bdb22
14:14:48:045 ZDP get node descriptor for 0x22DB
14:14:48:046 APS-DATA.request id: 82, addrmode: 0x03, addr: 0xa4c1385a6614bdfe, profile: 0x0000, cluster: 0x0002, ep: 0x00 -> 0x00 queue: 1 len: 3 tx.options 0x00
14:14:48:047 	asdu (length: 3): 1cdb22
14:14:48:047 Delay APS request id: 82 to 0x22DB, profile: 0x0000 cluster: 0x0002 node already has busy 1
14:14:48:107 Delay APS request id: 82 to 0x22DB, profile: 0x0000 cluster: 0x0002 node already has busy 1
14:14:48:170 Delay APS request id: 82 to 0x22DB, profile: 0x0000 cluster: 0x0002 node already has busy 1
14:14:48:232 Delay APS request id: 82 to 0x22DB, profile: 0x0000 cluster: 0x0002 node already has busy 1
14:14:48:294 Delay APS request id: 82 to 0x22DB, profile: 0x0000 cluster: 0x0002 node already has busy 1
14:14:48:325 APS-DATA.confirm id: 81, status: 0x00 SUCCESS
14:14:48:326 APS-DATA.confirm request id: 81 -> confirmed, timeout 92037096
14:14:48:396 APS-DATA.confirm id: 82, status: 0x00 SUCCESS
14:14:48:397 APS-DATA.confirm request id: 82 -> confirmed, timeout 92037376
14:14:48:507 APS-DATA.indication srcAddr: 0x22db, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8002, lqi: 255, rssi: -33
14:14:48:508 	asdu: 1b00db220240804111424200002a420000
14:14:48:509 APS-DATA.indication request id: 81 -> finished
14:14:48:510 APS-DATA.indication from child 0x22DB
14:14:48:510 APS-DATA.request id: 81 erase from queue
14:14:48:511 DB pushZdpDescriptorDb()
14:14:48:512 DB save zll database items 0x00000801
14:14:48:512 DB sql exec UPDATE devices SET nwk = 8923 WHERE mac = 'a4:c1:38:5a:66:14:bd:fe';INSERT INTO devices (mac,nwk,timestamp) SELECT 'a4:c1:38:5a:66:14:bd:fe', 8923, strftime('%s','now') WHERE (SELECT changes() = 0);
14:14:48:517 DB saved in 4 ms
14:14:48:517 DB UPDATE device_descriptors SET data = x'0240804111424200002a420000', timestamp = 1655205288 WHERE device_id = (SELECT id FROM devices WHERE mac = 'a4:c1:38:5a:66:14:bd:fe') AND endpoint = 0 AND type = 2
14:14:48:519 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'0240804111424200002a420000', 1655205288 FROM devices WHERE mac = 'a4:c1:38:5a:66:14:bd:fe'
14:14:48:528 APS-DATA.request id: 86, addrmode: 0x02, addr: 0x22db, profile: 0x0000, cluster: 0x0005, ep: 0x00 -> 0x00 queue: 1 len: 3 tx.options 0x00
14:14:48:529 	asdu (length: 3): 4fdb22
14:14:48:530 FP indication 0x0000 / 0x8002 (0xA4C1385A6614BDFE / 0x22DB)
14:14:48:531                       ...     (0xA4C1385A6614BDFE / 0x22DB)
[... 353 lines cut ...]
14:15:04:918 ZCL attribute report 0xA4C1385A6614BDFE for cluster: 0x0006, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
14:15:04:918 	payload: 0000100101402100000240210000
14:15:04:918 APS-DATA.request id: 195, addrmode: 0x02, addr: 0x22db, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x02 queue: 4 len: 5 tx.options 0x04
14:15:04:919 	asdu (length: 5): 10ae0b0a00
14:15:04:930 APS-DATA.confirm id: 195, status: 0x00 SUCCESS
14:15:04:930 APS-DATA.confirm request id: 195 -> erase from queue
14:15:05:019 aps request id: 195 finished, erase from queue
14:15:05:288 APS-DATA.indication srcAddr: 0x22db, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0xE001, lqi: 255, rssi: -33
14:15:05:289 	asdu: 08af0a10d03002
14:15:05:290 APS-DATA.indication from child 0x22DB
14:15:05:290 Node data 0xa4c1385a6614bdfe profileId: 0x0104, clusterId: 0xE001
14:15:05:291 FP indication 0x0104 / 0xE001 (0xA4C1385A6614BDFE / 0x22DB)
14:15:05:291                       ...     (0xA4C1385A6614BDFE / 0x22DB)
14:15:05:291 ZCL attribute report 0xA4C1385A6614BDFE for cluster: 0xE001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
14:15:05:291 	payload: 10d03002
14:15:05:291 APS-DATA.request id: 199, addrmode: 0x02, addr: 0x22db, profile: 0x0104, cluster: 0xE001, ep: 0x01 -> 0x02 queue: 4 len: 5 tx.options 0x04
14:15:05:292 	asdu (length: 5): 10af0b0a00
14:15:05:303 APS-DATA.confirm id: 199, status: 0x00 SUCCESS
14:15:05:304 APS-DATA.confirm request id: 199 -> erase from queue
14:15:05:390 aps request id: 199 finished, erase from queue

## Screenshots

[image]
[image]

Mimiix, 2022-06-14:
Not sure why you add all that logs, but please add all screenshots.

imeSven, 2022-06-14:
> Probably the attributes visible on the cluster 0x0006. But can you show the complete node, the bottom is trunkated so we can't see how many endpoint have the device.

[image]
oooh. here u go

Mimiix, 2022-06-14:
I want all  blue clusters in the original post  and being read. Check the wiki, there's a how to read clusters guide.

Mimiix, 2022-06-14:
If it's a duplicate I close this one and reopen the other.

closed 2022-06-14 by Mimiix