dresden-elektronik/deconz-rest-plugin #6141: WS5001

jacobhallgren, 2022-06-17:
## Device

- Product name: WS5001
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

### Basic 
[image]

### Identify 
[image]

### Other clusters that are not mentioned above 
[image]

Mimiix, 2022-06-17:
Missing screenshots of blue clusters. Please add them

jacobhallgren, 2022-06-17:
> Missing screenshots of blue clusters. Please add them
[image]
[image]
[image]
[image]
[image]

Mimiix, 2022-06-17:
Please add them to the original post.

jacobhallgren, 2022-06-17:
> Please add them to the original post.

@Mimiix Done

jacobhallgren, 2022-07-09:
Any Update on tvOS?

Mimiix, 2022-07-17:
@Smanar something you can do?

Smanar, 2022-07-18:
Can make a try, but from my memory schneider devices have 'special trick", we need to use a specific endpoint on the coordinator, but I don't know wich one devices.

Can you try  this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Scheider Electric",
  "modelid": "FLS/SYSTEM-M/4",
  "product": "4 gangs switch",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x15",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 22,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 23,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 24,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

It's a realy basic DDF, it just make the device includable in the API and configure bind for the cluster on/off for all 4 endpoints.

If it work the device will have a better name than 0x767C in deconz.
Then go in deconz, help / debug view, with flag "info" and "info_l2", and press button, if you are lucky you will see error about button map.

jensflorian, 2022-09-27:
please reopen.
This device is also branded as Merten MEG5001-0300 (https://eref.se.com/de/de/scope/product-pdf/MEG5001-0300).
How do I force the above DDF example onto this device? I tried the hot reload option
[image]

Mimiix, 2022-09-27:
Sure.

SwoopX, 2022-09-27:
@jensflorian You might want to type the correct manufacturer name 😋

jensflorian, 2022-09-28:
Thank you very much, after correcting the typo, the device is also shown in Phoscon.
[image]

Here is the output from help / debug view after pressing the top button:
[image]

 Can you create button maps from this?

Smanar, 2022-09-28:
Nope thoses logs ^^.
Enable flag "info_l2", you will have thoses usefull error lines

```
	Line 4570:         DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",
	Line 5607:         DBG_Printf(DBG_INFO_L2, "[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",
```

jensflorian, 2022-09-28:
The above screenshot was with flag "info_l2" enabled. I have only the 1gang cover.
Top button:
`17:22:58:143 [INFO] - No button map for: FLS/SYSTEM-M/4, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 33`

Bottom button:
`17:23:00:030 [INFO] - No button map for: FLS/SYSTEM-M/4, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 34`

This matches the AIRLINK device. So I added the device in button_maps.json as suggested by @Smanar and restarted deConz.

Now the log file returns after pressing the button:

`17:40:48:794 Websocket 192.168.1.130:36346 send message: {"attr":{"id":"113","lastannounced":null,"lastseen":"2022-09-28T15:40Z","manufacturername":"Schneider Electric","modelid":"FLS/SYSTEM-M/4","name":"Switch 113","swversion":"002.005.030 R","type":"ZHASwitch","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"},"e":"changed","id":"113","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:795 Websocket 192.168.1.130:36348 send message: {"attr":{"id":"113","lastannounced":null,"lastseen":"2022-09-28T15:40Z","manufacturername":"Schneider Electric","modelid":"FLS/SYSTEM-M/4","name":"Switch 113","swversion":"002.005.030 R","type":"ZHASwitch","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"},"e":"changed","id":"113","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:796 Websocket 192.168.1.130:36366 send message: {"attr":{"id":"113","lastannounced":null,"lastseen":"2022-09-28T15:40Z","manufacturername":"Schneider Electric","modelid":"FLS/SYSTEM-M/4","name":"Switch 113","swversion":"002.005.030 R","type":"ZHASwitch","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"},"e":"changed","id":"113","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:797 Websocket 192.168.1.130:36368 send message: {"attr":{"id":"113","lastannounced":null,"lastseen":"2022-09-28T15:40Z","manufacturername":"Schneider Electric","modelid":"FLS/SYSTEM-M/4","name":"Switch 113","swversion":"002.005.030 R","type":"ZHASwitch","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"},"e":"changed","id":"113","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:800 Websocket 192.168.1.130:36346 send message: {"e":"changed","id":"113","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-09-28T15:40:48.793"},"t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:801 Websocket 192.168.1.130:36348 send message: {"e":"changed","id":"113","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-09-28T15:40:48.793"},"t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:802 Websocket 192.168.1.130:36366 send message: {"e":"changed","id":"113","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-09-28T15:40:48.793"},"t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
17:40:48:804 Websocket 192.168.1.130:36368 send message: {"e":"changed","id":"113","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-09-28T15:40:48.793"},"t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760)
1`

The bindings tab in the DDF editor is still empty. Do I need to add anything here?

Smanar, 2022-09-29:
>192.168.1.130:36346 send message: {"e":"changed","id":"113","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-09-28T15:40:48.793"},"t":"event","uniqueid":"04:cd:15:ff:fe:c0:d1:14-ff-0006"} (ret = -1093345760) 17:40:48:801 Websocket

This one have worked, we can see `"buttonevent":1002` as the value is not null, it have worked at a moment.
You can still use the debug with "info_l2" (only, less talkative), it will be more visible.

>The bindings tab in the DDF editor is still empty

Not normal, you need to have thoses binding 
```
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 22,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 23,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 24,
      "dst.ep": 1,
      "cl": "0x0006"
```

You can see them in the DDF.
I making a bind (only for the cluster 0x0006 to test) but for all endpoint (If I m right you have 1 endpoint by button). 0x15 is the hexa version, 21 the decimal value.

If you haven't log "[INFO] - No button map for" for the second button, it's probably a problem with bind yes.

If you have them but no change on buttonevent return it s probably due to the button map file (The code use only endpoint 0x15 and 0x16, you have 2 mores on your device from capture)

It's a 4 gangs device ?

jensflorian, 2022-09-29:
> Not normal, you need to have thoses binding

Sorry, my fault, I forgot to reload the json file after restarting deConz. Now I have all four endpoints in the DDF. Thanks!

>If you haven't log "[INFO] - No button map for" for the second button, it's probably a problem with bind yes.

The second button has buttonevent 2002. 

>It's a 4 gangs device ?

Depending on the button cover you are using its either  1gang(2button) or a 2gang(4button) switch. This will be commissioned during the pairing process, see https://www.productinfo.schneider-electric.com/wiser_home/sysm_sysd-connected-wireless-switch-1-gang_2-gang_device-user-guide_wiser-home/English/SysM_SysD%20-%20Connected%20Wireless%20Switch%201-gang_2-gang_Wiser_Home_Device%20user%20guide_0000774577.xml/$/WHM-SysM_SysD-Connectedwirelessswitch1-gang_2-gang-PairingthedeviceTSK_0000735364
As I have only the two button cover, I'm getting only button events 1002 and 2002.

I think this information is stored in the Merten specific cluster FF17. To read the clusters you need to wake up the device by pressing the top button 3x quickly. The green light will flash and you can read all the clusters:
[image]

Can you make a PR for your DDF and button_maps.json ? I'd be happy to have this device included in future deconz releases.

Smanar, 2022-09-29:
Sure I can, but it's finished ?
Your device is only a on/off, no dimmer feature ?
If you have the dimmer feature (generaly long press) I thinkj you need to bind the cluster 0x0008 too with adding

```
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 22,
      "dst.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 23,
      "dst.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 24,
      "dst.ep": 1,
      "cl": "0x0008"
    }
```

> Pressed button 1 and 4 for about 5 seconds to enable individual buttons. :-)

BTW, you have the different value the attribute 0x0020 can have ?

jensflorian, 2022-09-30:
> Your device is only a on/off, no dimmer feature ?

According to the manufacturer manual it is capable to dim, use a shutter and trigger a moment. However I do not get any other messages with "info_l2" enabled as for button events 1002 and 2002. There is no other message or error after long press. So, it needs probably to be configured by the manufacturer app and this is probably in the Merten specific cluster FF17 somewhere. I haven't figured this out yet.

>BTW, you have the different value the attribute 0x0020 can have ?

For endpoint 15 its "0" for the other 16,17 and 18 it's "1"

The attribute 0x0001 is "unknown for endpoint 15 and for the other 16,17 and 18 it's "7f"

Smanar, 2022-09-30:
> For endpoint 15 its "0" for the other 16,17 and 18 it's "1"
> The attribute 0x0001 is "unknown for endpoint 15 and for the other 16,17 and 18 it's "7f"

Ok so need investigation, because I don't see the logic

>There is no other message or error after long press. So, it needs probably to be configured by the manufacturer app and this is probably in the Merten specific cluster FF17 somewhere. I haven't figured this out yet.

Or they are using the cluster 0x0008 , the level control, try to add them or can just bind one manualy to test.

Else I can make the PR as it ?

jensflorian, 2022-10-01:
> Or they are using the cluster 0x0008 , the level control, try to add them or can just bind one manualy to test.

Tested it without success, is probably not enabled by default.

>Else I can make the PR as it ?

Don't forget to correct the typo in "manufacturername": "Scheider Electric" to Schneider.
Thank you for your input and help!

Smanar, 2022-10-01:
Ok so have removed bind for cluster 0x0008

Final DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6373

Smanar, 2022-10-01:
Oups, it' perhaps possible to mix 2 DDF, can you try this one ?

It's an update of this one devices\wiser\fuga_4button_battery_switch.json

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": {"Schneider Electric", "Schneider Electric"],
  "modelid": ["FLS/AIRLINK/4", "FLS/SYSTEM-M/4"],
  "product": "LK Fuga 4 Button Switch",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x15",
        "0x0006"
      ],
      "fingerprint": { "profile": "0x0104", "device": "0x000c", "endpoint": "0x15", "in": ["0x0000", "0x0001", "0x0006", "0x0008"] },
      "items": [
        {
            "name": "config/battery",
            "parse": {"ep": 21, "cl": "0x0001", "at": "0x0021", "eval": "Item.val = Attr.val / 2"},
            "awake": true
        },
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 21,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 22,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 21,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 22,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 21,
      "cl": "0x0001",
      "report": [ {"at": "0x0021", "dt": "0x20", "min": 300, "max": 3600, "change": "0x01" } ]
    }
  ]
}
```

jensflorian, 2022-10-01:
> possible to mix 2 DDF?
unfortunately not yet. This one doesn't work.  This will cause:

`  "modelid": "FLS/SYSTEM-M/4",
  "product": "FLS/SYSTEM-M/4",`
and endpoints missing.

Smanar, 2022-10-01:
Oups my bad, have made a typo, have corrected the DDf on the previous post.

jensflorian, 2022-10-01:
This one works!

Smanar, 2022-10-02:
Nice, I will make a new PR with both device mixed, as it not possible to know if it's a 2 gang or 4 gang (same firmware, same hardware, just a setting) I don't think it can cause issue.

closed 2022-10-21 by manup

frehov, 2024-04-01:
I'm very sorry to necro this issue, but if you have any trouble with the FF17 cluster, please take a look at the zigbee implementation docs from Schneider, which details the inner workings of the FF17 cluster for the these switches/remotes.
https://drive.google.com/file/d/1FANEmbeh89I6Hn9e6ddt1YzSzk_GofJd/view?usp=sharing

This device can function in 1-gang, 2-gang and 4-pushbutton mode, depending on FF17 cluster values.

These documents were found on an aws cloudfront distribution that had been indexed by duckduckgo, along with 7 other zigbee implementation documents https://drive.google.com/drive/folders/1PaPQfbuMHdtiV199EIIirJE5RbujuJXn?usp=sharing for different Schneider devices.

Smanar, 2024-04-01:
Ha nice, usefull informations, thx