dresden-elektronik/deconz-rest-plugin #6142: Communication issue after an upgrade to 2.16.01

gkgeorgiev, 2022-06-17:
## Describe the bug
After upgrading to the latest firmware (2.16.01 / 4/21/2022) I no longer am able to connect the gateway with the Zigbee devices. I tried to delete one hue dimmer and re-pair it but without success. In the debug view I get only:

[image]
I restored the DB from the backup I took right before the upgrade but the same. 

## Steps to reproduce the behavior
Upgrade to the version from the screenshot above.

## Expected behavior
To be able to communicate with the Zigbee devices.

## Screenshots

## Environment
See the screenshot.

 - Host system: Ubuntu 20.04
 - Running method: Ubuntu
 - Firmware version: 26720700
 - deCONZ version: 2.16.01
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Logitech keyboard and mouse transceivers 

## deCONZ Logs
23:14:39:270 UPNP socket not bound, state: 0
23:18:07:770 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
23:18:07:770 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF051AC1, start reconfiguration
23:18:07:770 Skip automatic channel change, TODO warn user
23:18:17:770 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
23:18:17:770 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF051AC1, start reconfiguration
23:18:17:770 Skip automatic channel change, TODO warn user

## Additional context

Mimiix, 2022-06-18:
Hi,

As this is not a bug report as described in #5113 I am closing this issue. Feel free to open an topic on the forum.

Kind regards,

closed 2022-06-18 by Mimiix

gkgeorgiev, 2022-06-18:
Well IMHO it was a bug since I had a system working for 2+ years, and suddenly I decided to upgrade my Conbee II to the latest firmware and after this I was no longer able to use my device. 
For those who face the same issue and don't wanna waste time in pointless discussions, I downgraded my firmware to this version https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF  and worked like a charm.

Thx anyway

Mimiix, 2022-06-18:
You probably clicked the firmware update button 
which doesn't work in the ha addon.  Nevertheless, your apt ext ID changed (as shown in the logs). Changing it to the correct values as the logs suggested would've fixed it.

No clue why it got scrambled. Probably because of the ha addon or you might have started ZHA.

gkgeorgiev, 2022-06-18:
> 

I've started it from the native deConz app. What I found strange is that this ID got "scrambled" only with the latest firmware. As soon as re-flashed it with the version before (Commit: 1dcc54
Firmware: 0x26720700) seems to be back to normal. 
Thx for your clarification!