dresden-elektronik/deconz-rest-plugin #6143: Royal Gardineer ZigBee Water Valve

fphoefling, 2022-06-18:
## Device

- Product name: Royal Gardineer ZX-7155-675
- Manufacturer: _TZE200_2wg5qrjy
- Model identifier: TS0601
- Device type :
  - Other: Water Valve

## Screenshots

### Basic
[image]
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]

### Other clusters that are not mentioned above
[image]
[image]

Smanar, 2022-06-18:
Hello can you try this DDF ?

It will probably enable the on/off

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_2wg5qrjy",
  "modelid": "TS0601",
  "vendor": "$MF_TUYA",
  "product": "Royal Gardineer ZigBee Water Valve",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

I think we can add the water consumption, but I have serious doubt for the timer.

fphoefling, 2022-06-19:
Thanks for the quick answer. Works so far... Recognized by Phoscon App. on/off is enabled. Real world test is in progress. Will report result later.

Smanar, 2022-06-20:
Yes using light, can be the pairing procedure, but tuya make so much clone, better to check in deconz too if you realy have the same (You can include a device in deconz and not ahing it in the API)

fphoefling, 2022-06-20:
I paired the device by following this procedure:
### Step 1
Open Phoscon App -> Add new light -> Press button on device for 3 s until it flashes -> Release button
In this stage the device does not appear in the phoscon app, but it is visible in the deCONZ app. 
### Step 2
Change to the folder the deCONZ app is installed in. In my case its `/home/pi/.local/share/dresden-elektronik/deCONZ`.
Find the subfolder `devices` (or create one). In my case its path is `/home/pi/.local/share/dresden-elektronik/deCONZ/devices`
Create there a json file with the content Smanar provided. I called it `ZX-7155-675.json`.
### Step 3
Change to the deCONZ app. The device is visible here under its NWK address (in my case: `0xDBBE`) -> Select this node.
While the node is selected (has the blue border) press `CTRL+E`. A new window pops up: The DDF Editor.
Press `CTRL+O` and open the file you created in Step 2.
### Step 4
Now you find the device in the Phoscon Apps list of lights (if not, try rebooting deCONZ).

Smanar, 2022-06-21:
You can too just make the Step 2.
Then include the device, deconz will use the DDF to reconise it direclty.

olliox, 2022-06-22:
I‘m quite a noob to be honest  - just used deconz gui once - sorry.
I just use homebridge and phoscon webgui to add devices. 

Will these expert steps be added in a new version of phoscon/deconz to get the devices running without the steps?

that would be fantastic

MatKennel, 2022-06-23:
Good Job! Works so far for me also. Now I am excited to get the measurement running. Nevertheless I still miss any clue how to proceed. DPID 1 is obviously state/on. Any idea where to find the specification for the measurement?

Smanar, 2022-06-23:
Sure, dp for battery is 7, for water consumtion it's 5, can try to add a ZHAConsumption (to have a sensor with consumption) and add inside

```
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 5, "eval": "Item.val = Attr.val / 33.8140226;"},
          "read": {"fn": "none"}
        },
```

MatKennel, 2022-06-23:
Hi @Smanar thanks for the fast advice. Currently I succeed with Battery (not sure if the value needs some scaling) but I still have zero consumption (that's why I removed the scaling) and swversion is null too. Any recommendation?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_2wg5qrjy",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Silver",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "public": false,
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 300,
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          }
[... 149 lines cut ...]
                "lastupdated": "2022-06-23T19:39:04.540"
            },
            "type": "ZHAConsumption",
            "uniqueid": "04:0d:84:ff:fe:5c:bf:30-01-0702"
        },

"105": {
            "config": {
                "on": true,
                "reachable": true
            },
            "etag": "2255afff1b7c64f5109ccfdb45597754",
            "lastannounced": null,
            "lastseen": "2022-06-23T19:44Z",
            "manufacturername": "_TZE200_2wg5qrjy",
            "modelid": "TS0601",
            "name": "Battery 105",
            "state": {
                "battery": 50,
                "lastupdated": "2022-06-23T19:37:08.206"
            },
            "type": "ZHABattery",
            "uniqueid": "04:0d:84:ff:fe:5c:bf:30-01-0001"
        },
```

Smanar, 2022-06-24:
From that I m reading the battery is without convertion, so you are already at  50 % ?

Something that can help, look on logs with "ddf" "info" and "info_l2", you will see all request made by the device, even the one not used.

```
        DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",
                   rt, seq, dpid, dataType, dataLength, num.s32);
```

This one appear with the flag "info" (if other are too much talkative).
The problem is the device send request when it want, so can wait so long time before having something usefull.

And BTW why using a ZHABattery ? can use the config/battery in the ZHAConsumption ?

olliox, 2022-06-24:
If I try to create the json. file in the folder of step 2 it says: permission denied. Also tried to duplicate another json file in the devices folder and change the name and content - didn't work to and says permission denied.
I'm connected via VNC Viewer - stopped the deconz service and deconz is running. I can find all of my 60 nodes and also the node of the water valve - but I have the problem above to create the json. Can somebody help me?
VNC viewer connects via user: pi pw: raspberry.  I'm using homebridge and installed deconz/phoscon via "sudo hb-config" in terminal

Smanar, 2022-06-25:
On linux to increase right

`sudo chmod -R 777 folder_name`

MatKennel, 2022-06-25:
50% battery souds okay: I just took some old accu.
I  checked the behavior if all dpids here summary:

```
//DPID 1
//on/off
{"log":"10:10:41:363 TY_DATA_RESPONSE: seq 86, dpid: 0x01, type: 0x01, length: 1, val: 0\r\n","stream":"stdout","time":"2022-06-25T10:10:41.47135265Z"}
{"log":"10:10:36:993 TY_DATA_RESPONSE: seq 78, dpid: 0x01, type: 0x01, length: 1, val: 1\r\n","stream":"stdout","time":"2022-06-25T10:10:36.998293125Z"}

//DPID 5
//Consumption always "0"
{"log":"10:10:41:778 TY_DATA_RESPONSE: seq 89, dpid: 0x05, type: 0x02, length: 4, val: 0\r\n","stream":"stdout","time":"2022-06-25T10:10:41.792515512Z"}

//DPID 6
//?? always "0"
{"log":"10:10:41:608 TY_DATA_RESPONSE: seq 88, dpid: 0x06, type: 0x02, length: 4, val: 0\r\n","stream":"stdout","time":"2022-06-25T10:10:41.63111265Z"}

//DPID 7
//Battery
{"log":"10:10:08:596 TY_DATA_RESPONSE: seq 70, dpid: 0x07, type: 0x02, length: 4, val: 50\r\n","stream":"stdout","time":"2022-06-25T10:10:08.67129415Z"}

//DPID 11
//CountDown timer: Starts with 600=600s=10min and goes down to 0 --> valve is automatically turned off
{"log":"10:10:37:236 TY_DATA_RESPONSE: seq 80, dpid: 0x0B, type: 0x02, length: 4, val: 600\r\n","stream":"stdout","time":"2022-06-25T10:10:37.288975525Z"}
{"log":"10:05:06:614 TY_DATA_RESPONSE: seq 55, dpid: 0x0B, type: 0x02, length: 4, val: 420\r\n","stream":"stdout","time":"2022-06-25T10:05:06.670393137Z"}

//DPID 12
//?? Sometimes "1" sometimes "2": I see no logic
{"log":"10:10:41:832 TY_DATA_RESPONSE: seq 90, dpid: 0x0C, type: 0x04, length: 1, val: 2\r\n","stream":"stdout","time":"2022-06-25T10:10:41.95098895Z"}
{"log":"10:10:37:119 TY_DATA_RESPONSE: seq 79, dpid: 0x0C, type: 0x04, length: 1, val: 1\r\n","stream":"stdout","time":"2022-06-25T10:10:37.155116725Z"}

//DPID 15
//?? mostly 0 sometimes related to last DPID 11
{"log":"10:10:21:217 TY_DATA_RESPONSE: seq 73, dpid: 0x0F, type: 0x02, length: 4, val: 480\r\n","stream":"stdout","time":"2022-06-25T10:10:21.231302612Z"}

```
1) Might there be the need to set DPID 5 to the reqested amount of water before starting and see it go down to "0" during water flow?
2) I see in the documentation that there is the option to set the openingWith of the valve. Any Idea how this could work? DPID 6?

Smanar, 2022-06-25:
DPID 12 is the timer, enabled or not.
DIP 15 is the time during the valve is open
DPID 11 is the timer remain before the valve is closed

No information about the dpid 6

For the DPID 5, idk, perhaps need to use lot of water for the value increase, it's a counter, no ? And it seem the value is in fl. oz.

der-burn, 2022-06-25:
Concerning capabilities, that's what I found so far: https://www.zigbee2mqtt.io/devices/ZVG1.html
Maybe that sheds some lights on what DPID 6 is?

Smanar, 2022-06-25:
All my information are from Z2m ^^.
And it seem they are not using the dp 6 too. Or I have skipped something.

MatKennel, 2022-06-30:
After replacing batteries (now dpid 7 shows 100%) and letting the water run for some 10minute cycles I gave up. My assumption is that water/flow measurement has a defect in the device. Before replacing the device with a new one: Did anyone in our community succeed in getting some values on dpid 0x05 which are not equal to „0“?

Smanar, 2022-07-01:
The device haven't 2 mounting side ?

der-burn, 2022-07-02:
@MatKennel I'm fiddling around with these as well, though it's through ZHA for home assistant. Have 3 of them and none is reporting flow through DP 5.
Also I'm having trouble to set the device default valve open time ( factory value 600s). Has anyone succeeded to change it and if so it is which DP?

Smanar, 2022-07-02:
From z2M

DIP 15 is the time during the valve is open
DPID 11 is the timer remain before the valve is closed

But they use it only for return

To set value, they are using only the dp 11. Perhaps an issue on their side, 
They send 0 1 to dp 11 for enable disable
They send a timer in minut to the same dp to set the timer.

adx-git, 2022-07-05:
I have had the same device request a few weeks ago, the "on/off" DDF as provided by @Smanar is working fine for me.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6058
Is there anything I could help to verify?

I have two devices, one was very easy to pair, I just used the Phoscon webpage and searched for a new light, afterwards I added  the DTS file using the deCONZ GUI. Unfortunately I was not able to pair the second device, I will return this one, assuming it is defective.

MatKennel, 2022-07-11:
@adx-git I am still struggeling to get the water consumption working. Not sure if I should re-order the device after returning. See above you should see TY_DATA_RESPONSE messages in the debug logs. Do you see any value different than 0 for dpid 0x05 ?

der-burn, 2022-07-11:
> To set value, they are using only the dp 11. Perhaps an issue on their side, 
> 
> They send 0 1 to dp 11 for enable disable
> 
> They send a timer in minut to the same dp to set the timer.

That works for a one time setup, changing the valve close time. But once you close and open the valve again, value is rest to 600. Triggering the valve and overwrite the value each time through an automation would work but that's not really what I'm looking for...
Was hoping that there is a possibility to change the default value for once and all.

mehrdadparsa, 2022-07-20:
> I paired the device by following this procedure:
>
> ### Step 1
> Open Phoscon App -> Add new light -> Press button on device for 3 s until it flashes -> Release button In this stage the device does not [...]

I got it paired to my conbee II. The Problem ist I only can open the valve using state on/off. When change the item state from 0 to 1, the valve is opened and the signal is reseted automatically. I don't seem to be able to send any signal to turn it off. When I toggle the same signal, it receives another open signal but not close signal. Am I missing something?

adx-git, 2022-07-20:
Haven't tried that yet, but note that it will close automatically after some time.

mehrdadparsa, 2022-07-20:
> >When change the item state from 0 to 1
> 
> You are talking about wich one item if if it's not state/on ?

Well, i have it as an Item in my openhabian system. It's called for some reason brightness because i coupled the valve as a light in Phoscon Web GUI ( i assume).

Smanar, 2022-07-21:
Ha yes right, its something to correct in the DDF
```

  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "
```

Better to use $TYPE_ON_OFF_OUTPUT (and the DDF don't use state/bri)

closed 2022-08-19 by github-actions[bot]