dresden-elektronik/deconz-rest-plugin #6146: Perenio Power Link

inamesh, 2022-06-20:
## Device

- Product name: Perenio Power Link
- Manufacturer: Perenio
- Model identifier: PEHPL0X
- Device type : 
  - Switch

## Screenshots

### Node Info
[image]

### All Clusters
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### On/Off
[image]

### Simple Metering
[image]
[image]

### OTAU
[image]

Smanar, 2022-06-20:
> Device type :
>     Switch

Sure it's not a plug ?

Can try this DDF, made for plug

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Perenio",
  "modelid": "PEHPL0X",
  "vendor": "Perenio",
  "product": "Power Link Plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power"
        },
        {
          "name": "state/voltage"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Smanar, 2022-06-21:
Can you take a look direclty in the API, for exemple using Phoscon, help / API iInformation / sensor ?
Because there is no reason for the DDF create only the ZHAConsumption and not the ZHAPower.

On your captuere the value for consumption is 0 too
And for electrical measurement cluster, you forget to press the "read" button, so not possible to check working value or not.

You have see others DDF in the folder you have put the DDF ? (or have used custom/user folder ?)

inamesh, 2022-06-23:
Hi!
After a couple of restarts I now have 2 sensors in this device, but the values are still strange - 
[image]

I checked with API and below are the responses for each of these sensors -

**ZHAPower**

{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "a550f90994314909ab9c0767c2ed99ab",
    "lastannounced": null,
    "lastseen": "2022-06-23T12:19Z",
    "manufacturername": "Perenio",
    "modelid": "PEHPL0X",
    "name": "Power 47",
    "state": {
        "current": 0,
        "lastupdated": "none",
        "power": 0,
        "voltage": 0
    },
    "swversion": "1.3.11",
    "type": "ZHAPower",
    "uniqueid": "f4:ce:36:ce:42:ab:fc:55-01-0b04"
}

**ZHAConsumption**

{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "e24b3c5071493b3ac2a52d75ee2cf318",
    "lastannounced": null,
    "lastseen": "2022-06-23T12:24Z",
    "manufacturername": "Perenio",
    "modelid": "PEHPL0X",
    "name": "Consumption 46",
    "state": {
        "consumption": 18,
        "lastupdated": "2022-06-23T12:25:04.446"
    },
    "swversion": "1.3.11",
    "type": "ZHAConsumption",
    "uniqueid": "f4:ce:36:ce:42:ab:fc:55-01-0702"
}

The values on the cluster do not change on pressing "read" -

[image]

Finally, I don't have any other ddfs in the folder. I'm using the below folder (which the deCONZ GUI opens by default) - /usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices

Thank you once again, and do let me know if I can help troubleshoot further!

Smanar, 2022-06-23:
>The values on the cluster do not change on pressing "read" -

Seem realy bad, it s like your device don't support them.
As it's a powered device, no need to wake it, the "read" button worj immediatly.

For the consumption, I can't be sure, I can use a convertor, but from your capture, the value need to be as it, no divisor, no multiplicator

closed 2022-07-23 by github-actions[bot]