dresden-elektronik/deconz-rest-plugin #6158: Device name

Tscheurgen, 2022-06-30:
## Device

- Product name: Tuya Luminance Motion Sensor Model ZG-204ZL
- Manufacturer: _TZE200_3towulqd
- Model identifier: TS0601
- Device type : 
  - Sensor

## Screenshots

Node Info
[image]

Basic Cluster
[image]

Power Configuration
[image]

OTAU
[image]

Mimiix, 2022-07-01:
Hi,

Are these all the clusters in the cluster list

Tscheurgen, 2022-07-01:
Hello!
Here the clusterlist
[image]
Thank you!

Smanar, 2022-07-01:
Ha, it seem this device don't use the IAS cluster, but only the tuya one.

Can start with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_3towulqd",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "ZG-204ZL",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/lightlevel",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val == 0;"
          },
          "read": {
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/presence",
          "parse": {
            "fn": "tuya",
            "dpid": 12,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "state/dark"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lux"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

You can try state/presence and state/lightlevel

Smanar, 2022-07-08:
No difference ?
Your DDF look like that now 

```
        },
        {
          "name": "state/lightlevel",
          "parse": {
            "fn": "tuya",
            "dpid": 12,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/presence",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val == 0;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
```

If yes, can you enable the deconz log (help / debug view) with the flag "DDF" and "info_l2", and try to trigger a detection ?
(Can make a screen capture if you can make a copy/paste)

Tscheurgen, 2022-07-19:
Hello!
Sorry for delaying my answer!
I hadn't swapped the eval value. But now it is as well.
Here ist the log. Hopefully with enough information. I didn't find more entries with the Name or Mac Adress.
[Lightlevel62.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9139365/Lightlevel62.txt)

Smanar, 2022-07-19:
You have enable the flag "DDF" and "info_l2" ?
If yes it mean the device haven't make report at all, when a tuya device make request you will have something like that in logs

```
        DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",
                   rt, seq, dpid, dataType, dataLength, num.s32);
```

Oups, sorry my bad, this log is visible wih "info" not "info_l2"

Tscheurgen, 2022-08-01:
Finally here is the latest log. Mac-Adress is a4:c1:38:b6:7c:58:a8:be in Line 7 first occurance.
[Lighlevel62_Info_DDF.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9237008/Lighlevel62_Info_DDF.txt)
Thank you!

Smanar, 2022-08-02:
```
21:43:44:788 TY_DATA_REPORT: seq 160, dpid: 0x01, type: 0x04, length: 1, val: 0
21:43:44:829 a4:c1:38:b6:7c:58:a8:be-01-0400/state/presence expression: Item.val = Attr.val == 0; --> true
```
```
21:43:42:094 TY_DATA_REPORT: seq 158, dpid: 0x01, type: 0x04, length: 1, val: 1
21:43:42:132 a4:c1:38:b6:7c:58:a8:be-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false
```

So the presence seem working. Perhaps reversed ?

```
21:43:12:971 a4:c1:38:b6:7c:58:a8:be-01-0400/state/lightlevel expression: Item.val = Attr.val; --> 72
21:43:12:971 TY_DATA_REPORT: seq 154, dpid: 0x0C, type: 0x02, length: 4, val: 72
```

```
21:43:27:767 a4:c1:38:b6:7c:58:a8:be-01-0400/state/lightlevel expression: Item.val = Attr.val; --> 66
21:43:27:768 TY_DATA_REPORT: seq 156, dpid: 0x0C, type: 0x02, length: 4, val: 66
```

Lightlevel seem working too, but more complex for this one.

State/lux is not used previosuly in the DDF.
From the code It seem
- lux is The current Light level in lux
- lightlevel is The current Light level in 10000 log10 (lux) + 1.

I have found this table for this device.

>                         // Real lux - Measured value
>                         // 0          10
>                         // 30.1       778
>                         // 43.3       826
>                         // 57.4       850
>                         // 75.7       898
>                         // 98.6       922
>                         // 127.6      1000

But if value go from 0 to 1000 maximum there is a problem, we can have 500 Lux.
So I presume this value is in Lux and need to be converted in lightlevel

Can try this DDF, but it NEED to be in the tuya folder (because of the path file of JS script)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_3towulqd",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "ZG-204ZL",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/presence",
          "parse": {
            "fn": "tuya",
            "dpid": 12,
            "eval": "Item.val = Attr.val == 0;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "state/dark"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lux",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js"
          },
          "read": {
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}

```

Smanar, 2022-08-05:
>Where did you find this special table?

On z2m github https://github.com/Koenkk/zigbee2mqtt/issues/12364  , but they don't use it on final code, and display value as it, are you sure tuya gateway display standard value ?

>Do I have to add the Lightlevel conversion first? and also Lightlevel DDF Information?

Not sure to understand, but the file "/generic/illuminance_cluster/lux_to_lightlevel.js" is not present on your side ? (it is 3 month old)

And I have checked other project, no one use the good formula to have correct value, I have searched on my side, and don't found how to have thoses result 

```
                    // Real lux - Measured value
                    // 0          10
                    // 30.1       778
                    // 43.3       826
                    // 57.4       850
                    // 75.7       898
                    // 98.6       922
                    // 127.6      1000
```

closed 2022-09-04 by github-actions[bot]

llugo, 2022-10-15:
Hi,
i have this device with and without light meter.
Both get recognzied by Phoscon but only one of them connects to the zigbee interface (rasbee 2).
This is what i get from the API:
    "4": {
        "config": {
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "etag": "4d1069793df1f9b6cad8dd90a1b6d350",
        "lastannounced": "2022-10-15T11:18:48Z",
        "lastseen": "2022-10-15T11:19Z",
        "manufacturername": "_TZE200_3towulqd",
        "modelid": "TS0601",
        "name": "LightLevel 4",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2022-10-15T11:19:30.713",
            "lightlevel": 0,
            "lux": 0,
            "presence": false
        },
        "type": "ZHALightLevel",
        "uniqueid": "a4:c1:38:01:13:e6:33:1f-01-0400"
    },

So presence is false and lux is 0. Since the sensors aren't lying in a box, those values are wrong. So i guess the ddf just doesn't work for me?

Smanar, 2022-10-15:
>Both get recognzied by deCONZ, but only one of them connects to the zigbee interface (rasbee 2).

Men you can see both in deconz (the GUI application) but only one is visible in the API ?

You have put the DDF in the "tuya" folder ? Because it use a local script.

Can take a look in deconz log (help / debug view) with enabling "info" and "info_l2" and try to trigger a presence, you will see some line like

```
        DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",
                   rt, seq, dpid, dataType, dataLength, num.s32);
```

llugo, 2022-10-15:
> Men you can see both in deconz (the GUI application) but only one is visible in the API ?

They're both shown in deCONZ and through the API, but only one of them shows a link to the raspbee II / configuration tool.
The other one doesn't get updated for some reason. It surely is not the battersy since both do red flashes every few secs.
This is what i get from the unconnected one:
 >   "5": {
        "config": {
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "etag": "60daa20ba0641372e31214fe96514167",
        "lastannounced": "2022-10-12T12:53:00Z",
        "lastseen": "2022-10-12T16:09Z",
        "manufacturername": "_TZE200_3towulqd",
        "modelid": "TS0601",
        "name": "LightLevel 5",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2022-10-12T16:09:18.443",
            "lightlevel": 0,
            "lux": 0,
            "presence": false
        },
        "type": "ZHALightLevel",
        "uniqueid": "a4:c1:38:b5:d8:f7:f0:13-01-0400"
    },

> You have put the DDF in the "tuya" folder ? Because it use a local script.

Yep loaded the json file from /usr/share/.../devices/tuya/

> 
> Can take a look in deconz log (help / debug view) with enabling "ddf", "info" and "info_l2" [...]

When switching on all 3 there's **a lot** going on.
One recurring thing is the following:
>19:52:54:831 Force read attributes for ZHALightLevel SensorNode LightLevel 4
>19:52:54:832 Force binding of attribute reporting for node LightLevel 4

When switching on only DDF in debug view i see the following also recurring:
>19:50:46:682 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false
>19:58:21:881 a4:c1:38:01:13:e6:33:1f-01-0400/state/lux expression: const tholddark = R.item('config/tholddark').val;
const tholdoffset = R.item('config/tholdoffset').val;
const lux = Attr.val;
let ll = 0;
if (lux > 0 && lux < 0xffff) {
    ll = Math.round(10000 * Utils.log10(lux) + 1);
}
R.item('state/lightlevel').val = ll;
R.item('state/dark').val = ll <= tholddark;
R.item('state/daylight').val = ll >= tholddark + tholdoffset;
Item.val = lux; --> 0

 [...] and try to trigger a presence, you will see some line like
> 
> ```
>         DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",
>                    rt, seq, dpid, dataType, dataLength, num.s32);
> ```

This line sadly appears not to show up.

Smanar, 2022-10-15:
>This line sadly appears not to show up.

This line appear with flag "info", and start by "TY_DATA".
You have it probably around thoses ones

`19:50:46:682 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false`

On this one the device have trigger the presence dp, and is probably reversed.

llugo, 2022-10-15:
Ok, this is what i get when i have INFO switch enabled: 
>21:58:55:221 TY_DATA_REPORT: seq 56, dpid: 0x04, type: 0x02, length: 4, val: 100
21:58:55:223 TY_DATA_REPORT: seq 56, dpid: 0x04, type: 0x02, length: 4, val: 100
21:58:55:458 TY_DATA_REPORT: seq 57, dpid: 0x09, type: 0x04, length: 1, val: 2
21:58:55:460 TY_DATA_REPORT: seq 57, dpid: 0x09, type: 0x04, length: 1, val: 2
21:58:55:679 TY_DATA_REPORT: seq 58, dpid: 0x0A, type: 0x04, length: 1, val: 0
21:58:55:681 TY_DATA_REPORT: seq 58, dpid: 0x0A, type: 0x04, length: 1, val: 0
21:59:03:816 TY_DATA_REPORT: seq 59, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:03:817 TY_DATA_REPORT: seq 59, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:13:844 TY_DATA_REPORT: seq 60, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:13:851 TY_DATA_REPORT: seq 60, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:14:246 TY_DATA_REPORT: seq 61, dpid: 0x01, type: 0x04, length: 1, val: 1
21:59:14:249 TY_DATA_REPORT: seq 61, dpid: 0x01, type: 0x04, length: 1, val: 1
21:59:14:448 TY_DATA_REPORT: seq 62, dpid: 0x0C, type: 0x02, length: 4, val: 9
21:59:14:463 TY_DATA_REPORT: seq 62, dpid: 0x0C, type: 0x02, length: 4, val: 9
21:59:14:664 TY_DATA_REPORT: seq 63, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:14:665 TY_DATA_REPORT: seq 63, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:14:886 TY_DATA_REPORT: seq 64, dpid: 0x09, type: 0x04, length: 1, val: 2
21:59:14:888 TY_DATA_REPORT: seq 64, dpid: 0x09, type: 0x04, length: 1, val: 2
21:59:15:107 TY_DATA_REPORT: seq 65, dpid: 0x0A, type: 0x04, length: 1, val: 0
21:59:15:111 TY_DATA_REPORT: seq 65, dpid: 0x0A, type: 0x04, length: 1, val: 0
21:59:23:964 TY_DATA_REPORT: seq 66, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:23:966 TY_DATA_REPORT: seq 66, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:33:974 TY_DATA_REPORT: seq 67, dpid: 0x04, type: 0x02, length: 4, val: 100
21:59:33:976 TY_DATA_REPORT: seq 67, dpid: 0x04, type: 0x02, length: 4, val: 100

Smanar, 2022-10-16:
>21:58:55:221 TY_DATA_REPORT: seq 56, dpid: 0x04, type: 0x02, length: 4, val: 100

This one is the battery, have added it in the DDF

>21:58:55:458 TY_DATA_REPORT: seq 57, dpid: 0x09, type: 0x04, length: 1, val: 2

This one is the sensitivity

>21:58:55:679 TY_DATA_REPORT: seq 58, dpid: 0x0A, type: 0x04, length: 1, val: 0

Something to see with a timer ...

>21:59:14:246 TY_DATA_REPORT: seq 61, dpid: 0x01, type: 0x04, length: 1, val: 1

This one is the presence

>22:10:03:324 TY_DATA_REPORT: seq 39, dpid: 0x0C, type: 0x02, length: 4, val: 9
>22:10:03:337 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false

Fuck, this one is the lux, I reversed them.

New DDF to test

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_3towulqd",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "ZG-204ZL",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/presence",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val == 1;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/dark"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lux",
          "parse": {
            "fn": "tuya",
            "dpid": 12,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js"
          },
          "read": {
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

For exemple on logs 
>>21:59:14:246 TY_DATA_REPORT: seq 61, dpid: 0x01, type: 0x04, length: 1, val: 1

this one need to switch from 1 to 0 according to presence.

llugo, 2022-10-16:
Ok, this is what i get from the api using the modified DDF:  
 >"4": {
        "config": {
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "etag": "9b7fcecfcb2b079c2778c39037456129",
        "lastannounced": "2022-10-16T11:31:53Z",
        "lastseen": "2022-10-16T11:31Z",
        "manufacturername": "_TZE200_3towulqd",
        "modelid": "TS0601",
        "name": "LightLevel 4",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2022-10-16T11:32:10.577",
            "lightlevel": 1,
            "lux": 1,
            "presence": false
        },
        "type": "ZHALightLevel",
        "uniqueid": "a4:c1:38:01:13:e6:33:1f-01-0400"
    },

accompanying debug log (INFO and DDF being switched on):
> 13:34:40:576 TY_DATA_REPORT: seq 65, dpid: 0x01, type: 0x04, length: 1, val: 0
13:34:40:578 TY_DATA_REPORT: seq 65, dpid: 0x01, type: 0x04, length: 1, val: 0
13:34:42:372 TY_DATA_REPORT: seq 66, dpid: 0x0C, type: 0x02, length: 4, val: 1000
13:34:42:386 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false
13:34:42:388 TY_DATA_REPORT: seq 66, dpid: 0x0C, type: 0x02, length: 4, val: 1000
13:34:49:023 TY_DATA_REPORT: seq 67, dpid: 0x04, type: 0x02, length: 4, val: 100
13:34:49:025 TY_DATA_REPORT: seq 67, dpid: 0x04, type: 0x02, length: 4, val: 100
13:34:50:451 a4:c1:38:01:13:e6:33:1f-01-0400/state/lux expression: const tholddark = R.item('config/tholddark').val;
const tholdoffset = R.item('config/tholdoffset').val;
const lux = Attr.val;
let ll = 0;
if (lux > 0 && lux < 0xffff) {
    ll = Math.round(10000 * Utils.log10(lux) + 1);
}
R.item('state/lightlevel').val = ll;
R.item('state/dark').val = ll <= tholddark;
R.item('state/daylight').val = ll >= tholddark + tholdoffset;
Item.val = lux; --> 0
13:34:50:457 TY_DATA_REPORT: seq 68, dpid: 0x01, type: 0x04, length: 1, val: 0
13:34:50:459 TY_DATA_REPORT: seq 68, dpid: 0x01, type: 0x04, length: 1, val: 0
13:34:50:726 TY_DATA_REPORT: seq 69, dpid: 0x0C, type: 0x02, length: 4, val: 1000
13:34:50:739 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false
13:34:50:741 TY_DATA_REPORT: seq 69, dpid: 0x0C, type: 0x02, length: 4, val: 1000
13:34:50:948 TY_DATA_REPORT: seq 70, dpid: 0x04, type: 0x02, length: 4, val: 100
13:34:50:950 TY_DATA_REPORT: seq 70, dpid: 0x04, type: 0x02, length: 4, val: 100
13:34:51:170 TY_DATA_REPORT: seq 71, dpid: 0x09, type: 0x04, length: 1, val: 2
13:34:51:172 TY_DATA_REPORT: seq 71, dpid: 0x09, type: 0x04, length: 1, val: 2
13:34:51:407 TY_DATA_REPORT: seq 72, dpid: 0x0A, type: 0x04, length: 1, val: 0
13:34:51:409 TY_DATA_REPORT: seq 72, dpid: 0x0A, type: 0x04, length: 1, val: 0

Smanar, 2022-10-16:
Still missing the config/battery
Can you make a "hot reload" in the DDF editor, or re-include the device (without deleting, like that all will be invisible)

>13:34:42:388 TY_DATA_REPORT: seq 66, dpid: 0x0C, type: 0x02, length: 4, val: 1000

So the value is 1000
If I make a similutation need to have
lux = 1000
lightlevel = 30001

Why you have 0 as value ? It was working previously https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6158#issuecomment-1206440580

What is your deconz version ?

>13:34:50:457 TY_DATA_REPORT: seq 68, dpid: 0x01, type: 0x04, length: 1, val: 0

And you never have 1 as val for this debug line ? (It's the presence detection)

llugo, 2022-10-16:
>Still missing the config/battery
Can you make a "hot reload" in the DDF editor, or re-include the device (without deleting, like that all will be invisible)

When viewing the preview in the ddf editor after opening the ddf and doing a hot-reload the lines concerning battery won't show up.

>Why you have 0 as value ? It was working previously https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6158#issuecomment-1206440580

That was a post by Tscheurgen

>What is your deconz version ?

v.2.18.2

>And you never have 1 as val for this debug line ? (It's the presence detection)

Just got this:

>14:47:27:098 TY_DATA_REPORT: seq 133, dpid: 0x01, type: 0x04, length: 1, val: 1

Smanar, 2022-10-16:
>When viewing the preview in the ddf editor after opening the ddf and doing a hot-reload the lines concerning battery won't show up. I'm probalbly missing something here.

You are using HA ?

>That was a post by Tscheurgen

Yep but he was using the same DDF, with reversed DP

>4:47:27:098 TY_DATA_REPORT: seq 133, dpid: 0x01, type: 0x04, length: 1, val: 1

Ha nice ^^
So it mean the presence is working ? If  I m right you have
```

TY_DATA_REPORT: seq 133, dpid: 0x01, type: 0x04, length: 1, val: 1
a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 1; --> true
```

and

```
TY_DATA_REPORT: seq 133, dpid: 0x01, type: 0x04, length: 1, val: 0
a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 1; --> false
```

llugo, 2022-10-17:
>When viewing the preview in the ddf editor after opening the ddf and doing a hot-reload the lines concerning battery won't show up. I'm probalbly missing something here.

Something went wrong when adding your last modified DDF. I fixed it now and battery is showing up now.

> You are using HA ?

No i'm only using Phoscon (and Postman for REST API calls).

>So it mean the presence is working ? If I m right you have [...]

Yes, that's working now!

Smanar, 2022-10-18:
>edit: i guess it would help to learn how the presence detector works physically within the device

I m not sure to understand. It's a presence sensor, so it detect a presence and stay true during a periode after a detection (if there is a new detection, the timer restart)

llugo, 2022-10-18:
> ... the timer restart)

I wonder what the timer is set to.

This is me getting into the room i've left 5 min ago:

>19:51:29:060 TY_DATA_REPORT: seq 73, dpid: 0x0C, type: 0x02, length: 4, val: 112
19:51:43:642 a4:c1:38:01:13:e6:33:1f-01-0400/state/lux expression: const tholddark = R.item('config/tholddark').val;
const tholdoffset = R.item('config/tholdoffset').val;
const lux = Attr.val;
let ll = 0;
if (lux > 0 && lux < 0xffff) {
    ll = Math.round(10000 * Utils.log10(lux) + 1);
}
R.item('state/lightlevel').val = ll;
R.item('state/dark').val = ll <= tholddark;
R.item('state/daylight').val = ll >= tholddark + tholdoffset;

Opening the laptop, so lux increases:

>Item.val = lux; --> 136
19:51:43:650 TY_DATA_REPORT: seq 74, dpid: 0x0C, type: 0x02, length: 4, val: 136
19:51:43:652 TY_DATA_REPORT: seq 74, dpid: 0x0C, type: 0x02, length: 4, val: 136
19:51:43:654 TY_DATA_REPORT: seq 74, dpid: 0x0C, type: 0x02, length: 4, val: 136
19:51:43:937 TY_DATA_REPORT: seq 75, dpid: 0x01, type: 0x04, length: 1, val: 1
19:51:43:958 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 1; --> true
19:51:43:960 TY_DATA_REPORT: seq 75, dpid: 0x01, type: 0x04, length: 1, val: 1
19:51:43:961 TY_DATA_REPORT: seq 75, dpid: 0x01, type: 0x04, length: 1, val: 1

Presence gets set to false merely 5 sec later

>19:51:48:930 TY_DATA_REPORT: seq 76, dpid: 0x01, type: 0x04, length: 1, val: 0
19:51:48:959 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 1; --> false
19:51:48:962 TY_DATA_REPORT: seq 76, dpid: 0x01, type: 0x04, length: 1, val: 0
19:51:48:964 TY_DATA_REPORT: seq 76, dpid: 0x01, type: 0x04, length: 1, val: 0

Smanar, 2022-10-18:
The timer is the dp 10.

>21:58:55:679 TY_DATA_REPORT: seq 58, dpid: 0x0A, type: 0x04, length: 1, val: 0

If I have understand
0 -> 10s
1 -> 30s
2 -> 60s
3 -> 120s

>20:00:48:564 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 1; --> true
>20:00:48:885 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 1; --> false

Hu, right, relay too fast to go from true to false, I have not reversed them ?

Edit:

>Presence gets set to false merely 5 sec later

So Yes I think I have reversed them again ^^
try 

`"eval": "Item.val = Attr.val == 0;"`

For the state/presence.

llugo, 2022-10-18:
hrm, i cannot make any sense of it 🤷 

now with reversed attr.val
>22:04:37:279 TY_DATA_REPORT: seq 88, dpid: 0x0C, type: 0x02, length: 4, val: 90
22:04:37:280 TY_DATA_REPORT: seq 88, dpid: 0x0C, type: 0x02, length: 4, val: 90
22:04:37:281 TY_DATA_REPORT: seq 88, dpid: 0x0C, type: 0x02, length: 4, val: 90
22:04:37:569 TY_DATA_REPORT: seq 89, dpid: 0x01, type: 0x04, length: 1, val: 1
22:04:37:596 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false
22:04:37:598 TY_DATA_REPORT: seq 89, dpid: 0x01, type: 0x04, length: 1, val: 1
22:04:37:599 TY_DATA_REPORT: seq 89, dpid: 0x01, type: 0x04, length: 1, val: 1
22:04:41:180 TY_DATA_REPORT: seq 90, dpid: 0x01, type: 0x04, length: 1, val: 0
22:04:41:207 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> true
22:04:41:209 TY_DATA_REPORT: seq 90, dpid: 0x01, type: 0x04, length: 1, val: 0
22:04:41:210 TY_DATA_REPORT: seq 90, dpid: 0x01, type: 0x04, length: 1, val: 0

Smanar, 2022-10-18:
Make the same tries than before

- presence is  false*
- enter the room after 5s presence go to True
- stay in the room
- Presence stay true
- Leave the room
- Presence go to false after 10s

llugo, 2022-10-21:
Nor sure about the exact seconds, but it's working as expected! 
getting it out of the closet:
19:30:53:096 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> true
putting it back in to the closet 10 sec later
19:31:29:598 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> false
Getting it back out:
19:34:25:084 a4:c1:38:01:13:e6:33:1f-01-0400/state/presence expression: Item.val = Attr.val == 0; --> true
Finally my API output:
    "4": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "etag": "b89c9824dc46b7b9196ff476daec7f75",
        "lastannounced": "2022-10-20T19:13:40Z",
        "lastseen": "2022-10-21T17:34Z",
        "manufacturername": "_TZE200_3towulqd",
        "modelid": "TS0601",
        "name": "LightLevel 4",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2022-10-21T17:34:36.438",
            "lightlevel": 9032,
            "lux": 8,
            "presence": true
        },
        "type": "ZHALightLevel",
        "uniqueid": "a4:c1:38:01:13:e6:33:1f-01-0400"
    },

So Smanar, i think all is working well. Just wanted to say thanks at this point!
Now this just needs to be supported by Phoscon and i am happy as can be 👯

Smanar, 2022-10-21:
Ha ?
This device don't work natively with phoscon ?
Ha, perhaps because the state/presence is in the ZHAlightlevel, it's probalby possible to cheat, and create 2 sensors with this DDF 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_3towulqd",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "ZG-204ZL",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/dark"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lux",
          "parse": {
            "fn": "tuya",
            "dpid": 12,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js"
          },
          "read": {
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/presence",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val == 0;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

But need to delete previous entry, sensors will be different, state/presence is removed from the last one and added on a new one compatible with phoscon

And sure I can make the PR.

llugo, 2022-10-21:
>But need to delete previous entry

You mean delete the node in deCONZ, Or deleting prior DDFs?

Smanar, 2022-10-22:
You can just delete the device in phoscon it will remove it in the API and the zigbee network.
And I hope you have only 1 DDF for this device ?

WhoTheHeck, 2022-10-23:
Thanks a lot for your work @Smanar! I just got 2 of these sensors and was able to successfully integrate them into HA using deconz. Presence, light and battery all work. Not quite sure about the accuracy of the light level, but I just need some numerical triggers based on it anyway.

Smanar, 2022-10-23:
For the accuracy, I don't know the value you need to have, I have started something here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6158#issuecomment-1206730807

You just need to edit the line with "eval"

If the device is working, tell me if I can make the PR ?

WhoTheHeck, 2022-10-23:
Well, I don't need a specific value. I just looked at what values I get when it is light and when it is dark and chose my triggers accordingly. If I have the ddf file in the \tuya folder, I get light values in the range of up to 1000 when I use the script as in your last ddf file. The downside is that this folder is not persistent. But for now it works for me. Guess that makes it good to go for a PR?

I have now tried putting the file in the persistent folder /.local/.... and try the following instead of the script:
`"eval": "Measured value= (Math.round(10000 * Math.log10(Reallux) + 1))/20;`
but with this I don't get any more updates in HA. I'm not sure what the issue is, but it's not a big deal to me right now.

Smanar, 2022-10-23:
Ok, I make the PR

WhoTheHeck, 2022-10-24:
using the HA deconz addon files saved in `/data/.local/share/dresden-elektronik/deCONZ/devices/` are persistent, surviving container restarts. Based on https://community.home-assistant.io/t/permanent-changes-in-deconz-addon-container-survive-restart-update/443754/12

Smanar, 2022-10-24:
But I have made the try with an user the data folder contain only the "otau" folder and a file options.json.

WhoTheHeck, 2022-10-24:
The .local is a hidden folder. Is that maybe the issue? in the VNC server when opening a ddf-file the folder `/data/.local/share/dresden-elektronik/deCONZ/devices/` is always the default one that opens for me. When accessing it with the ssh-addon I also dont see the folder with e.g. `ls` but can access it.

Smanar, 2022-10-24:
Nice, thx a lot, I will add it :)

llugo, 2022-10-27:
So i did the following:
Update deCONZ to 2.19.1 and restart deconz-gui
Add file "__TZE200_3towulqd.json" to "/usr/share/deCONZ/devices/tuya/"
Delete node in deCONZ
Reset device
Node gets visible in deCONZ
Load above mentioned json within deconz DDF editor
Result is the following:

[image]

So since there's a question mark right beside "HA" (with orange background), I guess the device doesn't get recognized as a sensor?

>This device don't work natively with phoscon ?
I was told that it is "perfectly normal" most of the time for Phoscon not showing devices recognized by deCONZ since Phosocon "is closed source and needs devices to be added manually by their developer."

Seems like i need to switch to Home Assistant...

@WhoTheHeck does your device get recognized as a sensor in the deconz endpoint menu?

Smanar, 2022-10-28:
>So since there's a question mark right beside "HA" (with orange background), I guess the device doesn't get recognized as a sensor?

It's ok, unreconised device have a node name like 0xXXXX
The "?" is just an icon, some device have a special icon, but deconz can reconize your, it use same cluster than a door sensor for exemple, for me your device don't have cluster specific enought for deconz choose an icon (or just don't have one for this kind of device)

But in my mind support for classic sensor is native in phoscon, I don't see a reason for this one was not visible.

>i just realized that the node is not listed in deconz "node list" panel

This is not normal too.

WhoTheHeck, 2022-10-31:
not sure why things are behaving differently on your side.

> @WhoTheHeck does your device get recognized as a sensor in the deconz endpoint menu? I mean do you have a question mark symbol there too?
mine looks basically the same, but without the questionmark. For me there is just no symbol there.

> i just realized that the node is not listed in deconz "node list" panel for whatever reason..
for me they do show up in node list.

llugo, 2022-11-04:
>>i just realized that the node is not listed in deconz "node list" panel

>This is not normal too.

After moving the device closer to the configuration tool (RaspBee II) and reconnecting it, it does now appear in the deCONZ node list!

> But in my mind support for classic sensor is native in phoscon, I don't see a reason for this one was not visible.

Well it still does not show up in Phoscon. Maybe i'll just have to wait for the next Phoscon update. 
Switching to HA seems cumbersome since i have other services running on the Pi and i don't want to use it only for HA...

Smanar, 2022-11-05:
>Well it still does not show up in Phoscon. Maybe i'll just have to wait for the next Phoscon update.

IDK, perhaps just ask on the forum, seriously I don't see why phoscon will use black/white list for that.

>Switching to HA seems cumbersome since i have other services running on the Pi and i don't want to use it only for HA

I understand and I m agree, but there is other third app, realy lighter than HA, idk how fast phoscon will evolve in the future ....

llugo, 2022-11-06:
> But in my mind support for classic sensor is native in phoscon, I don't see a reason for this one was not visible.

Oh my.. i just found out that the "Presence 8"  device is indeed not visible in Phsoscons device list... HOWEVER if you add a sensor control option for a group of lights "Presence 8" is a selectable device! 

Sorry for the confusion and not trying this before...

So the device is now working perfectly and lights go on when presence is detected! Yaay!

Thousand thanks again Smanar and thank you too WhoTheHeck!

llugo, 2022-12-02:
One question.
The device checks for presence about every 30s or so. Is it possible to set a different duration?

Smanar, 2022-12-03:
Sure can you try to add for the ZHAPresence

```
                {
                    "name": "config/duration",
                    "write": {"fn": "tuya", "dpid": 10,  "dt": "0x2b",  "eval": "Item.val"},
                    "parse": {"fn": "tuya", "dpid": 10, "eval": "Item.val = Attr.val;" },
                    "read": {"fn": "none"},
                    "default": 0
                },
```
Available values are 10, 30, 60 and 120

llugo, 2022-12-04:
> The timer is the dp 10.
> 
> > 21:58:55:679 TY_DATA_REPORT: seq 58, dpid: 0x0A, type: 0x04, length: 1, val: 0
> 
> If I have understand 0 -> 10s 1 -> 30s 2 -> 60s 3 -> 120s

> Sure can you try to add for the ZHAPresence
>
> ```
>                 {
>                     "name": "config/duration",
>                     "write": {"fn": "tuya", "dpid": 10,  "dt": "0x2b",  "eval": [...]

First thank you again for helping out.

So I guess the magic happens in the "write" brackets. But i'm not sure which part of it sets the value to 0, 1, 2 or 3.
What does "dt" stand for? I guess it's a hex value and set to 43 in decimal?

Smanar, 2022-12-04:
>If I have understand 0 -> 10s 1 -> 30s 2 -> 60s 3 -> 120s

Ha yes, you are right, have forget that, so the type is wrong too, need to use 0x30
dt is the type

 ```
         bool           0x10
          s32 value      0x2b
          enum           0x30
          8-bit bitmap   0x18
          16-bit bitmap  0x19
          32-bit bitmap  0x1b
```

>edit: in the DDF editor you don't really "see" the "write part", right?

No, the support is not finished for tuya, need to use a text editor

So better to use this DDF, it will convert the value and use the good type.

```
                {
                    "name": "config/duration",
                    "write": {"fn": "tuya", "dpid": 10,  "dt": "0x30",  "eval": "[10, 30, 60, 120].indexOf(Item.val)" },
                    "parse": {"fn": "tuya", "dpid": 10, "eval": "Item.val = [10, 30, 60, 120][Attr.val]" },
                    "read": {"fn": "none"},
                    "default": 0
                },
```

It's a little hacky because we are using config/duration that need to be a numeric value

BabaIsYou, 2023-01-08:
Hi,

@Smanar, what do you think about adding duration into the opened PR before final review ? :-)

Smanar, 2023-01-08:
If someone can test it ?
Because it's a little hacky ...

WhoTheHeck, 2023-01-08:
I can try. Although I don't really understand what it's supposed to do 😄 As long as I move in front of the sensor it reports movement, when I don't, after round about 10sec it changes state. How exactly should duration affect the behavior?
I added the lines to the DDF as so:

<details>
    <summary> full DDF </summary>

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_3towulqd",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "ZG-204ZL Luminance Motion Sensor",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark",
          "default": 12000
        },
        {
          "name": "config/tholdoffset",
          "default": 7000
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/dark"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lux",
          "parse": {
            "fn": "tuya",
            "dpid": 12,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js"
          },
          "read": {
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/presence",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val == 0;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/duration",
          "write": {"fn": "tuya", "dpid": 10,  "dt": "0x30",  "eval": "[10, 30, 60, 120].indexOf(Item.val)" },
          "parse": {"fn": "tuya", "dpid": 10, "eval": "Item.val = [10, 30, 60, 120][Attr.val]" },
          "read": {"fn": "none"},
          "default": 10
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```
</details>

This exposes a duration input number to Home Assistant, where I can enter any number. But it doesnt change the behavior.

Is it normal that the DDF Preview in deconz shows the section differently? There it looks like so:
[image]

Note that I tried another default in the DDF (10), but this also does not seem to be used.

Smanar, 2023-01-08:
>Is it normal that the DDF Preview in deconz shows the section differently? There it looks like so:

Yeah don't worry, it's just a display form.
Value than can be used are 10,30,60 and 120.

>But it doesnt change the behavior

The value is updated in the API ?
Can you show logs when trying to change the value, on deconz / help / debug view.
with flag "DDF" and "info" ( the log start by "TY_DATA_")

WhoTheHeck, 2023-01-08:
Not sure how to copy the debug log from the VNC viewer... Please have a look if the screenshot already gives enough verification. It seems to be working. When I enter a valid number in Home Assistant the log shows something like this:
[image]
In this example I wrote 60 and index 2 was correctly chosen.

And wrong numbers return -1
[image]

But still... how should the behavior of the sensor change?

Smanar, 2023-01-08:
There is something wrong, but all seem fine.
The dpid is good 0x0A (=10) the type is good too 0x30 (=0x04 after tuya convertion), tha value too (=2)

But some line after we can see the report from the device for the dpid 0x04 and the value is 0.

Perhaps if we can see raw value, visible if enabling "APS" and "APS_l2", let he flag "info" else it will be not possible to locate the line ^^.

The interesting part will be just after the "write tuya data", we will see the raw request made by deconz to the device.

WhoTheHeck, 2023-01-08:
ok, there is quite a lot going on. 😄 
I tried to time it correctly, but unsure if it contains the correct moment. "APS_L2" doesnt give anything. But have a look if this APS snapshot contains anything helpful:
[image]

Smanar, 2023-01-09:
`11 b700 00 0a 0a 04 00 01 02`

Bad luck the request if fine ...
On the second capture the request is the request 72, and there is no success confirm for this one.
I realy have no clue to explain why it's not working.

>also: 0 is the default, but the list [10...120] doesnt include it. Is that wished?

Nope ^^, the tuya request support value 0 1 2 3, but value need to be 10, 30, 60, 120.
But, it's just a defaut vlaue, the first time the device send his own value, it need to be updated.
And this is strange because the device make lot of report for it with the 0 value so the value in the API need to be 10.
So the return is not working too.