dresden-elektronik/deconz-rest-plugin #6160: Hue tap dial switch

ebaauw, 2022-07-01:
## Device

- Product name: Hue tap dial switch
- Manufacturer: Signify Netherlands B.V.
- Model identifier: RDM002
- Device type : Please remove all unrelated device types. 
  - Remote

As mentioned on Discord, the device seems to be the love child of the Hue dimmer switch and the Lutron Aurora (#2305).  Unlike the Aurora, it pairs with deCONZ immediately, and It reports button/rotary-events on 0xFC00 without any configuration.  It is using the same [scheme](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2305#issuecomment-584359012) as the Aurora for rotary events.  Like the Hue in-wall module, it doesn't seem to support hybrid mode (reporting on 0xFC00 and sending _On/Off_ and _Level Control_ commands at the same time),despite the client clusters, even after writing 0x0B to 0x0000/0x0031.  It won't be able to control the lights directly.

The device feels well constructed and pleasantly heavy.  However it's not quite heavy enough to move the dial while it's laying on the table: the whole device moves as there's not enough friction between the switch and the table.  When pressing down a bit while dialling, it works fine, though.  Like the Hue dimmer switch, the buttons give good feedback when pressed, as does the dial when turned.

Exposing the buttons and battery will be straightforward.  For the dial, we'd need to introduce a ZHARelativeRotary `/sensors` type.  And we'll need a trick to handle the `swversion`, probably just faking another endpoint, cf. the IKEA Starkvind.

## Screenshots
[image]
[image]

### Basic
[image]
### Power Configuration
[image]
### Hue
[image]

## Hue API
As with the Aurora, the Hue bridge exposes it as two `sensors` resources.  A `ZLLRelativeRotary` for the dial:
```json
{
  "state": {
    "rotaryevent": 2,
    "expectedrotation": 90,
    "expectedeventduration": 400,
    "lastupdated": "2022-07-01T14:43:50"
  },
  "swupdate": {
    "state": "noupdates",
    "lastinstall": "2022-07-01T14:38:51"
  },
  "config": {
    "on": true,
    "battery": 100,
    "reachable": true,
    "pending": []
  },
  "name": "Hue tap dial switch 1",
  "type": "ZLLRelativeRotary",
  "modelid": "RDM002",
  "manufacturername": "Signify Netherlands B.V.",
  "productname": "Hue tap dial switch",
  "diversityid": "e5b68d4b-6249-436d-88c5-7e02ffabda37",
  "swversion": "2.59.19",
  "uniqueid": "00:17:88:01:0d:13:1f:97-01-fc00-0014",
  "capabilities": {
    "certified": true,
    "primary": true,
    "inputs": [
      {
        "repeatintervals": [
          400
        ],
        "events": [
          {
            "rotaryevent": 1,
            "eventtype": "start"
          },
          {
            "rotaryevent": 2,
            "eventtype": "repeat"
          }
        ]
      }
    ]
  }
}
```
and a `ZLLSwitch` for the buttons:
```json
{
  "state": {
    "buttonevent": 3002,
    "lastupdated": "2022-07-01T14:47:39"
  },
  "swupdate": {
    "state": "noupdates",
    "lastinstall": "2022-07-01T14:38:51"
  },
  "config": {
    "on": true,
    "battery": 100,
    "reachable": true,
    "pending": []
  },
  "name": "Hue tap dial switch 2",
  "type": "ZLLSwitch",
  "modelid": "RDM002",
  "manufacturername": "Signify Netherlands B.V.",
  "productname": "Hue tap dial switch",
  "diversityid": "e5b68d4b-6249-436d-88c5-7e02ffabda37",
  "swversion": "2.59.19",
  "uniqueid": "00:17:88:01:0d:13:1f:97-01-fc00",
  "capabilities": {
    "certified": true,
    "primary": false,
    "inputs": [
      {
        "repeatintervals": [
          800
        ],
        "events": [
          {
            "buttonevent": 1000,
            "eventtype": "initial_press"
          },
          {
            "buttonevent": 1001,
            "eventtype": "repeat"
          },
          {
            "buttonevent": 1002,
            "eventtype": "short_release"
          },
          {
            "buttonevent": 1003,
            "eventtype": "long_release"
          },
          {
            "buttonevent": 1010,
            "eventtype": "long_press"
          }
        ]
      },
      {
        "repeatintervals": [
          800
        ],
        "events": [
          {
            "buttonevent": 2000,
            "eventtype": "initial_press"
          },
          {
            "buttonevent": 2001,
            "eventtype": "repeat"
          },
          {
            "buttonevent": 2002,
            "eventtype": "short_release"
          },
          {
            "buttonevent": 2003,
            "eventtype": "long_release"
          },
          {
            "buttonevent": 2010,
            "eventtype": "long_press"
          }
        ]
      },
      {
        "repeatintervals": [
          800
        ],
        "events": [
          {
            "buttonevent": 3000,
            "eventtype": "initial_press"
          },
          {
            "buttonevent": 3001,
            "eventtype": "repeat"
          },
          {
            "buttonevent": 3002,
            "eventtype": "short_release"
          },
          {
            "buttonevent": 3003,
            "eventtype": "long_release"
          },
          {
            "buttonevent": 3010,
            "eventtype": "long_press"
          }
        ]
      },
      {
        "repeatintervals": [
          800
        ],
        "events": [
          {
            "buttonevent": 4000,
            "eventtype": "initial_press"
          },
          {
            "buttonevent": 4001,
            "eventtype": "repeat"
          },
          {
            "buttonevent": 4002,
            "eventtype": "short_release"
          },
          {
            "buttonevent": 4003,
            "eventtype": "long_release"
          },
          {
            "buttonevent": 4010,
            "eventtype": "long_press"
          }
        ]
      }
    ]
  }
}
``` 
I'm not sure what the `long_press` x010 button events are; haven't seen these before, and the device doesn't seem to send them.

ebaauw, 2022-07-03:
Above PR adds support.  The `ZHASwitch` resource is very familiar:
```
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true
  },
  "etag": "fb844e55b3f8e374db8944d75b65b25c",
  "lastannounced": "2022-07-03T13:57:59Z",
  "lastseen": "2022-07-03T14:02Z",
  "manufacturername": "Signify Netherlands B.V.",
  "mode": 1,
  "modelid": "RDM002",
  "name": "RDM002 43",
  "state": {
    "buttonevent": 4002,
    "eventduration": 1,
    "lastupdated": "2022-07-03T11:37:09.362"
  },
  "swversion": "2.59.19",
  "type": "ZHASwitch",
  "uniqueid": "00:17:88:01:0d:13:09:7b-01-fc00"
}
```
The `ZHARelativeRotary` resource is modelled after the Hue API:
```
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true
  },
  "etag": "463728970bdb7d04048fc4373654f45a",
  "lastannounced": "2022-07-03T13:57:59Z",
  "lastseen": "2022-07-03T14:02Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "RDM002",
  "name": "RDM002 44",
  "state": {
    "expectedeventduration": 400,
    "expectedrotation": 75,
    "lastupdated": "2022-07-03T11:37:49.586",
    "rotaryevent": 2
  },
  "swversion": "2.59.19",
  "type": "ZHARelativeRotary",
  "uniqueid": "00:17:88:01:0d:13:09:7b-14-fc00"
}
```
`rotaryevent` is 1 for new movements (start) and 2 for repeat movements.  `expectedrotation` reports the angle, positive for clockwise, negative for counter-clockwise.  I'm not sure what scale it uses.  While turning, a rorary event is issued every 400ms (the `expectedeventduration`):
```
Jul 03 16:07:59 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedeventduration":400,"expectedrotation":15,"lastupdated":"2022-07-03T14:07:59.382","rotaryevent":1}
Jul 03 16:07:59 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":105,"lastupdated":"2022-07-03T14:07:59.787","rotaryevent":2}
Jul 03 16:08:00 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":211,"lastupdated":"2022-07-03T14:08:00.193"}
Jul 03 16:08:00 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":256,"lastupdated":"2022-07-03T14:08:00.595"}
Jul 03 16:08:01 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":286,"lastupdated":"2022-07-03T14:08:00.996"}
Jul 03 16:08:09 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":0,"lastupdated":"2022-07-03T14:08:09.486","rotaryevent":1}
Jul 03 16:08:09 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":-30,"lastupdated":"2022-07-03T14:08:09.598"}
Jul 03 16:08:09 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":-166,"lastupdated":"2022-07-03T14:08:09.977","rotaryevent":2}
Jul 03 16:08:10 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":-226,"lastupdated":"2022-07-03T14:08:10.374"}
Jul 03 16:08:10 pi5 dc_eventlog[692]: /sensors/46/state: {"expectedrotation":-271,"lastupdated":"2022-07-03T14:08:10.801"}
```

closed 2022-07-16 by manup

hjri, 2022-08-07:
for some reason my Hue Tap Dial barely wants to work for me - it doesn't start blinking orange after holding button 1 for some time but it does get onto the network, reads either don't work or work slowly, the amount of elements in "Philips Specific" section in basic cluster varies with each pair attempt, nothing seem to connect to the switch or maybe 1 or 2 other devices, it also reports battery as 0 even with fresh battery.

I really don't know if my unit is defective or i'm doing something wrong.

SwoopX, 2022-08-07:
Are you sure you copied all files from the respective PR correctly?

hjri, 2022-08-07:
no, i just updated to 2.18.0 beta which should have the changes from this MR

hjri, 2022-09-19:
Ok so my tap dial had borked firmware that made it mostly unresponsive, luckily i managed to update firmware with deconz just fine, now it blinks when i push buttons and rotate dial, and i see those events in Hue cluster in deconz but i guess those events aren't exported anywhere yet - i don't see events in node-red at least even when watching for all the events.

HKUser1, 2022-11-25:
Hello Same for me, Tap events are ok, Dial events does not work. For lights The Rotary Function to increase and decrease 
can be configured with the Hue essentials app. But it wood be nice to control for example a group of RollerShutters with the Rotary button