dresden-elektronik/deconz-rest-plugin #6171: TS011F _TZ3000_typdpdpg - 20A EU Smart Plug Tuya Zigbee3.0 Remote Power Socket With Energy Monitoring Function

phwagner82, 2022-07-04:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: _TZ3000_typdpdpg
- Model identifier: TS011F
- Device type : 
  - Switch

## Screenshots
[image]

[image]

Smanar, 2022-07-04:
Hello can try with this complete DDF, you probably need to adjust values convertions

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_typdpdpg",
  "modelid": "TS011F",
  "vendor": "Tuya",
  "product": "Another tuya plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val * 10;"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

phwagner82, 2022-07-14:
Hi Smanar, 
sorry, but my skills to read the ddf are not good enough. i cant adjusting values convertions.

Kind regards
Philipp

Smanar, 2022-07-14:
I can adjust them, just tell me the value you have and the one you need to have (to know if I need to *10 for exemple)

phwagner82, 2022-08-08:
Hello Smanar.
I need the default values for plug, current energy, total energy, voltage, power, ... like other Tuya plugs.

Thanks in advance
Philip

Smanar, 2022-08-09:
And none of them are working ?
Can you show the device json ? (The sensor part on phoscon / help / api information / sensors)

closed 2022-09-07 by github-actions[bot]