dresden-elektronik/deconz-rest-plugin #6173: Aqara presence sensor fp1 (lumi.motion.ac01)

stefanuytterhoeven, 2022-07-05:
what is the status of this sensor?
my phoscon version is the latest version : 2.17.01 / 1-7-2022  with firmware : 26720700

Phoscon gives me this:

Vendor aqara
Product lumi.motion.ac01
Version 20210121
Motion detected
Refreshed 07/05/2022 3:58:12 PM

in Home assistant: I have only the "motion" entity?

looking around here, I saw you guys were talking about "near, far"... Also, it should give some kind of "direction" of where to movement is going.

Are you still trying to figure it out? (no critisicm, just asking...)

Does it generate events? 'Cause I could find an event.

thanks for your work !

ps: the sensor just arrived here. So, I'm trying to figure out what i can do about it. I saw that Zigbee2mqtt already could give more data.

closed 2022-07-05 by close-issue-app[bot]

Kane610, 2022-07-05:
Implementation cycles of deconz and home assistant are not in sync. FP1 support will come at the earliest with HA 2022.8 as I just added support for it in pydeconz.

stefanuytterhoeven, 2022-07-05:
Ok. That explains it😄
Thanks! So I'll have to wait.