dresden-elektronik/deconz-rest-plugin #6180: Update to 2.17.01 - Aqara Motion sensors have lost Temperature sensor

evan-shaw, 2022-07-07:
## Describe the bug

## Steps to reproduce the behavior
 It is constant since the upgrade, for the past 3hrs.

## Expected behavior

## Screenshots

## Environment

[image]

![Selection_887](https://user-images.githubusercontent.
[image]

[image]
com/3452050
[image]

[image]
6/177771550-fca947e6-ef4f-419e-8f5c-a696968a7e55.png)

 - Host system: Raspberry Pi4B
 - Running method: Raspbian Buster 
 - Firmware version: 26720700
 - deCONZ version: 2.17.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

evan-shaw, 2022-07-07:
Hi Mimlix,

Thanks for the response, does this help: -

"8": {
            "config": {
                "battery": 100,
                "on": true,
                "reachable": true,
                "tholddark": 12000,
                "tholdoffset": 7000
            },
            "ep": 1,
            "etag": "c8ef78d1f38242dbb2c454d781dd56db",
            "lastannounced": null,
            "lastseen": "2022-07-07T13:36Z",
            "manufacturername": "LUMI",
            "modelid": "lumi.sensor_motion.aq2",
            "name": "FrontDoorMotionSensor",
            "state": {
                "dark": false,
                "daylight": true,
                "lastupdated": "2022-07-07T12:41:53.480",
                "lightlevel": 30001,
                "lux": 1000
            },
            "swversion": "20170627",
            "type": "ZHALightLevel",
            "uniqueid": "00:15:8d:00:03:3a:d8:49-01-0400"
        };

and

"9": {
            "config": {
                "battery": 100,
                "duration": 90,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "fa7989be29512a64fd19ec5c7e58a0be",
            "lastannounced": null,
            "lastseen": "2022-07-07T13:36Z",
            "manufacturername": "LUMI",
            "modelid": "lumi.sensor_motion.aq2",
            "name": "FrontDoorMotionSensor",
            "state": {
                "lastupdated": "2022-07-07T12:43:23.663",
                "presence": false
            },
            "swversion": "20170627",
            "type": "ZHAPresence",
            "uniqueid": "00:15:8d:00:03:3a:d8:49-01-0406"
        },

Thanks

Mimiix, 2022-07-07:
Is there a ZHATemperature for this device?

Mimiix, 2022-07-07:
It looks like the DDF is missing the Temperature part here. I think you should be able to solve it now by changing the state of the DDF to Draft : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Mimiix, 2022-07-07:
I've asked @SwoopX  to check in.

Mimiix, 2022-07-07:
I just tested it with my own. Re-paired on 2.17.01. @Thomas-Vos did the same on his environment. Here we see the temperature:

[image]

This seems to be a issue on your own environment i guess. I recommend re-pairing and see from there. As it's not a bug, i am closing this issue. If you still need some help, please use the forums. 

Kind regards

ebaauw, 2022-07-07:
Like most Xiaomi devices, these report a coarse-grained internal device temperature, which deCONZ exposes as `config.temperature`, and, for reasons beyond my comprehension, Phoscon shows the same way as a proper temperature sensor.  Looks like the `config` attribute is missing from the DDF.

bcutter, 2022-07-19:
> This seems to be a issue on your own environment i guess.

No it's not. Same happened for me: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6203

> I recommend re-pairing and see from there.

I repaired (soft-repairing without deletion) and it changed nothing.

> As it's not a bug, i am closing this issue.

Why do you think it's not a bug? There's evidence 2.17.1 broke it. Two issues exist on the same problem. If you can't replicate it, it's not automatically "no bug".

> If you still need some help, please use the forums.
It's an issue and GitHub is for issues, right.

closed 2022-09-09 by manup