dresden-elektronik/deconz-rest-plugin #6182: 2.17.1 update breaks Phoscon gateway page (no details shown)

bcutter, 2022-07-08:
## Describe the bug
Updated from 2.16.1 to 2.17.1 using Home Assistant addon (6.14.2 to 6.15.0).
Since the update the gateway settings page is almost empty:
Deleted browser cache, issue persits.

## Steps to reproduce the behavior
1. Update to 2.17.1
2. Turn to Phoscon gateway page

## Expected behavior
Details shown as before, including backup options (VERY important!)

## Screenshots
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26720700)
 - deCONZ version: (2.17.1)
 - Device: ConBee II
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? - It's clearly not about hardware here.

## deCONZ Logs
Logs

## Additional context
Interestingly, https://github.com/dresden-elektronik/deconz-rest-plugin/releases lists `Fix crash on Phoscon App gateway settings page.`.

bcutter, 2022-07-08:
Update: it seems to be a browser issue. Tested on several other systems/devices/browsers, page just loads fine.
More precisely: only the browser which surfed the page and initiated the update is affected.

I deleted the browser cache once again, force-fresh-reloaded the addon / phoscon page - everything is working now.

**Solved for me. Please close if that's acceptable to you. I'll keep it open for now just in case if you want to further investigate.**

Mimiix, 2022-07-08:
Hi,

Thank you for your report.

As this is not a bug report as described in #5113 I am closing this issue. Phoscon bug reports are collected at the forums. In your case, this already is reported.

https://forum.phoscon.de/t/phoscon-app-broken-with-deconz-2-17-0/2099/11

Kind regards,
Dennis

closed 2022-07-08 by Mimiix