dresden-elektronik/deconz-rest-plugin #6187: IKEA Symfonisk controller sends left and right turn events when rotated

jakemauer, 2022-07-10:
## Describe the bug
It seems that the Symfonisk controller sends both events for left/right movement regardless of what direction you spin the dial. 

## Steps to reproduce the behavior
1. Pair the Symfonisk controller.
2. View the DeConz GUI debug log and turn the dial one direction then stop.
3. Notice if there are multiple simultaneous "Move Up" and "Move Down" events.

## Expected behavior
Turning the dial left should fire only one movement direction event until movement stops.

## Screenshots
[image]

## Environment

 - Host system: Synology
 - Running method: DeConz and Home Assistant running via Docker on a Synology DS1019+
 - Firmware version: (26720700)
 - deCONZ version: (2.17.01)
 - Home Assistant (2022.7.2)
 - Symfonisk Controller Firmware (2.3.080)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes, 6ft)
 - No other USB or Serial devices connected.

## deCONZ Logs
These are the logs that resulted from me turning the dial to the left in a continuous motion for a few rotations.
```
16:43:11:463 [INFO] - Button 2001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Up, payload: 00C30000, zclSeq: 116
16:43:11:822 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 117
16:43:12:181 [INFO] - Button 3003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 118
16:43:12:287 [INFO] - Button 2001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Up, payload: 00C30000, zclSeq: 119
16:43:12:370 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 120
16:43:13:686 [INFO] - Button 2001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Up, payload: 00C30000, zclSeq: 121
16:43:13:912 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 122
16:43:14:076 [INFO] - Button 2001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Up, payload: 00C30000, zclSeq: 123
16:43:14:332 [INFO] - Button 2003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 124
16:43:14:337 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 125
16:43:14:847 [INFO] - Button 2001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Up, payload: 00C30000, zclSeq: 126
16:43:15:180 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 127
16:43:15:459 [INFO] - Button 2001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Up, payload: 00C30000, zclSeq: 0
16:43:15:764 [INFO] - Button 2003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 1
```

## Additional context
My zigbee network has around 20 devices. Most of them are various IKEA bulbs, switches, and outlets. I have two smart outlets from Innr and one from Enbrighten.

jakemauer, 2022-08-09:
Howdy, this issue still persists. Can this be reopened?

Mimiix, 2022-08-09:
Sure :)

jakemauer, 2022-08-10:
@Mimiix I'd love to contribute more to get this fixed. Are there specific things I can do to help debug or areas of the codebaseI should look to further diagnose the issue? I realize that the issue may not be with DeConz.

Mimiix, 2022-08-19:
@manup need to fix it i suppose. he promised me ages ago, but yet not there :(

jakemauer, 2022-09-11:
Would love input on how I can help debug this.

Paule-Panter, 2022-09-11:
just for Information my 2cent:

My Symfonisk Controller works as aspected.

`10:28:06:773 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 97
10:28:06:943 [INFO] - Button 3003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 98
10:28:07:121 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 99
10:28:07:301 [INFO] - Button 3003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 100
10:28:07:467 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 101
10:28:07:646 [INFO] - Button 3003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 102
10:28:07:895 [INFO] - Button 3001 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Move Down, payload: 01C30000, zclSeq: 103
10:28:08:075 [INFO] - Button 3003 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 104`

However my Controller have the "swversion": "2.1.024"

`{
    "config": {
        "alert": "none",
        "battery": 60,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "c223ea1cf62e869c09c53ea199fXXXXXX",
    "lastannounced": "2022-09-11T07:56:58Z",
    "lastseen": "2022-09-11T08:37Z",
    "manufacturername": "IKEA of Sweden",
    "mode": 1,
    "modelid": "SYMFONISK Sound Controller",
    "name": "WZ Lautstärkeregler Couchtisch",
    "state": {
        "buttonevent": 2003,
        "lastupdated": "2022-09-11T08:37:01.505"
    },
    "swversion": "2.1.024",
    "type": "ZHASwitch",
    "uniqueid": "ec:1b:bd:ff:fe:02:4b:2c-xxxxxxxx"
}`

Maybe there was something changed in the IKEA-FW.!?

jakemauer, 2022-09-13:
@Paule-Panter The firmware might have something to do with it, my Symfonisk controller is on firmware 2.3.080

closed 2022-10-13 by github-actions[bot]